AltBeacon/android-beacon-library #421: when several iBeacon transmitter is setup, startRangingBeaconsInRegion function can only get one UUID?

BoraDon, 2016-08-29:
### Expected behavior
when i use startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null);) in a room where two or three iBeacon transmitter is setup, i hope i can get all the UUID of these transmitters. then i cat pick up the one i needed.

### Actual behavior
didRangeBeaconsInRegion callback just print one UUID

### Steps to reproduce this behavior
print out the UUID in didRangeBeaconsInRegion function. call the function startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null);),  wait several seconds before stopRangingBeaconsInRegion("myRangingUniqueId", null, null, null);) and see the logcat in the eclipse.  just the same UUID is print out with different RSSI.

### Mobile device model and OS version
HUAWEI 7i & Android 5.1.1

### Android Beacon Library version
android-beacon-library-2.9

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2016-08-29:
Can you please test the [reference app](https://github.com/AltBeacon/android-beacon-library-reference) or the Locate app to see if you see the same behavior?  

I have verified this is not the behavior using tyese apps on the Nexus 5x and Nexus 9.

BoraDon, 2016-08-29:
thanks for you quick response, @davidgyoung. as i don't have android studio environment, i tried locate app in my HUAWEI 7i mobile phone. the same result. it can get one UUID when i put two iBeacon transmitters nearby.

davidgyoung, 2016-08-29:
See answer here: http://stackoverflow.com/a/39204928/1461050

closed 2016-08-29 by davidgyoung