AltBeacon/android-beacon-library #422: requestStateForRegion() method seems to be different from iOS

paulbao, 2016-09-01:
### Expected behavior
`didDetermineStateForRegion()` will be called when enter/exit the region on iOS devices

### Actual behavior
`didDetermineStateForRegion()` will be called only once with an `outside` state

### Steps to reproduce this behavior
Call `requestStateForRegion() ` only (**without calling 'startMonitoringBeaconsInRegion'**)

### Mobile device model and OS version
Any

### Android Beacon Library version
Version 2.9

Cheers

davidgyoung, 2016-09-01:
I don't understand. If you do not call startMonitoringBeaconsInRegion, why would you expect a callback with anything other than an outside state?

paulbao, 2016-09-01:
If you try to call requestStateForRegion **only** on iOS, it do have a callback with an inside state. I guess iOS called sth like `startMonitoringBeaconsInRegion ` inside `requestStateForRegion` because you'll have regions that you've passed to `requestStateForRegion` which have been  stored in `monitoredRegions`.

davidgyoung, 2016-10-02:
I see  I did not understand that iOS worked that way.  While it is possible to implement the equivalent behavior in this library, it doesn't fit in easily with the existing design.  I would be inclined to address this simply through documentation, noting that it works differently than iOS and that you must explicitly start monitoring for it to do anything. 

However, if you feel strongly that it should be implemented in an equivalent way to iOS and want to work on creating a pull request for it, I'd be happy to take a look.

paulbao, 2016-10-02:
I think adding this difference in the documentation is good enough.
Thanks again for providing such an amazing Lib.

davidgyoung, 2016-10-02:
Reopening so I at least have a reminder to update the documentation...