AltBeacon/android-beacon-library #423: How to ignore initial trigger if device is already inside a beacon region

LudySu, 2016-09-01:
Hi David,

This is an enhancement request. We read the document and searched stackoverflow.com. We have been using the AltBeacon library for a while and it works great, but now we need to do a small change. 

Our issue is, when the device is already inside a beacon region and we start the beacon scanning (`BeaconManager.startMonitoringBeaconsInRegion`), a beacon entry event will be triggered, which we don't want. We only want it triggered when crossing the boundary.

The idea comes from the Google Location API, which has a method `setInitialTriggerer(int)`, if it is set to `0` then no event will get triggered at the time the geofence monitoring is started (i.e. if the device is already inside a region, entry event won't get triggered). Is it possible to add a similar method to AltBeacon library?

### Mobile device model and OS version
Samsung S6 with Android 6.0.1

### Android Beacon Library version
2.6.1

Thanks,

davidgyoung, 2016-09-05:
Hi, @LudySu.  I'm not sure you are aware, but as of version 2.9, if a device is already inside a beacon region when scanning starts, didEnterRegion will not be called, whereas didDetermineStateForRegion will be called with an indicator that the region is inside.

I know this is not the API you suggest, but I thought it important to note that there is a different way to accomplish the same thing with the latest version of the library.

LudySu, 2016-09-13:
Hi @davidgyoung, thanks for your reply.

I tried the new version 2.9 and 2.9.1 and found the behavior is the same as the version we were using (2.6.1). That means if the device is already inside a beacon region and starts beacon monitoring by calling `BeaconManager.startMonitoringBeaconsInRegion`, the app can still get the beacon entry callback from `didEnterRegion` immediately after monitoring starts. But we don't want this kind of entry, what we want is only get beacon events when the device moves across the region boundary.

Thanks,

davidgyoung, 2016-09-13:
@LudySu,

Do you stop monitoring?  If you start monitoring for a region but never call stop, this should not happen with 2.9+.  You should only get an enter callback the first time you enter, provided the beacon us always in the vicinity and stays advertising at a sufficient rate.  

If you are not seeing this, then I am afraid whatever is causing the discrepancy would also prevent your enhancement request from working.

LudySu, 2016-09-13:
Hi @davidgyoung , thanks for your reply.

We never stop beacon monitoring, and our service is running in the background all the time. During the test, the device was no more than 0.5 meter away from the beacon, and an iOS device can ignore the entry event when already inside the beacon region. To be frank, the reason why we want this feature is to keep in consistent with our iOS app.

Is there anything else that may cause this discrepancy? If not, I will try to have a look at the source code to see how to achieve this. If you can suggest which part or commit should I look into that will be great.

Thanks,

davidgyoung, 2016-09-13:
Understood, consistency with iOS was on of the driving forces behind the change put in with version 2.9.

Specifically when do you see the second didEnterRegion callback?  Is it only at app startup?  Or do you see it periodically while the app is running.  Do you  see a didExitRegion callback before the didEnterRegion callback?

LudySu, 2016-09-14:
Hi @davidgyoung 

`didEnterRegion` only happens once when we start beacon monitoring. No any `didExitRegion` found.

The log is as below:

```
BeaconScannerService: ========= didDetermineStateForRegion "id1: 61687109-905f-4436-91f8-e602f514c96d id2: null id3: null"
BeaconScannerService: ========= didDetermineStateForRegion "id1: 00000000-0000-0000-0000-000000000100 id2: null id3: null"
BeaconScannerService: ========= didDetermineStateForRegion "id1: 23a01af0-232a-4518-9c0e-323fb773f5ef id2: null id3: null"
BeaconScannerService: ========= didDetermineStateForRegion "id1: 2f234454-cf6d-4a0f-adf2-f4911ba9ffa6 id2: null id3: null"
BeaconScannerService: ========= didDetermineStateForRegion "id1: 61687109-905f-4436-91f8-e602f514c96d id2: null id3: null"
BeaconScannerService: ========= didEnterRegion "id1: 61687109-905f-4436-91f8-e602f514c96d id2: null id3: null"
BeaconScannerService: ========= didRangeBeaconsInRegion "id1: 61687109-905f-4436-91f8-e602f514c96d id2: null id3: null"
```

The attached file is the full log with `BeaconManager.setDebug(true)`.
[beacon.txt](https://github.com/AltBeacon/android-beacon-library/files/473592/beacon.txt)

Thanks,

davidgyoung, 2016-09-15:
@LudySu, that log file was very helpful, particularly this line: 

```
09-15 10:19:49.381  5464  5464 D MonitoringStatus: Not restoring monitoring state because it was recorded too many milliseconds ago: 1473891589388
```

What this says is that your app has never saved the state of the beacon regions to persistent storage, perhaps suggesting some kind of failure to do so.  You can see another line where it thinks it saves its state to storage at a later time:

```
09-15 10:19:49.381  5464  5464 D MonitoringStatus: saveMonitoringStatusIfOn()
```

This is supposed to write the state to a file called "org.altbeacon.beacon.service.monitoring_status_state" in your application's home folder.  That may be failing for some reason (although no errors are logged, which means the code thinks it wrote to the file successfully.)  If we can figure out why, we may be able to solve your problem.  Can you check and see if that file is there, and if so what timestamp is has?

LudySu, 2016-09-15:
Hi @davidgyoung 

I know where the discrepancy comes from, what I thought was different from how this feature was implemented. The scenario was that our app launched first time since fresh installation, so there's no saved beacon data there. I thought the library is able to ignore entry event in that scenario.

Then I checked iOS again and found it behaved the same way. So everything works perfect.

Thanks very much for your help!

closed 2016-09-15 by LudySu

davidgyoung, 2016-09-15:
Ah, that explains it!  Glad this will work for you, @ludysu