AltBeacon/android-beacon-library #428: ArrayIndexOutOfBoundsException in Eddystone URL

alberolimpica, 2016-09-08:
**### Expected behavior**
Find Eddystone URL and show them in a log

**### Actual behavior**
kills the activity due to an error:
FATAL EXCEPTION: IntentService[BeaconIntentProcessor]
               java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
               at org.altbeacon.beacon.utils.UrlBeaconUrlCompressor.uncompress

**### Steps to reproduce this behavior**
I have 4 beacons, two of them have set an URL and the other don't. When reaching the ones that have URL is fine, they show the URL in the log, but when they receive the signal from the ones that don't have it gets an error and kills the activity.
We took a look to the code and realized that the problem was because of the non-URL beacons (we are only trying to show the URL.

The code given:
for (Beacon beacon: beacons) {
            if (beacon.getServiceUuid() == 0xfeaa && beacon.getBeaconTypeCode() == 0x10) {
                // This is a Eddystone-URL frame
                String url = UrlBeaconUrlCompressor.uncompress(beacon.getId1().toByteArray());
                Log.d(TAG, "I see a beacon transmitting a url: " + url +
                        " approximately " + beacon.getDistance() + " meters away.");
            }
}
When trying to uncompress, since the beacon detected doesn't have a URL the value is 0. We fixed this by changing a bit the code:
        for (Beacon beacon: beacons) {
            if (beacon.getServiceUuid() == 0xfeaa && beacon.getBeaconTypeCode() == 0x10) {
                byte[] array = beacon.getId1().toByteArray();
                if(array.length == 0)
                        return;
                final String url = UrlBeaconUrlCompressor.uncompress(array);

**### Mobile device model and OS version**
Xiaomi Redmi Note 3
Project TANGO tablet

**### Android Beacon Library version**
android-beacon-library:2.3.5+

**IMPORTANT**:  I tried and it works. I don't know if this issue happens often, but I thought that it might be interesting to post it in case someone needs it.

davidgyoung, 2016-09-08:
Thanks for this report.  If the identifier has a length of zero, what would you expect the call to `UrlBeaconUrlCompressor.uncompress(...)` to do?

It cannot be a valid URL, so it would have to return null, an empty String, or throw a different exception.

alberolimpica, 2016-09-08:
I'm using beacons that once you input an URL, the http://www.   cannot be erased. Could it be the problem?

alberolimpica, 2016-09-08:
What do you mean?

davidgyoung, 2016-09-08:
How do you suggest that the `UrlBeaconUrlCompressor.uncompress` method behave when passed a zero length array?

davidgyoung, 2020-02-28:
closing due to inactivity

closed 2020-02-28 by davidgyoung