AltBeacon/android-beacon-library #429: support bluetooth permissions with attribute required="false"

giorgio-zamparelli, 2016-09-09:
### Expected behavior
The library should work with the following permissions in the manifest.xml (note the android:required="false")
<uses-permission android:name="android.permission.BLUETOOTH" **android:required="false"** />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" **android:required="false"** />

The required="false" otherwise the app will not be installed on devices that do not have bluetooth.
In many apps the bluetooth functionality is a nice plus but should not forbid devices without bluetooth to use the app anyway without the bluetooth features

### Actual behavior
The app crashes at startup with error "android required false Need BLUETOOTH permission: Neither user 10077 nor current process has android.permission.BLUETOOTH"

### Steps to reproduce this behavior
Add the android:required="false" after the BLUETOOTH and BLUETOOTH_ADMIN permissions in the manifest

### Mobile device model and OS version
Samsung Galaxy S6 Edge SM-G925F
Android 6.0.1

### Android Beacon Library version
2.9.1

davidgyoung, 2016-09-09:
Good point, thanks for the report.  I know this was tested with earlier Android versions so I suspect that this crash may be new to Android 6.

chyrta, 2016-10-01:
Had the same issue on Samsung device came from my Rollbar stats. Definitely needs to be fixed.

davidgyoung, 2016-10-02:
@giorgio-zamparelli and @chyrta, do either of you have a test device you can use to reproduce this to see if my proposed fix in #438 solves the problem?

Also, @giorgio-zamparelli in your initial report, you say this happened with a Samsung Galaxy S6 Edge SM-G925F.  But that device has Bluetooth LE.  How is it possible that you saw the crash on that device?

giorgio-zamparelli, 2016-10-05:
yes I do own Samsung Galaxy S6 Edge SM-G925F and I will try to test whether #438 fixed the problem.
Not sure how but the crash is caused by the android:required="false" attribute if though the device has both Bluetooth and BLE.

davidgyoung, 2016-10-08:
I suspect this crash may be caused by Samsung Knox, a security framework which allows disabling bluetooth permissions for apps installed inside a Knox container.  The solution to this problem probably the same as the solution to that problem. 

See here:  https://github.com/AltBeacon/android-beacon-library/issues/440