AltBeacon/android-beacon-library #431: StartupBroadcastReceiver throws NullPointerException

Eylen, 2016-09-20:
### Expected behavior
StartupBroadcastReceiver should start without exception

### Actual behavior
StartupBroadcastReceiver is throwing NullPointer exception while trying to retrieve applicationContext as the context is null.

Line 29: 
`BeaconManager beaconManager = BeaconManager.getInstanceForApplication(context.getApplicationContext());`

In AndroidManifest.xml:
```
<receiver android:name="org.altbeacon.beacon.startup.StartupBroadcastReceiver">
            <intent-filter>
                <action android:name="android.intent.action.ACTION_EXTERNAL_APPLICATIONS_AVAILABLE"/>
                <action android:name="android.intent.action.BOOT_COMPLETED"/>
                <action android:name="android.intent.action.QUICKBOOT_POWERON"/>
                <action android:name="android.intent.action.ACTION_POWER_CONNECTED"/>
                <action android:name="android.intent.action.ACTION_POWER_DISCONNECTED"/>
            </intent-filter>
        </receiver>
```

Full stacktrace:
```
E/AndroidRuntime: FATAL EXCEPTION: main
       Process: net.ds.batto.debug, PID: 13024
       Theme: themes:{}
        java.lang.RuntimeException: Unable to start receiver org.altbeacon.beacon.startup.StartupBroadcastReceiver: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.Context android.content.Context.getApplicationContext()' on a null object reference
        at android.app.ActivityThread.handleReceiver(ActivityThread.java:2766)
        at android.app.ActivityThread.-wrap14(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1440)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5461)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.Context android.content.Context.getApplicationContext()' on a null object reference
       at android.content.ContextWrapper.getApplicationContext(ContextWrapper.java:107)
       at net.ds.batto.geofences.GeofenceObservationService.addGeofences(GeofenceObservationService.java:215)
      at org.altbeacon.beacon.startup.StartupBroadcastReceiver.onReceive(StartupBroadcastReceiver.java:51)
       at android.app.ActivityThread.handleReceiver(ActivityThread.java:2759)
       at android.app.ActivityThread.-wrap14(ActivityThread.java) 
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1440) 
       at android.os.Handler.dispatchMessage(Handler.java:102) 
       at android.os.Looper.loop(Looper.java:148) 
       at android.app.ActivityThread.main(ActivityThread.java:5461) 
       at java.lang.reflect.Method.invoke(Native Method) 
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) 
```

### Mobile device model and OS version
At least in Android 6.0.1
Device: Moto G

### Android Beacon Library version
2.9.1

davidgyoung, 2016-09-20:
Thanks for the report.  It is strange that the context is null when passed in to the broadcast receiver.  I see that you have added additional events in the manifest.  Is there one event in particular that causes this?

Eylen, 2016-09-20:
I have tried with BOOT and ACTION_POWER_CONNECTED and I think that happens in both cases.
It's true that's really strange as it has happened right now and I have been using this library with those same parameters for a long time.

davidgyoung, 2016-10-02:
@Eylen, are you sure this crash was seen with an unmodified version 2.9.1 of the library?  It appears as though somebody may have used a forked version of the Android Beacon Library here.

The crash log shows that the NullPointerException happened at org.altbeacon.beacon.startup.StartupBroadcastReceiver.onReceive(StartupBroadcastReceiver.java:51

But that file does not have a line 51.  It is only 34 lines long.  And it certainly doesn't make a call to net.ds.batto.geofences.GeofenceObservationService.addGeofences, as mentioned on the next item up on the stack trace.  

You can see the 2.9.1 version of StartupBroadcastReceiver here: 

https://github.com/AltBeacon/android-beacon-library/blob/7bfc78e044f7ea7db3b8514bcbcf51f4071524a7/src/main/java/org/altbeacon/beacon/startup/StartupBroadcastReceiver.java

Eylen, 2016-10-03:
It seems that's a copy of altbeacon a bit modified, so the problem could come from another place...
Close this one if you want and I will try to investigate further.
Thanks

davidgyoung, 2016-10-06:
OK, I'm closing this.  If you find there is an issue with the unmodified library, please re-open.

closed 2016-10-06 by davidgyoung