AltBeacon/android-beacon-library #435: Eid, Region, EidResolver and DataFactory

yodutouf, 2016-09-26:
I have to develop an application that detects Eid.
I start looking the eddystone-eid branch.

If I understand correctly the process  the eid resolution is done on demand for one beacon when the developper calls the EidResolver.
This means the MonitorNotifier interface will work correctly with a Region that contains eid as Identifier but not the fixed resolved id. 

What do you thing about:
* resolving the eid automatically when the EidResolver is configured
* Making works the region detection process based on the resolved id. This permits:
 * to keep it identical to the eddystone uid with region based on the namespace-id
 * to avoid region monitoring error when the eid switches

I look as well to the DataFactory, asking myself if we can merge both process. What do you think about? 

For both, DataFactory and EidResolver the calls to get the content is done beacon by beacon. 
What do you think about an other implementation that permits to get a beacon content or an eid resolution for an array of beacons ?

davidgyoung, 2016-09-29:
Yes, @yodutouf, in general I think this approach sounds good, although I'm not sure whether it makes sense to use `BeaconDataFactory` or a new construct to make the remote requests.  (`BeaconDataFactory` was designed for the 1.x version of the library and has not been used since the 2.0 release.)

I think the tricky part of this is making region monitoring work.  The scanning service would have to make calls to resolve each unique EID beacon identifier it sees, and after it gets a result, try to match to monitored regions and issue notifier callbacks on matches.  It would be critical to design this in a way that doesn't overwhelm the system with lots of ongoing network requests in the case where a bunch of EID beacons are in the vicinity.

yodutouf, 2016-09-29:
I see two solutions to the network requests problems : using the batch system describes in the [google documentation](https://developers.google.com/beacons/proximity/how-tos/batch) or doing the batch job from an "other than google" backend server (with the possibility to resolved and getting the content in one ws call).

I see what you mean when you speak about the tricky part with the region monitoring.
I have to move on it, so I propose to do a fork to make a try. 
When I finish, can I make you PR to get your views ?

davidgyoung, 2016-09-29:
Good plan, @yodutouf

davidgyoung, 2019-10-24:
Closing due to inactivity

closed 2019-10-24 by davidgyoung