AltBeacon/android-beacon-library #440: Bluetooth isEnabled() call can give an exception.

inbeacon, 2016-10-07:
For some reason, it is sometimes not allowed to call bluetoothAdapter.isEnabled() resulting in an app crash with
`Fatal Exception: java.lang.SecurityException: Need BLUETOOTH permission: Neither user 10196 nor current process has android.permission.BLUETOOTH.
`
See also http://stackoverflow.com/questions/33245977/java-lang-securityexception-bluetooth-permission-crash-samsung-devices-only

This seems to be a Samsung issue, but a simple try/catch for SecurityException around the call will solve this issue.

Crash happened in altbeacon version 2.9  and is still an issue in 2.9.1

```
#0. Crashed: main: 0 0 0x0000000000000000
       at android.os.Parcel.readException(Parcel.java:1540)
       at android.os.Parcel.readException(Parcel.java:1493)
       at android.bluetooth.IBluetooth$Stub$Proxy.isEnabled(IBluetooth.java:1219)
       at android.bluetooth.BluetoothAdapter.isEnabled(BluetoothAdapter.java:707)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:250)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:239)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:235)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:145)
       at android.app.ActivityThread.main(ActivityThread.java:6145)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

```

the crash in finishScanCycle happens here:
```
                    if (getBluetoothAdapter() != null) {
                        if (getBluetoothAdapter().isEnabled()) {   <-- crash
```

in ScanLeDevice there is already a try/catch around this call:

```
                try {
                    if (getBluetoothAdapter() != null) {
                        if (getBluetoothAdapter().isEnabled()) {
                    ....
                } catch (Exception e) {
                    LogManager.e(e, TAG, "Exception starting Bluetooth scan.  Perhaps Bluetooth is disabled or unavailable?");
                }
```

so this might simply be copied into finishScanCycle

davidgyoung, 2016-10-08:
@inbeacon, thank you for this report!  I suspect this may be the cause of a similar issue described here:

See this related issue: https://github.com/AltBeacon/android-beacon-library/issues/429

Do you happen to have  a test device with Samsung Knox that could be use to try out possible fixes?

inbeacon, 2016-10-18:
hi David, 
no, we have only a crashlytics report. We encounter this in an 0,5% update of an app on a number of samsung devices, so we don't have the actual device on which it happens.
However, a simple try/catch will solve the problem. 

[image]

nishkarsh, 2016-12-13:
@davidgyoung, This seems to be an issue with not only Samsung devices. We have been getting similar crash on **Panasonic ELUGA Z** (version 4.4.2), **Lenovo P1a42** (version 5.1.1) and **Moto G (4)** (version 6.0.1)

```
Fatal Exception: java.lang.SecurityException: Need BLUETOOTH permission: Neither user 10136 nor current process has android.permission.BLUETOOTH.
       at android.os.Parcel.readException(Parcel.java:1546)
       at android.os.Parcel.readException(Parcel.java:1499)
       at android.bluetooth.IBluetooth$Stub$Proxy.isEnabled(IBluetooth.java:849)
       at android.bluetooth.BluetoothAdapter.isEnabled(BluetoothAdapter.java:603)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(Unknown Source)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(Unknown Source)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(Unknown Source)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.app.ActivityThread.main(ActivityThread.java:5383)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:939)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:734)
```
We had been using library version 2.7.1 till now. We have not got the crash on our test devices so can't confirm whether latest release 2.9.2 fixes this.

davidgyoung, 2016-12-13:
Thanks for the info, @nishkarsh.  Please advise when you get 2.9.2 out into production whether you continue to see this.

davidgyoung, 2017-03-11:
Any update here?

nishkarsh, 2017-03-16:
@davidgyoung We haven't pushed the app to production yet so still not sure whether it's fixed in the newer version. We are planning to push in a few days, will update you after observing for sometime after we push the app update.