AltBeacon/android-beacon-library #441: Not able to delete All Regions in case you do not know their exact name

mmarshak, 2016-10-13:
### Expected behavior

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

mmarshak, 2016-10-13:
I created one beacon region to monitor (it had minor number of 1 and identifier called "beacon1"), and I wanted to test if I have away to delete it at all times, if I do not know its name.  I made sure that the sample app can detect entry and exit. At that time I stopped (via Android studio) the app, and change the region's identifier (to "beacon2") and the minor number (to minor 2) I was looking for, and restarted the app. The sample (reference) app detected the original beacon (the one with identifier "beacon1").

Upon entry to didEnterRegion I wanted to see which regions the beaconManager is aware of so I did -
ArrayList<Region> beaconRegions = (ArrayList<Region>) beaconManager.getMonitoredRegions();
int i = 0;
for (Region region1 : beaconRegions){
            Log.d(TAG, "Region # " + i + " Identifier = "+  region1.getUniqueId() + " UUID = " + region1.getId1() + " Major = " + region1.getId2() + " Minor = " + region1.getId3());
            ++i;
}

To my surprise it listed only 1 beach, the beacon with identifier "beacon2" (as expected) but did not listed "beacon1" even thought we just detected entry.

So there is no way to stop monitoring for the first region because I do not know his name (I kinda know as this was the argument for the entry). 

So the problem is that if an app lose a track getMonitoredRegions() will return all the regions that really exist.

The first beacon is being recreated every time I restart the app because the library save it on Persistence storage. It look like the first beacon is around after the restart of the app due to the call of restoreMonitoringStatus that load the saved regions from the file "org.altbeacon.beacon.service.monitoring_status_state". 

I found the following two methods to remove the file and enable persistence storage, but the docs says not to use them by client. The first one delete the file. Is there any clean solution to delete all regions if I lose track of the regions names? I want to be able to stop all regions.

   
public synchronized void stopStatusPreservation() {
        this.mContext.deleteFile("org.altbeacon.beacon.service.monitoring_status_state");
        this.mStatePreservationIsOn = false;
    }

    public synchronized void startStatusPreservation() {
        if(!this.mStatePreservationIsOn) {
            this.mStatePreservationIsOn = true;
            this.saveMonitoringStatusIfOn();
        }

    }

davidgyoung, 2016-10-13:
> ArrayList beaconRegions = (ArrayList) beaconManager.getMonitoredRegions();

Good point, @mmarshak.  The problem is that
`beaconManager.getMonitoredRegions()` returns a collections stored on the
`BeaconManager`, which only includes regions that the user actually started
monitoring since startup.  Whereas state restoration of previously
monitored regions comes from `MonitoringStatus`.

I agree this needs to be fixed.  The simplest solution would probably be to
get rid of the storage of the collection of monitored regions on
`BeaconManager` and simply use the one from `MonitoringStatus`.  That way
two collections do not have to be kept in sync.

mmarshak, 2016-10-14:
I found another solution, which is kind of a hack.
If I want to get rid of all the regions on disk, you can use -
 beaconManager.setRegionStatePeristenceEnabled(false);
beaconManager.setRegionStatePeristenceEnabled(true);

mmarshak, 2016-10-14:
But again, this will not help in the general case where I want to stop all the beacon during run.

mmarshak, 2016-10-14:
@danialgoodwin why not keep the region on storage (as they are required for restarts and power cycle), and when somebody define a new beaconManager, just asscoiate the existing regions from disk with this beaconManager, this way the app will have a consistance way to get all the regions from its beaconManager?

davidgyoung, 2016-10-14:
Yes, I'm not suggesting that the `beaconManager.getMonitoredRegions()` be eliminated.  I am just saying that it could behind the scenes just fetch the regions from `MonitoringStatus`.

mmarshak, 2016-11-21:
Hi @davidgyoung , do you know when you will be able to fix that?

davidgyoung, 2016-11-22:
Let me try to take a crack at it today

davidgyoung, 2016-11-25:
Closing as fix has been merged to master and is pending release.

closed 2016-11-25 by davidgyoung