AltBeacon/android-beacon-library #443: Ranging Issue

arpitjoshi08, 2016-10-19:
hello ... i want that ranging continuous detect beacon after app closed but i checked log my ranging start when i get any event of enter or exit. i follow this steps:
1) start ranging when beacon detect
2) app killed from background 
3) ranging stop not showing anything
4) switched off beacon and then again on then ranging start automatically and my app has killed.
so how can i continuous ranging when app closed without enter or exit beacon manually

davidgyoung, 2016-10-19:
@arpitjoshi08, you can kick off ranging without an entry event by putting your code inside the `didDetermineStateForRegion`, which is always called at app startup.

If you have further questions please post on StackOverflow, as that is the preferred forum for support questions.  This forum is reserved for feature requests and bug reports with the library itself.

closed 2016-10-19 by davidgyoung

arpitjoshi08, 2016-10-19:
but i want that without open app again how can i perform ranging continuously for detection of beacon ?