AltBeacon/android-beacon-library #448: java.lang.IndexOutOfBoundsException from Beacon.getId3() and Eddystone

bluetooth-mdw, 2016-10-28:
### Expected behavior
If application encounters an Eddystone beacon, it will only have two IDs and thus id3 in a beacon object will not contain a value. Calling getId3, which is supposed to be a convenience method, just behave nicely in this situation, ideally returning an object whose value means "not present" as opposed to simply returning null. Then calling methods like toHexString on such an Identifier object could return an empty String.

### Actual behavior
If you call getId3() on a Beacon object created for an Eddystone beacon you get an IndexOutOfBoundsException like this:

```
java.lang.IndexOutOfBoundsException: Invalid index 2, size is 2
                                                                                        at java.util.ArrayList.throwIndexOutOfBoundsException(ArrayList.java:255)
                                                                                        at java.util.ArrayList.get(ArrayList.java:308)
                                                                                        at org.altbeacon.beacon.Beacon.getId3(Beacon.java:337)
```

### Steps to reproduce this behavior
Define a Redion with Identifier objects set to null so that all beacons are detected. Place an Eddystone beacon in range. Code which calls getId3() on a Beacon object will fail with this Exception.

### Mobile device model and OS version
Nexus 5, Android 6

### Android Beacon Library version
2.9.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

Thanks! :-)

davidgyoung, 2016-10-28:
Good point, @bluetooth-mdw.  

The simplest solution would be to have the method return null.  However, this may still cause a  NullPointerException later on in application code, and improvement for sure, but not as convenient as your suggestion of returning an Identifier instance that wraps a null using the Null Object pattern.

Adding a special null Identifier instance concerns me a bit, because it seems like it would lead to a breaking API change.  Would the API then require passing this in when creating regions?  Like this:  `new Region("my region", Identifier.nullInstance(), Identifier.nullInstance(), Identifier.nullInstance())` instead of `new Region("my region", null, null, null)`

Bottom line:  I think the simplest incremental change is to return null, since passing null for Identifier instances is allowed for other cases of the API.

bluetooth-mdw, 2016-10-28:
You're right. Nothing's ever simple :-) 

Cheers

Martin

Oupsla, 2016-12-02:
Hello,

I got the same Error with an Eddystone beacon.
When I set a null region to detect all beacons, I got a NullPointerException when my device detects the beacon.

```
java.lang.RuntimeException: An error occurred while executing doInBackground()
                                                                   at android.os.AsyncTask$3.done(AsyncTask.java:309)
                                                                   at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
                                                                   at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
                                                                   at java.util.concurrent.FutureTask.run(FutureTask.java:242)
                                                                   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                   at java.lang.Thread.run(Thread.java:818)
                                                                Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'void org.altbeacon.beacon.service.RssiFilter.addMeasurement(java.lang.Integer)' on a null object reference
                                                                   at org.altbeacon.beacon.service.RangedBeacon.addMeasurement(RangedBeacon.java:72)
                                                                   at org.altbeacon.beacon.service.RangedBeacon.updateBeacon(RangedBeacon.java:39)
                                                                   at org.altbeacon.beacon.service.RangedBeacon.<init>(RangedBeacon.java:34)
                                                                   at org.altbeacon.beacon.service.RangeState.addBeacon(RangeState.java:56)
                                                                   at org.altbeacon.beacon.service.BeaconService.processBeaconFromScan(BeaconService.java:439)
                                                                   at org.altbeacon.beacon.service.BeaconService.access$500(BeaconService.java:78)
                                                                   at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:484)
                                                                   at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:459)
                                                                   at android.os.AsyncTask$2.call(AsyncTask.java:295)
                                                                   at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                                   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
                                                                   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
                                                                   at java.lang.Thread.run(Thread.java:818) 
```

What is the solution when I create my region ?
```java
public void onBeaconServiceConnect() {
        Region region = new Region("all-beacons-region", null, null, null);
        try {
            beaconManager.startRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        beaconManager.addRangeNotifier(this);
    }
```