AltBeacon/android-beacon-library #453: Android 6, never getting onScanResult()

eckoj433, 2016-11-18:
Im having trouble getting altbeacon framework to work properly on Android 6. 
At first it was not scanning at all, but i found a simmilar issue and managed to get it to scan by enabling location services.

Now the framework started scanning, however i never get any onScanResult(), despite the beacon being right next to me. The device im testing on is a Samsung Galaxy s6. The problem might be samsung specifik, as they modify the android software with all kinds of stupid stuff (pardon my bias), but the samsung is a really popular device, so we need to make it work on it somehow.

davidgyoung, 2016-11-18:
Can you please turn on debug logging with `beaconManager.setDebug(true)` and capture and attach a logcat sequence when beacons are in the vicinity?  Also, cam you detect beacons on the same eevice with an off the  shelf detector based on the same library like Locate:  https://play.google.com/store/apps/details?id=com.radiusnetworks.locate

eckoj433, 2016-11-21:
Found the bug, aperently despite having targetframework = 22, and having permissions for fine_location coarse_location and bluetooth in the manifest, the app did not recieve the permissions to be able to scan. had to go to settinings find the app and manually turn on location for the app in order to get the scanning to work.

So it appears that the scanning will not work without runtimepermissions despite targetting api 22. Any one experience simmilar issue?Certainly there must be a way to scan on android 6 without targetting api 23?

davidgyoung, 2016-11-21:
From your description it sounds like the OS-wide "Location" setting was turned off, which must always be turned on for an app to be able to perform Bluetooth scans as of Android 6 regardless of target SDK.  Don't confuse  the OS-wide location on/off setting with the app-specific  dynamic location permission, which is a separate thing, and is also required to scan  if targeting SDK 23.

eckoj433, 2016-11-21:
No, the OS-wide location was on, along with the improved accurasy "bluetooth scanning", however when i went to setting->applications and klicked on my app, it had the location permission switched off. Like when you click no on a runtime permission, however the application  targetted api 22 so im not sure why this happend.

davidgyoung, 2016-11-21:
Can you reproduce after uninstalling reinstalling?

eckoj433, 2016-11-21:
I'll see if i can reproduce the bug when i get the time. My application is actually supposed to target API 24, so its not really an issue for me sence i need to use runtime permissions anyway, just figured id post about it incase someone else is having the same issue.

eckoj433, 2016-11-21:
just realised that i was overwriting the manifest targetsdk in the the gradle build file, so i was actually targetting 24 when i thought i was targeting 22.

closed 2016-11-21 by davidgyoung