AltBeacon/android-beacon-library #454: Android 6 UI freeze when scanning

eckoj433, 2016-11-21:
Im having lagg issues with Android 6.0, simmilar to issue #313 
However asking for permissions runtime and allowing them, does not have any effect. 

periodically about once every second, the ui just freezes for 200-300 ms,  and this is what i get in the log when the lagg occurs:

`11-21 12:34:01.843 15358-15358/com.cgi.beaconframework D/BluetoothAdapter: STATE_ON
11-21 12:34:01.853 15358-15358/com.cgi.beaconframework D/BluetoothAdapter: STATE_ON
11-21 12:34:01.853 15358-15358/com.cgi.beaconframework D/BluetoothLeScanner: Stop Scan
11-21 12:34:01.863 15358-15358/com.cgi.beaconframework D/BluetoothLeScanner: Start Scan
11-21 12:34:01.873 15358-15358/com.cgi.beaconframework D/BluetoothAdapter: STATE_ON
11-21 12:34:01.873 15358-15358/com.cgi.beaconframework D/BluetoothAdapter: STATE_ON
11-21 12:34:01.873 15358-15358/com.cgi.beaconframework D/BluetoothAdapter: STATE_ON
11-21 12:34:01.883 15358-15358/com.cgi.beaconframework D/BluetoothAdapter: STATE_ON
11-21 12:34:01.923 15358-15370/com.cgi.beaconframework D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=7`
this is with setDebug(false).

attatching a more extensive log with setDebug(true).

[log.txt](https://github.com/AltBeacon/android-beacon-library/files/603425/log.txt)

If i move to an isolated area where there are no beacons in range, the lagg stops, but as soon as there are beacons in range the lagg starts again. Could there be some other permission that im missing which is causing this?

eckoj433, 2016-11-21:
found an issu #136 which seems to be the same issu, which actually has a solution pullrequest #430  with a seperate thread being used for the scans, which was merged into master on october 20th. However i cant seem to find a binary download for this version. the 2.9.2-beta binary seems to include these changes, so ill give that a try. Any idea on when a stable release including these changes will be up?

davidgyoung, 2016-11-21:
Yes, please try 2.9.2-beta.  I was hoping to get confirmation on two hard to reproduce bugs, one specific to LG and the other to Samsung Knox before a stable  2.9.2 release.  If I cannot get that confirmation and you need this change sooner, I can hold off those fixes until 2.9.3.  Please let me know if this works for you.

eckoj433, 2016-11-21:
2.9.2-beta solved the issue. For me it doesnt really matter if i use this beta or a release labled "stable", however this really is a huge improvement in performance for android 6+ devices, so i think it would be good if there was an official binary for it asap.

davidgyoung, 2016-11-25:
Closing issue as the referenced solution has been merged to master and is pending the next release.

closed 2016-11-25 by davidgyoung

himanshu-android, 2018-02-12:
Hi, I am using the latest version 2.12.4 which has same behavior of UI freeze when binding and unbinding to the scan service. I am able to reproduce this issue on my device that is Moto G4 Plus running Nougat (7.0).