AltBeacon/android-beacon-library #455: Long time to detect my beacon in the background after passing by somebody else's beacon is present

mmarshak, 2016-11-22:
My app is able to detect iBeacon in the background, I configure few regions with UUID, major and minor. If the app is running in the background it detect the iBeacon immediately, however after the initial detect, it will wait for 5 minutes until it will be detect again. I read from the documentation that this is the expected default behavior in order to save battery, but is there away to shorten the 5 minutes only after detect my beacon?

However a more serious issue is that if my app is running in the background, and detect somebody else Beacon, I see from the logs that is start the 5 minute wait, however if shortly after detecting somebody else Beacon, my app get close to my beacon that is define by the region I set, it does not detect my beacon for 5 minutes, and this is a big problem as the user my be far away from my beacon and I want to trigger as soon as I see him.

I tested it with Android L, as well with Android M

In the code -
https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/scanner/CycledLeScannerForLollipop.java#L137

It says-
Android 5 background scan algorithm (largely handled in this method)
      Same as pre-Android 5, except when on the between scan period.  In this period:
      If a beacon has been seen in the past 10 seconds, don't do any scanning for the between scan period.
         Otherwise:
           - create hardware masks for any beacon regardless of identifiers
           - look for these hardware masks, and if you get one, report the detection
      when calculating the time to the next scan cycle, make it be on the seconds modulus of the between scan period plus the scan period
      This is an improvement over the current state, but the disadvantages are:
         _**- If a somebody else's beacon is present and yours is not yet visible when the app is in
           the background, you won't get the accelerated discovery.  You only get the accelerated
           discovery if no beacons are visible before one of your regions appears.  Getting around
           this would mean setting up filters that are specific to your monitored regions, which is
           a possible future improvement.**_

What can I do if I want to get the accelerated discovery after seeing somebody else beacon?
This is really wrong, what is the point of setting iBeacon region that is being ignored in the background if the user by chance passed by somebody else iBeacon?

Logs bellow -
11-21 22:08:31.289 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 79555 milliseconds
11-21 22:08:32.291 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 78554 milliseconds
11-21 22:08:33.292 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 77553 milliseconds
11-21 22:08:34.293 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 76551 milliseconds
11-21 22:08:35.294 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 75550 milliseconds
11-21 22:08:36.296 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 74549 milliseconds
11-21 22:08:37.297 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 73547 milliseconds
11-21 22:08:38.298 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 72546 milliseconds
11-21 22:08:38.907 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
11-21 22:08:38.916 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got a filtered scan result in the background.
11-21 22:08:38.931 29111-29571/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
11-21 22:08:38.931 29111-29571/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215a78762a64fa24e988024bc5b71e0893e93f400b0b308094b6f6e74616b74020af40a160dd030427a4b333162000000000000000000 with startIndex: 5, endIndex: 29
11-21 22:08:38.931 29111-29571/org.altbeacon.beaconreference D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
11-21 22:08:38.931 29111-29571/org.altbeacon.beaconreference D/BeaconParser: Bytes are: 0201061aff4c000215a78762a64fa24e988024bc5b71e0893e93f400b0b308094b6f6e74616b74020af40a160dd030427a4b333162000000000000000000
11-21 22:08:38.936 29111-29571/org.altbeacon.beaconreference D/BeaconService: beacon detected : id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:38.937 29111-29571/org.altbeacon.beaconreference D/MonitoringStatus: This region (id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 12) does not match beacon: id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:38.937 29111-29571/org.altbeacon.beaconreference D/MonitoringStatus: This region (id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 11) does not match beacon: id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:38.937 29111-29571/org.altbeacon.beaconreference D/MonitoringStatus: This region (id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 1) does not match beacon: id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:38.937 29111-29571/org.altbeacon.beaconreference D/BeaconService: looking for ranging region matches for this beacon
11-21 22:08:39.299 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Delivering Android L background scanning results
11-21 22:08:39.299 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 71546 milliseconds
11-21 22:08:40.300 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Delivering Android L background scanning results
11-21 22:08:40.300 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 70544 milliseconds
11-21 22:08:41.156 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
11-21 22:08:41.156 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got a filtered scan result in the background.
11-21 22:08:41.160 29111-29620/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
11-21 22:08:41.161 29111-29620/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 0201061aff4c000215a78762a64fa24e988024bc5b71e0893e93f400b0b308094b6f6e74616b74020af40a160dd030427a4b333162000000000000000000 with startIndex: 5, endIndex: 29
11-21 22:08:41.162 29111-29620/org.altbeacon.beaconreference D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
11-21 22:08:41.163 29111-29620/org.altbeacon.beaconreference D/BeaconParser: Bytes are: 0201061aff4c000215a78762a64fa24e988024bc5b71e0893e93f400b0b308094b6f6e74616b74020af40a160dd030427a4b333162000000000000000000
11-21 22:08:41.167 29111-29620/org.altbeacon.beaconreference D/BeaconService: beacon detected : id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:41.169 29111-29620/org.altbeacon.beaconreference D/MonitoringStatus: This region (id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 12) does not match beacon: id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:41.171 29111-29620/org.altbeacon.beaconreference D/MonitoringStatus: This region (id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 11) does not match beacon: id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:41.172 29111-29620/org.altbeacon.beaconreference D/MonitoringStatus: This region (id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 1) does not match beacon: id1: a78762a6-4fa2-4e98-8024-bc5b71e0893e id2: 45322 id3: 176
11-21 22:08:41.174 29111-29620/org.altbeacon.beaconreference D/BeaconService: looking for ranging region matches for this beacon
11-21 22:08:41.301 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Delivering Android L background scanning results
[... cut ...]
11-21 22:09:29.391 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 21453 milliseconds
11-21 22:09:30.392 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 20452 milliseconds
11-21 22:09:31.394 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 19451 milliseconds
11-21 22:09:32.395 29111-29111/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 18449 milliseconds

davidgyoung, 2016-11-22:
> master/src/main/java/org/altbeacon/beacon/service/scanner/
> CycledLeScannerForLollipop.java#L137

Yes, you are correct about the limitations of the background scanning
strategy.  If another recognized beacon is detected in the vicinity, even
if it is not yours, it will slow down background scans to once every 5
minutes.

The reason this happens is because fast background detections rely on
hardware filters which are based on matching a known byte sequence at the
start of the advertising packet.  If hardware filters cannot be used, then
full scans must be executed which use far more battery, so they must be
throttled, by default to every 5 minutes.

The library's hardware filter byte sequence is based on the beacon type
pattern without any identifiers or partial identifiers included.  Because
of this, any other detected beacon using the same registered format as
yours will match the filter.  If a filter is matched, scan results will
start returning packets to software which consumes extra battery.  To
mitigate this, the library backs off scans to 10 seconds once every five
minutes once this happens.

You can increase this background scan rate to one second every six seconds
with commands like below:

`beaconManager.setBackgroundScanPeriod(1000l);
beaconManager.setBackgroundBetweenScanPeriod(5000l);`

This will use a similar duty cycle, but will use somewhat more battery
because of forcing more context switching on the operating system.

The alternative is to implement partial identifiers as part of the hardware
filter, which would provide a faster detection of only your beacons even of
other beacons are in the vicinity. **I generally think this is a bad
idea.** This is the strategy iOS uses, and apps often fail to get hardware
filters because they are a limited resource that can be taken up by other
apps before your app gets one, leading to silent failures and unexpected
detection delays.  By not making filters specific to an identifier, it is
much less likely that the scarce resource will run out on any given phone.

I would be open to  PR for a code change that would allow making the filter
specific to one or more region identifiers, but I do not think this should
be the default behavior in the library for the reason described above.

mmarshak, 2016-11-22:
@davidgyoung thank you for the answer.

If I plan to use the background scanning for at most for 1 hour every few days on the app, what will be the impact of using the solution you suggested -

`beaconManager.setBackgroundScanPeriod(1000l);
beaconManager.setBackgroundBetweenScanPeriod(5000l);` ?

Alternative solution, which is more complicated is define only one region without any UUID, major or minor, and in the methods didEnterRegion, didExitRegion take action on beacon that match the UUID, major, minor that you care about. Now the trick is in the method didDetermineStateForRegion that will be now called for every beacon (because we did not specify the UUID, major, minor) is to start a short timer of 5 minutes.  Every time we enter didDetermineStateForRegion we call -
beaconManager.setBackgroundScanPeriod(1000l);
beaconManager.setBackgroundBetweenScanPeriod(5000l);
And we save in the shared pref a timestamp when we set them. 
If no other beacon is detected in 5 minutes, the timer will expire and we can go to the default values -
  beaconManager.setBackgroundScanPeriod(BeaconManager.DEFAULT_BACKGROUND_SCAN_PERIOD);
		beaconManager.setBackgroundBetweenScanPeriod(BeaconManager.DEFAULT_BACKGROUND_BETWEEN_SCAN_PERIOD);

If during these 5 minutes we find another beacon we cancel the old timer, and create a new timer to expire in 5 minutes.    
This is more complicated then your first suggested, but with this method we will need to pay the penalty of of faster scan in the background only during 5 minutes after each time we find a new beacon in the background.
We can add even an optimization by removing this region we build for the background scanning when we detect the app went to be in the foreground (the library already track the application life cycle, so this should not be too hard). Thoughts?

If your 1st suggest have small impact on the battery if I use it 1 hour every 2-3 days, then I will go with that option.

davidgyoung, 2016-11-22:
Doing this for one hour every few days is no problem at all from a battery perspective.  Best to keep it simple if you can!

mmarshak, 2016-11-23:
Hi @davidgyoung, I implemented a logic that when I set the background scan parameters as you suggested -
`beaconManager.setBackgroundScanPeriod(1000l);
beaconManager.setBackgroundBetweenScanPeriod(5000l);`

And after few minutes I call those methods with the default values -
mBeaconManager.setBackgroundScanPeriod(BeaconManager.DEFAULT_BACKGROUND_SCAN_PERIOD);
		mBeaconManager.setBackgroundBetweenScanPeriod(BeaconManager.DEFAULT_BACKGROUND_BETWEEN_SCAN_PERIOD);

I did an experiment where I am near the beacon and I have the short time duration for the scan and the interval in the background, I put the app in the background. The app is doing ranging. After few minute my timer expire and I call the methods to set the very slow scan (the 5 minutes with 10 seconds scan). What I notice that it did not make a difference, I still get ranging every 5 seconds, and it did not revert to the 5 minutes. Only when I took the beacon far away the library detected the EXIT event and only then it revert to the new setting for 5 minutes scan in the background.

I was expected that when I will call setBackgroundScanPeriod and setBackgroundBetweenScanPeriod it will take effect very quickly, but it does not. Is this a bug or I am doing something wrong?

Thanks

davidgyoung, 2016-11-23:
If changing scan periods after you are already bound to the service, you must call beaconManager.updateScanPeriods() to make the change take effect.

mmarshak, 2016-11-23:
Thank you for the pointer, I just verified that this works.

closed 2016-12-10 by davidgyoung