AltBeacon/android-beacon-library #458: DidEnterRegion is not called when app starts with beacon in immediate vicinity

mikhailmelnik, 2016-11-22:
### Expected behavior

DidDetermineStateForRegion and DidEnterRegion are called sequentially after app start.

### Actual behavior

DidDetermineStateForRegion  is called but DidEnterRegion is not.

### Mobile device model and OS version

LG G3 (D855), Android 5.0

### Android Beacon Library version

Starting from 2.9

davidgyoung, 2016-11-22:
This is expected behavior.  Please see:

http://altbeacon.github.io/android-beacon-library/state-persistence.html

closed 2016-11-22 by davidgyoung

mikhailmelnik, 2016-11-22:
Sorry @davidgyoung, somehow was not able to find this before and thanks for explanation. 

The application I'm working on is using DidEnterRegion callback to start ranging process on Application level so as I understand in my case I have to use one of the suggested "workarounds".

davidgyoung, 2016-11-23:
Understood.  I recommend using the didDetermineStateForRegion callback for starting ranging.  You are guaranteed to get a callback to that method after app startup.