AltBeacon/android-beacon-library #459: Empty collection of beacons in didRangeBeaconsInRegion

andretietz, 2016-11-23:
### Expected behavior
At least one beacon in the collection of beacons on the didRangeBeaconsInRegion call

### Actual behavior
No item in the collection
[image]
### Steps to reproduce this behavior

### Mobile device model and OS version
Google Pixel - Android v7.1

### Android Beacon Library version
2.9.1

davidgyoung, 2016-11-25:
This is expected behavior.  When ranging is turned on, even when no beacons are visible, the `didRangeBeaconsInRegion` callback will be made at the end of each scan cycle (by default every 1100 ms in the foreground and 310000 ms in the background.)  When no beacons are visible, this callback will be made with a collection containing zero beacons.  This is equivalent to the way the CoreLocation APIs work on iOS.

See code excerpt and sample LogCat output below.

     @Override
      public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {  
        Log.d(TAG, "didRangeBeaconsInRegion count:"+beacons.size());
     }

Logcat output:

```
11-25 09:37:13.186 1298-4703/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:14.316 1298-4721/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:15.439 1298-4740/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:16.563 1298-4759/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:17.682 1298-4779/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:18.792 1298-4797/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:19.909 1298-4816/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:21.033 1298-4834/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:22.146 1298-4855/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:23.259 1298-4876/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:24.374 1298-4902/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
11-25 09:37:25.538 1298-4925/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion count:0
```

closed 2016-11-25 by davidgyoung

andretietz, 2016-11-26:
@davidgyoung 
This doesn't explain, why I receive beacon url's on all other devices except the Google Pixel

davidgyoung, 2016-11-27:
@andretietz, indeed, if you are not detecting beacons s on the Google Pixel but you are on other devices, that is a different problem.  There are many reasons why this may happen, most of which are OS-specific.  First things to check:

* Verify location services are turned on in settings
* Verify location permission has been granted to the app http://altbeacon.github.io/android-beacon-library/requesting_permission.html

                  a  .

andretietz, 2016-11-27:
@davidgyoung There was no change in the permissions API between Android 6 (which comes with the Nexus 6P as min) and Android 7 (Pixel). This tells you, that if I wouldn't have requested the permissions correctly, it wouldn't work on non of the devices (6P and Pixel as described in the first post)

davidgyoung, 2016-11-28:
Sorry for the misunderstanding, @andretietz.  I misunderstood your statement above that it is working on the Nexus 6P Android 7.0.  I had read that as saying it was NOT working on that device. 

Can you please open a new issue on this, perhaps with the title "Pixel 7.1 fails to detect beacons"?  This issue describes an empty collection as being the problem, but that is only a problem if beacons are not detected on the device.  Making a new issue and linking to it will make this clear that the core problem is that the Pixel with 7.1 is not detecting for you at all.

A couple of things you might try to collect to help me figure out what is going on and include in this issue:

1. Try to use the Locate App (which uses the same library) on the Pixel and see if it works out of the box.  https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en

2. Scan for bluetooth devices in the operating system Bluetooth Settings, and see if you can detect anything.

3. In your app, turn on the Android Beacon Library debug logging and attach a capture of its output from LogCat.  You can turn this on with this line of code:  `beaconManager.setDebug(true);`