AltBeacon/android-beacon-library #462: Eddystone and Distance

Oupsla, 2016-12-02:
Similar to https://github.com/AltBeacon/android-beacon-library/issues/448

### Expected behavior

Working with Eddystone beacon and trying to get a distance.

### Actual behavior

Got a null pointer exception when my device detects the beacon.

```
java.lang.RuntimeException: An error occurred while executing doInBackground()
                                                                   at android.os.AsyncTask$3.done(AsyncTask.java:309)
                                                                   at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
                                                                   at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
                                                                   at java.util.concurrent.FutureTask.run(FutureTask.java:242)
                                                                   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                   at java.lang.Thread.run(Thread.java:818)
                                                                Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'void org.altbeacon.beacon.service.RssiFilter.addMeasurement(java.lang.Integer)' on a null object reference
                                                                   at org.altbeacon.beacon.service.RangedBeacon.addMeasurement(RangedBeacon.java:72)
                                                                   at org.altbeacon.beacon.service.RangedBeacon.updateBeacon(RangedBeacon.java:39)
                                                                   at org.altbeacon.beacon.service.RangedBeacon.<init>(RangedBeacon.java:34)
                                                                   at org.altbeacon.beacon.service.RangeState.addBeacon(RangeState.java:56)
                                                                   at org.altbeacon.beacon.service.BeaconService.processBeaconFromScan(BeaconService.java:439)
                                                                   at org.altbeacon.beacon.service.BeaconService.access$500(BeaconService.java:78)
                                                                   at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:484)
                                                                   at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:459)
                                                                   at android.os.AsyncTask$2.call(AsyncTask.java:295)
                                                                   at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                                   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
                                                                   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
                                                                   at java.lang.Thread.run(Thread.java:818) 
```
### Steps to reproduce this behavior

Adding Eddystone parser 
```java
@Override
    public void onResume() {
        super.onResume();
        beaconManager = BeaconManager.getInstanceForApplication(this.getApplicationContext());
        // Detect the URL frame:
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20v"));
        beaconManager.bind(this);
    }
```

Setting a region to get all beacons
```java
public void onBeaconServiceConnect() {
        Region region = new Region("all-beacons-region", null, null, null);
        try {
            beaconManager.startRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        beaconManager.addRangeNotifier(this);
    }
```

### Mobile device model and OS version

LG G4 - Android 6.0

### Android Beacon Library version

Last version with gradle
compile 'org.altbeacon:android-beacon-library:2+'

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2016-12-10:
Thanks for this report.    I don't think this is caused by using Eddystone.  It seems the error reported in this line:

 `E/RangedBeacon: Could not construct RssiFilterImplClass org.altbeacon.beacon.service.RunningAverageRssiFilter`.  

Is caused by a failure to call a constructor using reflection in this code from RangedBeacon.java:

        try {
            Constructor cons = BeaconManager.getRssiFilterImplClass().getConstructors()[0];
            filter = (RssiFilter)cons.newInstance();
        } catch (Exception e) {
            LogManager.e(TAG, "Could not construct RssiFilterImplClass %s", BeaconManager.getRssiFilterImplClass().getName());
        }

I can't think of why it would fail, and unfortunately the log line does not print out the Exception so we can't see why.

Can you reproduce this problem?  If so, I can build you a custom library binary that logs this exception and perhaps we can figure out why it is happening.  @Oupsla, please let me know if you are game for trying this.

Oupsla, 2016-12-12:
Hello @davidgyoung ,

Yes I'm down to help you for this !
Maybe it is important to mention that I use Estimote Beacon ?

Let me know for the custom library !

tikorth, 2017-04-28:
Hello, i post a full sample here above

https://github.com/tikorth/EddyStone-UID-TLM-sample

allliiii, 2017-06-08:
how can i find the measuring technique or formula to get distance(m) in iBeacon & Eddystone app.i am so curious to know about path loss propagation coefficient value used in this app to evaluate distance from beacon to mobile sensor. i request all of you to help me with useful suggestions and links to this particular issue as i got involved in trouble to predict my results for localization in indoor environments.i uploaded csv file to you can also paste all your suggestion to my email address alin33600@gmail.com

HadenHiles, 2024-10-20:
I'm experiencing the same error when trying to use the flutter_beacon package on the latest android of my S24 Ultra.

```collapsable
I/BluetoothAdapter(10889): BluetoothAdapter() : com.groupofseven.g7trailapp
E/RangedBeacon(10889): Could not construct RssiFilterImplClass org.altbeacon.beacon.service.RunningAverageRssiFilter
E/AndroidRuntime(10889): FATAL EXCEPTION: pool-20-thread-3
E/AndroidRuntime(10889): Process: com.groupofseven.g7trailapp, PID: 10889
E/AndroidRuntime(10889): java.lang.NullPointerException: Attempt to invoke interface method 'void org.altbeacon.beacon.service.RssiFilter.addMeasurement(java.lang.Integer)' on a null object reference
E/AndroidRuntime(10889): 	at org.altbeacon.beacon.service.RangedBeacon.addMeasurement(RangedBeacon.java:79)
E/AndroidRuntime(10889): 	at org.altbeacon.beacon.service.RangedBeacon.updateBeacon(RangedBeacon.java:39)
E/AndroidRuntime(10889): 	at org.altbeacon.beacon.service.RangedBeacon.<init>(RangedBeacon.java:29)
E/AndroidRuntime(10889): 	at org.altbeacon.beacon.service.RangeState.addBeacon(RangeState.java:61)
E/AndroidRuntime(10889): 	at org.altbeacon.beacon.service.ScanHelper.processBeaconFromScan(ScanHelper.java:372)
E/AndroidRuntime(10889): 	at org.altbeacon.beacon.service.ScanHelper.access$200(ScanHelper.java:61)
E/AndroidRuntime(10889): 	at org.altbeacon.beacon.service.ScanHelper$ScanProcessorRunnable.run(ScanHelper.java:455)
E/AndroidRuntime(10889): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
E/AndroidRuntime(10889): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
E/AndroidRuntime(10889): 	at java.lang.Thread.run(Thread.java:1012)
I/Process (10889): Sending signal. PID: 10889 SIG: 9
Lost connection to device.
```

hiru-vn, 2025-03-05:
I am facing the same issues with a faulty beacon that crashes the app whenever it is near on Flutter app. Can we try to catch the null field?

davidgyoung, 2025-03-15:
Moved to android beacon library issue here: https://github.com/AltBeacon/android-beacon-library/issues/1231

closed 2025-03-15 by davidgyoung