AltBeacon/android-beacon-library #464: Start monitoring without start an activity

Carmelo992, 2016-12-08:
I would like to start the monitoring without starting the main activity. It's possible? 
I saw the sample code, but as soon as he sees a beacon, the app is launched.
I would like to show a notification when a beacon is seen and launch the app when the user clicks on the notification or on its icon.

I tried to add to the monitorNotifier beaconManager, but does not work.

arpitjoshi08, 2016-12-08:
yes this is possible. put your code in background service class like this.
in your background service class:
`
public class BackgroundSrvce extends Service implements BeaconConsumer,MonitorNotifier {
    @Override
    public IBinder onBind(Intent intent) {
        return null;
    }

    @Override
    public void onCreate() {
       Appclass.bind(this);
        super.onCreate();

    }

    @Override
    public int onStartCommand(Intent intent, int flags, int startId) {
        try {
            Appclass.startMonitoringBeaconsInRegion(Appclass.region1);

        } catch (Exception e) {
        }
        return START_STICKY;
    }

    @Override
    public void onBeaconServiceConnect() {

        Appclass.beaconManager.setMonitorNotifier(this);

    }
    public void didEnterRegion(Region region) {
Log.d("monitoring", "Enter");        
    }

    @Override
    public void didExitRegion(Region region) {
Log.d("monitoring", "Exit");
    }

    @Override
    public void didDetermineStateForRegion(int i, Region region) {
Log.d("monitoring", "didDetermineStart");
    }
}`

and in your Application class:

`
public class Appclass extends Application implements BootstrapNotifier {
     private RegionBootstrap regionBootstrap;
    private BackgroundPowerSaver backgroundPowerSaver;
    private boolean haveDetectedBeaconsSinceBoot = false;
 
    public static Region region1;
    public static BeaconManager beaconManager;

    @Override
    public void onCreate() {

        super.onCreate();
        instances = this;
        beaconManager = org.altbeacon.beacon.BeaconManager.getInstanceForApplication(this);

        beaconManager.getBeaconParsers().add(new BeaconParser()
                .setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.setForegroundScanPeriod(1100l);
        beaconManager.setForegroundBetweenScanPeriod(0l);
        beaconManager.setAndroidLScanningDisabled(true);
        beaconManager.setBackgroundBetweenScanPeriod(01);
        beaconManager.setBackgroundScanPeriod(1100l);

        try {
            beaconManager.updateScanPeriods();
        } catch (Exception e) {
        }
        Log.d("", "setting up background monitoring for beacons and power saving");
        // wake up the app when a beacon is seen
        region1 = new Region("backgroundRegion",
                null, null, null);
        regionBootstrap = new RegionBootstrap(this, region1);
       backgroundPowerSaver = new BackgroundPowerSaver(this);
        
    }
  @Override
    public void didEnterRegion(Region region) {
            Intent i = new Intent(getApplicationContext(), BackgroundSrvce.class);
                    startService(i);
      }   
    }
    @Override
    public void didExitRegion(Region region) {
    
    }

    @Override
    public void didDetermineStateForRegion(int j, Region region) {
                    Intent k = new Intent(getApplicationContext(),   BackgroundSrvce.class);
              startService(k);
            }
       
}
`
check and let me know if is it working properly or not  .

davidgyoung, 2016-12-08:
@Carmelo992, in addition to the solution from @arpitjoshi08 above, the reference application will do exactly what you want if you simply disable the automatic app launching in the [BeaconReferenceApplication class.](https://github.com/AltBeacon/android-beacon-library-reference/blob/master/app/src/main/java/org/altbeacon/beaconreference/BeaconReferenceApplication.java)

Just change this:

        if (!haveDetectedBeaconsSinceBoot) {
            Log.d(TAG, "auto launching MainActivity");

            // The very first time since boot that we detect an beacon, we launch the
            // MainActivity
            Intent intent = new Intent(this, MonitoringActivity.class);
            intent.setFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
            // Important:  make sure to add android:launchMode="singleInstance" in the manifest
            // to keep multiple copies of this activity from getting created if the user has
            // already manually launched the app.
            this.startActivity(intent);
            haveDetectedBeaconsSinceBoot = true;
        } else {
            if (monitoringActivity != null) {
                // If the Monitoring Activity is visible, we log info about the beacons we have
                // seen on its display
                monitoringActivity.logToDisplay("I see a beacon again" );
            } else {
                // If we have already seen beacons before, but the monitoring activity is not in
                // the foreground, we send a notification to the user on subsequent detections.
                Log.d(TAG, "Sending notification.");
                sendNotification();
            }
        }

To this:

            Log.d(TAG, "Sending notification.");
            sendNotification();

If you need more help with your project, please open a new question on StackOverflow.com, as this forum is reserved for feature requests and bug reports with the library itself.

closed 2016-12-08 by davidgyoung

Carmelo992, 2016-12-11:
@arpitjoshi08 I believed that the MonitorNotifier had to be declared within an application or an activity.

@davidgyoung I apologize for having the wrong place to ask for help for my project.

I apologize for the delay but I was pretty busy these days. However thanks for the help