AltBeacon/android-beacon-library #467: Not detect beacon in lock screen (sleep mode)

polaris0227, 2016-12-21:
### Expected behavior

### Actual behavior

### Steps to reproduce this behavior

### Mobile device model and OS version

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

polaris0227, 2016-12-21:
Hi, all.
I'm developing an android app using beacon library.
App doesn't detect beacon after 30 seconds I click power button to lock screen.
Seems it doesn't detect beacon in sleep mode.
Also when I switch to other apps, it stops detecting immediately.
If any solution, please share with me.
Thanks.

davidgyoung, 2016-12-22:
@polaris0227, can reproduce this issue with the [reference application here](https://github.com/AltBeacon/android-beacon-library-reference)?

If you can reproduce this with the reference app, please attach info about the device model  where you see this.  Also, if possible, please modify the reference app to enable debug logging with `beaconManager.setDebug(true);` and attach a LogCat file during the time of the test.

If you only see this problem with a custom app, the issue may be with the way the library is used with the app.  In this case, please open a new question on StackOverflow.com and paste a code excerpt where you set up background beacon detection with the library.

zabi90, 2016-12-22:
Hi,
I also saw this issue. When i put my mobile in sleep mode  it was working fine ,detecting beacon through didRangeBeaconsInRegion callback  but as soon as i unlock mobile it stopped working. I have put the library code inside my own background services.

polaris0227, 2016-12-22:
@zabi90.
Hi.
Thank you for your kind reply.
Can you share with me your code of background service?
Many thanks.

davidgyoung, 2016-12-22:
To be clear, this library is designed to range and monitor beacons continually even in the background, if properly used.  The reference app demonstrates this.

arpitjoshi08, 2017-01-07:
public class Appclass extends Application implements BootstrapNotifier {
    private RegionBootstrap regionBootstrap;
    private BackgroundPowerSaver backgroundPowerSaver;
    public static Region region1;
       public static BeaconManager beaconManager;

    @Override
    public void onCreate() {

        super.onCreate();
      
        beaconManager = org.altbeacon.beacon.BeaconManager.getInstanceForApplication(this);
        beaconManager.getBeaconParsers().add(new BeaconParser()
                .setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.setForegroundScanPeriod(1100l);
        beaconManager.setForegroundBetweenScanPeriod(0l);
        beaconManager.setAndroidLScanningDisabled(true);
        beaconManager.setBackgroundBetweenScanPeriod(01);
        beaconManager.setBackgroundScanPeriod(1100l);

        try {
            beaconManager.updateScanPeriods();
        } catch (Exception e) {
        }
        Log.d("", "setting up background monitoring for beacons and power saving");
        // wake up the app when a beacon is seen
        region1 = new Region("backgroundRegion",
                null, null, null);
        regionBootstrap = new RegionBootstrap(this, region1);
        backgroundPowerSaver = new BackgroundPowerSaver(this);
  
    }

    @Override
    public void didEnterRegion(Region region) {
        Log.d("log", "AppClass"+"Enter");
        Log.d("log", "AppClass"+region.getId2() + " " + region.getId3());
        try {
          Intent i = new   Intent(getApplicationContext(), BeaconService.class);
                  startService(i);
            }

        } catch (Exception e) {

        }

    }

    @Override
    public void didExitRegion(Region region) {
        Log.d("log", "AppClass"+"exit");
        try {
                        Intent k = new Intent(getApplicationContext(), BeaconService.class);
                k.putExtra("didDetermineStateForRegion", false);
                k.putExtra("enterorexit", "exit");
                startService(k);
            }
        }
        catch (Exception e) {
        }
    }

    @Override
    public void didDetermineStateForRegion(int j, Region region) {
            try {
                   Intent k = new Intent(getApplicationContext(),         BeaconService.class);
                k.putExtra("didDetermineStateForRegion", true);
                k.putExtra("enterorexit", "");
                startService(k);
            }
        }
        catch (Exception e) {
        }
    }
}

// My BeaconServiceCode:

public class BeaconService extends Service implements BeaconConsumer, RangeNotifier {
    BeaconManager beaconManager;
  
    @Override
    public void onCreate() {
        super.onCreate();
   
            Appclass.beaconManager.bind(this);
        } catch (Exception e) {

        }
    }

    @Override
    public int onStartCommand(Intent intent, int flags, int startId) {
      //here you can check intent data and start or stop service according to use 
     Appclass.beaconManager.startRangingBeaconsInRegion(Appclass.region1);
                  
         
        return START_STICKY;
    }

    @Override
    public void onBeaconServiceConnect() {
        try {
            
            Appclass.beaconManager.setRangeNotifier(this);
        } catch (Exception e) {

        }
  
    }

@Override
    public void didRangeBeaconsInRegion(Collection<Beacon> collection, Region region) {
        String valuesAll = "";
        if (collection.size() > 0 && collection != null) {
            try {
                
check beacon here 

            } catch (Exception e) {
            }
        }
    }

    @Override
    public IBinder onBind(Intent intent) {
        return null;
    }

    @Override
    public void onTaskRemoved(Intent rootIntent) {
        super.onTaskRemoved(rootIntent);

        Intent restartServiceIntent = new Intent(getApplicationContext(), this.getClass());
        restartServiceIntent.setAction("");
         PendingIntent restartServicePendingIntent = PendingIntent.getService(getApplicationContext(), 1, restartServiceIntent, PendingIntent.FLAG_ONE_SHOT);
        AlarmManager alarmService = (AlarmManager) getApplicationContext().getSystemService(Context.ALARM_SERVICE);
        alarmService.set(
                AlarmManager.ELAPSED_REALTIME,
                SystemClock.elapsedRealtime() + 1000,
                restartServicePendingIntent);
    }

   
}

piotrek1543, 2017-01-07:
Well, it might be also a problem with Android system overlay provided by device manufacturer. For example, Huawei Protected Apps on Android Marshmallow based devices (I've already have P8) can kill an app during sleep mode if it takes too much resources.  The similar thing comes with Sony Xperia devices.

piotrek1543, 2017-01-07:
@arpitjoshi08 could you edit your answer - your code is not formatted pretty well.

dreimanis, 2017-01-31:
@piotrek1543 I have Huawei P8 also, for it to scan in background you have to enable it as Protected. But it also kills my app after few minutes. It doesn't really kill it, just pauses it. When you start charging it will start to scan again. I have also LG G4, Nexus 5, Galaxy J5 an Galaxy S6 everything works fine.

pankaj-syscraft, 2017-05-20:
I am also getting this issue "Not detect beacon in lock screen (sleep mode)". please help me !

didRangeBeaconsInRegion--->18:47:34 (Scan time)
 didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:47:49 (Scan time)
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:48:04 (Scan time)
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:48:19 (Scan time)
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:48:34 (Scan time)
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:48:49 (Scan time)
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:49:04 (Scan time)
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:49:19 (Scan time)
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->18:49:28 (Scan time)
didRangeBeaconsInRegion--->1
didRangeBeaconsInRegion--->18:49:38 (Scan time)
didRangeBeaconsInRegion--->0

davidgyoung, 2017-05-20:
@pankaj-syscraft, can you please test with this release:

 https://github.com/AltBeacon/android-beacon-library/releases/tag/2.11-beta1

pankaj-syscraft, 2017-05-22:
@davidgyoung 
Unable to complete this process in your given link to use 2.11-beta1 version 
getting this error "Error:Failed to resolve: android-beacon-library:2.11-beta1:"

To use this release, you must download the aar file below, and configure it with your project like so:

Edit your innermost build.gradle file to add the beta library AAR as a dependency like so:
      dependencies {
        compile 'android-beacon-library:2.11-beta1@aar'
      }
Create a /libs folder next to the build.gradle above, then download and copy the the android-beacon-library-2.11-beta1.aar file below into this folder.

Configure your app's outermost build.gradle File to reference the /libs directory from above in a flatDir declaration:

      allprojects {
          repositories {
              jcenter()
              flatDir {
                  dirs 'libs'
              }
          }
      }

arpitjoshi08, 2017-05-22:
hey pankaj-syscraft please check on this link i think it will be helpful for you  [http://stackoverflow.com/questions/29082634/how-to-import-the-altbeacon-library-inside-my-android-studio-1-0-empty-project](AltBeacon-implment)

arpitjoshi08, 2017-05-22:
hey pankaj-syscraft check this link i think it would be helpful to resolved this issue :-http://stackoverflow.com/questions/29082634/how-to-import-the-altbeacon-library-inside-my-android-studio-1-0-empty-project

davidgyoung, 2017-05-22:
You can see a working example of how to set this up in the reference project here:

https://github.com/AltBeacon/android-beacon-library-reference/tree/multi-process-support?files=1

pankaj-syscraft, 2017-05-23:
didRangeBeaconsInRegion--->19:37:53
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:38:03
didRangeBeaconsInRegion--->0
 didRangeBeaconsInRegion--->19:38:13
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:38:23
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:38:33
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:38:43
didRangeBeaconsInRegion--->1
didRangeBeaconsInRegion--->19:38:53
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:39:03
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:39:13
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:39:23
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:39:33
didRangeBeaconsInRegion--->0
didRangeBeaconsInRegion--->19:39:43
didRangeBeaconsInRegion--->0

davidgyoung, 2017-05-23:
@pankaj-syscraft, can you please try to reproduce this using the reference app?  You can clone it with a single command, then open it in Android Studio to build and run.

`git clone -b multi-process-support git@github.com:AltBeacon/android-beacon-library-reference`

Using a standard simple reference app to troubleshoot problems like this is critical to eliminate complexities added by custom apps.

allnash, 2017-10-09:
What is the status on this issue?

Liel208, 2020-07-07:
I had the same issue even when running the beacon library as a###  foreground service. 
Eventually I changed the app permission of '**battery saving'** to "**no restrictions**" and the detection worked when screen was locked.
**BUT** it stopped after an ~hour. I tried it several times with the same result.
I think a fix for this is to run the beacon ranging activity as a foreground service as well, in order to keep it running for a long period of time. - still didn't test this. will update

davidgyoung, 2020-07-07:
@Liel208, you are most certainly experiencing a proprietary model-specific customization of Android.  Such battery restrictions are not part of AOSP.  Please let us know your manufacturer, model and OS version as your solution is most certainly specific to that.

Liel208, 2020-07-07:
Thank you for your response
These are the specs:
[image]

So if run the same app on a different device (I am guessing not xiomi) it should run the foreground service for more than one hour?

davidgyoung, 2020-07-07:
Generally speaking Xioami/Redmi devices have lots of secret closed-source
customizations to the operating system that affect backgrounded apps.
Because the rules are not published, it is impossible to comply with them
without reverse-engineering what the rules are.  See here:
https://dontkillmyapp.com/  That list is about backgrounded apps in
general, not bluetooth scanning, so it rates other manufacturers in terms
of general background operations.   Xioami/Redmi is particularly known to
block bluetooth scans in some undefined cases.