AltBeacon/android-beacon-library #468: detection issue

shanu776, 2016-12-29:
### Expected behavior  : working on mycellphone

### Actual behavior : only working on lollipop

### Steps to reproduce this behavior

### Mobile device model and OS version htc 820 marshmallow

### Android Beacon Library version 2.9

issue :- beacon library not working on android 6,my cellphone note detect beacon but same app when i install on a lollipop device its working.

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

arpitjoshi08, 2017-01-07:
could you please  location ON  of your device  with bluetooth to check working or not actually in some devices i also got issue in 6.0 then i switched on location with bluetooth then its working fine.

piotrek1543, 2017-01-07:
@shanu776 First of all check your project's `build.gradle` file if your `targetSdkVersion` is higher than `22`.
If yes you would need to define runtime permissions for Marshmallow like in link below:
https://developer.android.com/training/permissions/requesting.html

On Marshamalllow in that case you need to check if `ACCESS_COARSE_LOCATION` permissions are granted.

Here's an example: https://github.com/vitas/beaconloc/blob/8207969c5ccbd91119aa8963657d789ce55887e1/app/src/main/java/com/samebits/beacon/locator/ui/activity/MainNavigationActivity.java

If you don't want to do it downgrade your `targetSdkVersion` to 22 or lower.

Hope it will help

tylerbrockett, 2018-03-26:
I spent a lot of time debugging this because the documentation states it's only necessary for background monitoring, not foreground monitoring. It seems as though it's the case for foreground mode as well (beaconManager.getBackgroundMode() returns "false")

Can we add code to the library to throw an Exception if we're trying to range beacons when we do not have either the ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permissions?

davidgyoung, 2018-03-26:
@tylerbrockett, if you can link to the documentation page that says this, I'll fix it.  I see that [this page](http://altbeacon.github.io/android-beacon-library/requesting_permission.html) does mention both foreground and background.

Throwing an exception is problematic because it breaks the API for existing apps.  We could however put an error level message in LogCat if you think that would be helpful.  I am open to other ideas to reduce frustration.

tylerbrockett, 2018-03-28:
@davidgyoung you're absolutely right, it is already in the documentation, I must have misread that. It also looks like that documentation states there should be an entry in the log if the proper permissions aren't granted. I didn't see that in my LogCat, but I'll have to go back through and double check.

davidgyoung, 2018-03-28:
Right, that error message is output by the OS, which you may not see if you are filtering messages for the app.

tylerbrockett, 2018-03-28:
Ahh, okay, makes sense. Thanks for your help!