AltBeacon/android-beacon-library #469: NPE in BackgroundPowerSaver onActivityPaused and onActivityResume

jeffkelsey, 2017-01-11:
### Expected behavior
No crash reports

### Actual behavior
Top 2 crashers in our app based on Fabric data is:

#1 
BackgroundPowerSaver line 66
org.altbeacon.beacon.powersave.BackgroundPowerSaver.onActivityResumed

Fatal Exception: java.lang.RuntimeException: Unable to resume activity {com.ticketmaster.mobile.android.na/com.ticketmaster.mobile.android.library.ui.activity.EventDetailsAndFindTicketsCombinedActivity}: java.lang.NullPointerException
       at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2958)
       at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2987)
       at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2380)
       at android.app.ActivityThread.access$900(ActivityThread.java:164)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1268)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:157)
       at android.app.ActivityThread.main(ActivityThread.java:5377)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:515)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1265)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1081)
       at dalvik.system.NativeStart.main(NativeStart.java)

Crash #2 
BackgroundPowerSaver line 76
org.altbeacon.beacon.powersave.BackgroundPowerSaver.onActivityPaused

Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'void org.altbeacon.beacon.BeaconManager.setBackgroundMode(boolean)' on a null object reference
       at org.altbeacon.beacon.powersave.BackgroundPowerSaver.onActivityPaused(SourceFile:76)
       at android.app.Application.dispatchActivityPaused(Application.java:247)
       at android.app.Activity.onPause(Activity.java:1652)
       at android.support.v4.app.FragmentActivity.onPause(SourceFile:440)
       at com.ticketmaster.mobile.android.library.activity.DrawerActivity.onPause(SourceFile:223)
       at com.ticketmaster.mobile.android.library.activity.MainActivity.onPause(SourceFile:218)
       at android.app.Activity.performPause(Activity.java:7033)
       at android.app.Instrumentation.callActivityOnPause(Instrumentation.java:1339)
       at android.app.ActivityThread.performPauseActivity(ActivityThread.java:4572)
       at android.app.ActivityThread.performPauseActivity(ActivityThread.java:4545)
       at android.app.ActivityThread.handlePauseActivity(ActivityThread.java:4520)
       at android.app.ActivityThread.access$1300(ActivityThread.java:221)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1805)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:158)

### Mobile device model and OS version

### Android Beacon Library version
2.5.1 used within the ExactTarget 4.7.1 sdk 
(example in https://github.com/salesforce-marketingcloud/LearningAppAndroid/blob/master/app/build.gradle)

davidgyoung, 2017-01-11:
Thanks for this report.  I am quite surprised to see it.

Reviewing the code,  it seems for this crash to happen a previously initialized variable must have mysteriously become null, something I believe should be impossible.

Do you see this with any specific devices or OS versions?

jeffkelsey, 2017-01-24:
Seems to be across most devices and versions:
of the 1209 CRASHES 1210 USERS:

Devices
samsung81%
LGE7%
motorola4%
Other…8%
  
OS:
Operating Systems
6 84%
5 7%
4 7%
Other 2%

jmvines, 2017-02-22:
Hi @davidgyoung,

I have the same crash as jeffkelsey when I use this class in some devices.

Did you had the chance to review this?

Best,

Chema

davidgyoung, 2017-02-23:
Since I cannot reproduce this problem or even explain how it could happen, the best I can do is protect against a crash by adding a null check in the next version.  I'll put this in the next release.

jmvines, 2017-02-23:
Thanks for your effort!!

Best,

Chema.

jeffkelsey, 2017-02-27:
Is there a release that has this fix included or a timeline for next release?  This is still the top crasher in our production application.

davidgyoung, 2017-02-28:
@jeffkelsey how would you feel about putting a prerelease with this fix into your app?  If so I can make you a special build with the latest release binary and only this change for you to deploy and see if these crashes go away.

It is a low risk change. But I am reluctant to put this in a full release since I have no way to reproduce or test that it helps.

jeffkelsey, 2017-02-28:
That works for me

jeffkelsey, 2017-03-01:
How can you send me the pre-release?  We have a new build getting ready to RC test on Monday and I would love to get this in before then

davidgyoung, 2017-03-02:
@jeffkelsey, I have prepared a release with instructions here:  https://github.com/AltBeacon/android-beacon-library/releases/tag/fix-power-saver-crash

Please give this a try and let me know how it affects these crash reports.

cupakromer, 2017-04-09:
This may be a symptom of a problem with singleton object creation fixed in #494.