AltBeacon/android-beacon-library #470: BeaconParser  java.lang.ArrayIndexOutOfBoundsException: length=62; index=-1

cgiMalaga, 2017-01-11:
### Expected behavior
Detect beacons with my own application. 
Application is working well except with de Android 6.0 OS version
### Actual behavior
App is killed when a beacon is detected (altBeacon, iBeacon,...)
### Steps to reproduce this behavior
1.- Turn off all beacons.
1.- Launch an app based on AltBeaconLibrary (RadBeacon, LocateBeacon,...)
2.- Turn on at least one beacon.
3.- App is killed throwing a ArrayIndexOutOfBoundsException in BeaconParser class , line 849.

This behavior is the same in all applications based on altbeaconlibrary downloaded from google play as well as in my own application

StackTrace below:

java.lang.RuntimeException: An error occurred while executing doInBackground()
                                                                                    at android.os.AsyncTask$3.done(AsyncTask.java:309)
                                                                                    at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
                                                                                    at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
                                                                                    at java.util.concurrent.FutureTask.run(FutureTask.java:242)
                                                                                    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                                    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                                    at java.lang.Thread.run(Thread.java:818)
                                                                                 Caused by: java.lang.ArrayIndexOutOfBoundsException: length=62; index=-1
                                                                                    at org.altbeacon.beacon.BeaconParser.byteArrayToFormattedString(BeaconParser.java:849)
                                                                                    at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:587)
                                                                                    at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:417)
                                                                                    at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:474)
                                                                                    at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:459)
                                                                                    at android.os.AsyncTask$2.call(AsyncTask.java:295)

### Mobile device model and OS version
 .- Device : Bq Aquaris A4.5 (Android One)
 .- Android Version : 6.0 marshmallow

### Android Beacon Library version

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-01-12:
Can you please modify your app to set `beaconManager.setDebug(true);` then re-produce the crash, and capture the last entry in LogCat before the crash that looks something similar to this:

```
    04-20 17:10:08.765  24817-25200/com.company.MyApp D/BeaconParser﹕This is a recognized beacon advertisement -- BEAC seen
    04-20 17:10:08.766  24817-25200/com.company.MyApp D/BeaconParser﹕Bytes are: ABCD1234
 --- crash here  ---
```

cgiMalaga, 2017-01-12:
Hi David, here you are the LogCat.
The library version used is 2.9.2, but with any other older version the behavior is the same.
For any other OS version the app work fine.

01-12 09:47:26.812 4033-4150 D/BeaconParser: Processing pdu type FF: 02011a1aff4c0002152f52aa80f5e549e0a4735432e9f5badb0000a966c50000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
01-12 09:47:26.813 4033-4150 D/BeaconParser: This is a recognized beacon advertisement -- 4c 00 02 15 seen
01-12 09:47:26.813 4033-4150D/BeaconParser: Bytes are: 02011a1aff4c0002152f52aa80f5e549e0a4735432e9f5badb0000a966c50000000000000000000000000000000000000000000000000000000000000000
                                                                            
                                                                            --------- beginning of crash

cgiMalaga, 2017-01-12:
Hi David, here you are the LogCat.
The library version used is 2.9.2, but with any other older version the behavior is the same.
For any other OS version the app work fine.

```
01-12 09:47:26.812 4033-4150 D/BeaconParser: Processing pdu type FF: 02011a1aff4c0002152f52aa80f5e549e0a4735432e9f5badb0000a966c50000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
01-12 09:47:26.813 4033-4150 D/BeaconParser: This is a recognized beacon advertisement -- 4c 00 02 15 seen
01-12 09:47:26.813 4033-4150D/BeaconParser: Bytes are: 02011a1aff4c0002152f52aa80f5e549e0a4735432e9f5badb0000a966c50000000000000000000000000000000000000000000000000000000000000000
--------- beginning of crash
```

davidgyoung, 2017-01-12:
The packet bytes look fine.   Can you please tell me the BeaconParser string you are using?

cgiMalaga, 2017-01-12:
I have been trying with several strings beacon parser with the same issue. Even if I set up the beacon (Radius DOT beacon) as AltBeacon the behavior do not change, fail too

```
beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24"));
```
```
beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
```

As I said before is strange because the applicaction only work wrong in this OS , so I think the source are rigth.

Maybe could be the chipset Mediatek integrated into this device, I do not know...

davidgyoung, 2017-01-12:
Yes, I suspected a chipset problem, too.  But if that is the case, I would expect to have seen a corrupted bytestream.   The bytestream you posted above looks good, and parses just fine.  

02011a1aff4c0002152f52aa80f5e549e0a4735432e9f5badb0000a966c50000000000000000000000000000000000000000000000000000000000000000

Are you sure this bytestream causes the crash?  I cannot reproduce that.

cgiMalaga, 2017-01-12:
Yes David, I am sure, but not only this bytestream, any other bytestream cause the crash too.
As soon as any beacon is detected, the app crash, but not only my app. 'RadBeacon' app or 'Locate Beacon' app crash too.

I would like to try with other device with this chipset to confirm our suspected, but it is impossible for me.

But I can confirm to you the application work fine in other devices with mediatek chipset, but not the same chipset

cgiMalaga, 2017-01-12:
Some more information about it.

I have been testing using 
```
mBluetoothAdapter.startLeScan(mLeScanCallback) 

```
 and iBeacon is detected fine.

So I am not sure the problem is the chipset

Here you are the callback code and traces

```
// Device scan callback.
    private BluetoothAdapter.LeScanCallback mLeScanCallback = new BluetoothAdapter.LeScanCallback() {
        @Override
        public void onLeScan(BluetoothDevice device, int rssi, byte[] scanRecord) {

            int startByte = 2;
            boolean patternFound = false;
            while (startByte <= 5) {
                if (    ((int) scanRecord[startByte + 2] & 0xff) == 0x02 && //Identifies an iBeacon
                        ((int) scanRecord[startByte + 3] & 0xff) == 0x15) { //Identifies correct data length
                    patternFound = true;
                    break;
                }
                startByte++;
            }

            if (patternFound) {
                //Convert to hex String
                byte[] uuidBytes = new byte[16];
                System.arraycopy(scanRecord, startByte+4, uuidBytes, 0, 16);
                String hexString = bytesToHex(uuidBytes);

                //Here is your UUID
                String uuid =  hexString.substring(0,8) + "-" +
                        hexString.substring(8,12) + "-" +
                        hexString.substring(12,16) + "-" +
                        hexString.substring(16,20) + "-" +
                        hexString.substring(20,32);

                //Here is your Major value
                int major = (scanRecord[startByte+20] & 0xff) * 0x100 + (scanRecord[startByte+21] & 0xff);

                //Here is your Minor value
                int minor = (scanRecord[startByte+22] & 0xff) * 0x100 + (scanRecord[startByte+23] & 0xff);
                Log.d(TAG,"Device :: " + device + " UUID :: " + uuid + "  Major :: " + major + "  Minor :: " + minor);
            }
        }
    };
```

```
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
Device :: E0:55:3D:B7:56:F0 UUID :: 2F52AA80-F5E5-49E0-A473-5432E9F5BADB  Major :: 0  Minor :: 43366
```

davidgyoung, 2017-01-12:
Understand that the only difference in what this library does to parse the beacon and what is shown above (based what it looks like on an older version of the library's parsing algorithm) is a more robust and technically solid parser.  I suspect what is happening is that a normal scan result byte array gets parsed just fine, but garbled ones do not.  (The less robust parser ironoically may be more tolerant of this garbling.)

The normal byte array you captured looks like this:
  02011a1aff4c0002152f52aa80f5e549e0a4735432e9f5badb0000a966c50000000000000000000000000000000000000000000000000000000000000000

But I suspect that at some times an incorrect scan result comes back with some of the leading bytes missing, or perhaps one of the length headers set incorrectly in those leading bytes.  Can you please try again to do a capture of the crash and make sure that this is what the byte stream looks like?

davidgyoung, 2017-03-11:
Closing this due to inactivity.  If you can capture a byte stream per the instructions in the last couple of comments, please reopen this issue and attach that info.

closed 2017-03-11 by davidgyoung