AltBeacon/android-beacon-library #471: Time of flight algorithm to get distance

am2222, 2017-01-19:
Hi,
I have faced a problem using current distance algorithm which is highly unstable and is not possible to use as location finding. anyway I have found some new algorithms like Time of flight. Does any one have tried to implement those? Is it possible to add those algorithm in altbeacon?

davidgyoung, 2017-01-19:
Hi, @am2222, I am always open to new ideas for making distance estimates more stable and accurate.  

While I am not familiar with the Time of flight algorithm, I have experimented with others such as a path loss approach, which I have never been able to make work better than the existing algorithm.  You can see that work I did in the pull request here: #251

Looking at that pull-request might be useful for you to see how you could hook in a new algorithm like you describe, and you could use this as a reference for making your own pull request for a Time of flight algorithm if you want to try to implement it.

am2222, 2017-01-19:
@davidgyoung  thanks for your reference, Now I am looking for those algorithms which is used in ToF and I hope I can make some progress. I'll share my findings here

TheWizz, 2017-03-12:
I'm curious as to what you envisage measuring the ToF turn-around relative to? Beacons communication is generally one way. Even if there is some bidirectional request/response style command, I seriously doubt the response time (not mentioning the OS stack propagation time) is predictable enough to use for ToF-style measurements.

am2222, 2017-03-13:
@TheWizz  based on my research it is not possible to achieve it using beacons, But I have seen some companies such as estimote and pixes claim that they are using ToF. How it is possible?

heypiotr, 2017-03-24:
To be strict, ToF is not possible using 2.4 GHz Bluetooth. We (Estimote) and Pixie use UWB (https://en.wikipedia.org/wiki/Ultra-wideband) for that. The thing is, UWB is not in smartphones, so this method is currently restricted to measuring distances between beacons with UWB themselves.

am2222, 2017-03-30:
@heypiotr can you please explain a bit more about how you use them? I mean Your beacons can communicate each other but how they send their distance to mobile phone?Is it even possible? I mean The only problem here is the distance between beacons and the device

heypiotr, 2017-04-13:
That's exactly my point (: Just wanted to confirm your:

>based on my research it is not possible to achieve it using beacons

As in, to my knowledge this is not currently possible to do between the beacons and the smartphone. Pixie (AFAIK) and Estimote (I do know) only do beacon-to-beacon ToF.

am2222, 2017-04-20:
@heypiotr yes they are doing it beacon-to-beacon ToF and I have no idea how can we increase accuracy, Is it possible to use them base on Kalman filter and thing like that? Do you have any ideas about doing it using statistical methods?

lakshmanpasala, 2017-04-24:
@am2222 Kalman Filter helped only marginally. The approach I took was to have already mapped the beacons on a map and using Kalman Filter and other statistical methods to find the "relatively closest" beacon. My use-case is indoor-navigation and surprisingly this approach works quite well.

am2222, 2017-04-24:
@lakshmanpasala  Hi, Really? Can you please share some results or other stuffs if it is possible? I am working on indoor mapping same as you but I have faced applying Kalman Filter, Why you have focused on finding closest beacon instead on the result of 3 point positioning?

lakshmanpasala, 2017-04-24:
The RSSI values of Battery powered BLEs is not consistent, especially indoors. To add to that, each mobile device keeps throwing different values of RSSI. For a fixed position of the beacon, my Samsung and OnePlus phones gave different values even though both were exactly the same distance away from the device.

Coming to the trilateration, I did implement that too but since faced different set issues with that.
1.) The map needs to be perfectly to scale. Which is a bit difficult for large venues. Might be easier for smaller venues.
2.) The distances that you do get even after the kalman filter and other statistical models will still be not accurate distances even though they are relatively correct. What the filter and models do is at least guarantee that they are relatively correct by taking care of the environment noise.

My conclusion was that BLE beacons are not the way to go for indoor navigation. If you still want to proceed then the mapping the beacons, kalman filter and some statistical model to normalize the noise is the way to go.
Interestingly powered WiFi beacons are better than BLE beacons for this use case.

lakshmanpasala, 2017-04-24:
You can check out my comment here https://github.com/AltBeacon/android-beacon-library/issues/498  for the Kalman Filter implementation

am2222, 2017-04-24:
@lakshmanpasala  thanks for your help,Can you please give me a bit more information about wifi beacons?