AltBeacon/android-beacon-library #472: StackOverflowError

tekinalper, 2017-02-09:
### Expected behavior
No crash
### Actual behavior
Fatal Exception: java.lang.StackOverflowError
       at android.os.Parcel.writeString(Parcel.java:516)
       at org.altbeacon.beacon.Region.writeToParcel(Region.java:262)
       at android.os.Parcel.writeParcelable(Parcel.java:1292)
       at org.altbeacon.beacon.service.RangingData.writeToParcel(RangingData.java:63)
       at android.os.Parcel.writeParcelable(Parcel.java:1292)
       at android.os.Parcel.writeValue(Parcel.java:1211)
       at android.os.Parcel.writeArrayMapInternal(Parcel.java:618)
       at android.os.Bundle.writeToParcel(Bundle.java:1692)
       at android.os.Parcel.writeBundle(Parcel.java:643)
       at android.content.Intent.writeToParcel(Intent.java:7251)
       at android.app.ActivityManagerProxy.startService(ActivityManagerNative.java:3024)
       at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1566)
       at android.app.ContextImpl.startService(ContextImpl.java:1548)
       at android.content.ContextWrapper.startService(ContextWrapper.java:494)
       at org.altbeacon.beacon.service.Callback.call(Callback.java:69)
       at org.altbeacon.beacon.service.BeaconService.processRangeData(BeaconService.java:402)
       at org.altbeacon.beacon.service.BeaconService.access$300(BeaconService.java:78)
       at org.altbeacon.beacon.service.BeaconService$1.onCycleEnd(BeaconService.java:364)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:251)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:243)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:215)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:284)
...
       at 
       at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:239)
       at android.os.Handler.handleCallback(Handler.java:733)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:136)
       at android.app.ActivityThread.main(ActivityThread.java:5118)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:515)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:789)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:605)
       at dalvik.system.NativeStart.main(NativeStart.java)

### Steps to reproduce this behavior
No idea
### Mobile device model and OS version 
(OS Versions: 4.3, 4.4.2, 4.4.3, 5.0, 5.1)
- Sony C2005, C200424 
- LGE LG-D682TR, LG-D686 
- Lenovo A1000

### Android Beacon Library version
2.9.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-02-21:
This stack trace looks normal:  it indicates a ranging event to your app from the library that will trigger the `didRangeBeaconsInRegion` callback.  

Is this the full stack trace?

Is it possible you can show the code you have inside the `didRangeBeaconsInRegion` callback in your app?  It is possible something in there is causing a recursive loop.

davidgyoung, 2017-02-21:
It looks like this may be caused by a loop here:

```
org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop (CycledLeScanner.java:243)
org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice (CycledLeScanner.java:215)
org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle (CycledLeScanner.java:284)
org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop (CycledLeScanner.java:243)
org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice (CycledLeScanner.java:215)
org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle (CycledLeScanner.java:284)
```

davidgyoung, 2017-02-21:
Is there anywhere in code where you set both the scan period and the between scan period to 0?  This is the only thing I can see that would cause this behavior.

tekinalper, 2017-02-21:
- No, I set them once(both are not 0) but calling the method that sets them when bluetooth/network connection status changes in 2 places. Is it a problem to set twice with the same periods?
- I have sent full stack trace reports to you via email. Nothing much more.
- I'm extending `BeaconConsumer` in an Android Service and restarting the Service in some cases. However, I do unbind `beaconManager` in `onDestroy` method and bind in `onStartCommand` again.

davidgyoung, 2017-02-22:
I need to investigate this a bit more.  What you describe should not cause any issues.  I need to figure out how this can happen.

Do you have the ability to reproduce this or do you just see it in crash reports?

If you could reproduce, I could generate a special debug build that logs extra messaging that might help track down the cause of the problem.

tekinalper, 2017-02-22:
Unfortunately I don't have any ability to reproduce it but I appreciate if you may send debug build so that we try to find some devices that crash and test in it.

tekinalper, 2017-02-22:
I don't know if it matters for your investigation but what I see on the reports is that the devices with the OS version 4 and 5 crash. Not 6 and 7. If we assume that I do something wrong, it had to cause problem for all OS versions.

tekinalper, 2017-02-24:
What I noticed is that although the device [LG-D682TR](http://www.lg.com/tr/akilli-telefon/lg-D682TR) has android version 4.1(upgradable to 4.4.2) and Bluetooth 3.0 on the specifications, I see the crashes for this device on the reports. Do you think there might be any possibility that the device pass over the BLE control as below when it's android version is 4.4.2 but Bluetooth version is 3.0? Otherwise, somehow the device is ready to use Bluetooth 4.0 after upgraded to 4.4.2.

`getPackageManager().hasSystemFeature(PackageManager.FEATURE_BLUETOOTH_LE)`

davidgyoung, 2017-02-24:
Interesting idea.  I will check the code to see if a scanning failure could somehow cause this.

tekinalper, 2017-02-27:
I have provided the device I mentioned. Here what I noticed:

- startService > onCreate() > stop service if device !supports BLE and also OS < 18
- onStartCommand() > stop service if device OS < 18 (Here BLE control is missing in my code)
- startScan() > stopScan() > startScan() > stopScan() .... (Here, there might be a problem in your lib as it started scanning without BLE support)

By the way, I have an Alarm which try to start my Service after a minute later if it stops.

I didn't encounter the crash yet but I guess BLE control in onStartCommand will be solution for some devices. Hovewer, I see other devices with OS > 18 and supports BLE on the reports.

davidgyoung, 2017-02-27:
Good find!  I'll roll a debug build and send a link for download when ready.

tekinalper, 2017-03-02:
Did you get a chance to prepare the build? :)

davidgyoung, 2017-03-02:
@tekinalper, please see the binary release here:  https://github.com/AltBeacon/android-beacon-library/pull/476

Sorry for the delay, it's been a busy week.

tekinalper, 2017-03-14:
@davidgyoung I have tested and still getting [crash](http://crashes.to/s/3be91f94b2b) as soon as scan starts. I have provided solution for the devices OS > 18 but NOT supports BLE by adding BLE control in onStartCommand(). For the rest of devices(OS > 18 and supports BLE), I will let you know if any crash still occurs after the SDK updates by the clients. Thank you.

davidgyoung, 2017-03-15:
@tekinalper, can you please share this solution?

tekinalper, 2017-03-15:
As I have Service class, I have checks like below in OnCreate method.
` if (!getPackageManager().hasSystemFeature(PackageManager.FEATURE_BLUETOOTH_LE)) {
            Log.e(TAG, "BLE is not supported on this device! Blesh service will be terminated.");
            stopSelf();
            return;
        }
        if (android.os.Build.VERSION.SDK_INT < android.os.Build.VERSION_CODES.JELLY_BEAN_MR2) {
            Log.e(TAG, "Blesh service is not supported below Android API Level 18!");
            stopSelf();
            return;
        }
 `

Although I had version control, first check was missing in my OnStartCommand method. Now I added it as well so Service stops,  not starting to scan and not causing crash. It is just quite different solution for the devices > 18 and not supports BLE in our SDK, not Altbeacon.

davidgyoung, 2021-08-06:
closed due to inactivity

closed 2021-08-06 by davidgyoung