AltBeacon/android-beacon-library #478: Scan beacon doesn't work - Eddystone

damienld22, 2017-03-08:
Hello,

I want to create an application to use Eddystone Beacon. 
I created a functional application (All work) but today, without reason, I can't see any beacon on my phone.
Since, even if I try to create an application from the example (Monitoring or Ranging), the problem is always here.

The application seem run but I don't receive any beacon.

To reproduce this behavior, I just create an application with the example from the website...

For my experimentation, I use a Huawei P9 Lite based on Android 6.0. (But I repeat, the application has already work many times). I use Android Beacon Library version 2.9.2.

Here my code from an example in the website : 

`
public class MainActivity extends Activity implements BeaconConsumer {

    protected static final String TAG = "MonitoringActivity";
    private BeaconManager beaconManager;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        beaconManager = BeaconManager.getInstanceForApplication(this);
        beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));
        beaconManager.bind(this);
        beaconManager.setDebug(true);
    }
    @Override
    protected void onDestroy() {
        super.onDestroy();
        beaconManager.unbind(this);
    }
    @Override
    public void onBeaconServiceConnect() {
        beaconManager.addMonitorNotifier(new MonitorNotifier() {
            @Override
            public void didEnterRegion(Region region) {
                Log.d(TAG, "I just saw an beacon for the first time!");
            }

            @Override
            public void didExitRegion(Region region) {
                Log.d(TAG, "I no longer see an beacon");
            }

            @Override
            public void didDetermineStateForRegion(int state, Region region) {
                Log.d(TAG, "I have just switched from seeing/not seeing beacons: "+state);
            }
        });

        try {
            beaconManager.startMonitoringBeaconsInRegion(new Region("myMonitoringUniqueId", null, null, null));
        } catch (RemoteException e) {    }
    }

}
`

And here the output for the debug : 

03-08 17:35:08.213 29131-29156/com.zetapush.damien.testbeacon D/BluetoothAdapter: STATE_ON
03-08 17:35:08.213 29131-29156/com.zetapush.damien.testbeacon D/BluetoothLeScanner: could not find callback wrapper
03-08 17:35:08.213 29131-29131/com.zetapush.damien.testbeacon D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
03-08 17:35:08.213 29131-29131/com.zetapush.damien.testbeacon D/CycledLeScanner: Scan started
03-08 17:35:09.215 29131-29131/com.zetapush.damien.testbeacon D/CycledLeScanner: Waiting to stop scan cycle for another 98 milliseconds
03-08 17:35:09.314 29131-29131/com.zetapush.damien.testbeacon D/CycledLeScanner: Done with scan cycle
03-08 17:35:09.316 29131-29131/com.zetapush.damien.testbeacon D/CycledLeScanner: stopping bluetooth le scan
03-08 17:35:09.316 29131-29131/com.zetapush.damien.testbeacon D/CycledLeScannerForLollipop: Stopping scan
03-08 17:35:09.318 29131-29156/com.zetapush.damien.testbeacon I/BluetoothLeScanner: stopScan is called
03-08 17:35:09.318 29131-29131/com.zetapush.damien.testbeacon D/CycledLeScanner: starting a new scan cycle
03-08 17:35:09.320 29131-29156/com.zetapush.damien.testbeacon D/BluetoothAdapter: STATE_ON

Thank you for your help.

davidgyoung, 2017-03-08:
@damienld22, this sounds like a device-specific problem.  For issues like this, please create a question on StackOverflow.com and tag it with ibeacon-android.  Happy to help there.  I'd suggest you reboot your phone and test again before trying this, and also try out an off-the-shelf scanning app like https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en, which is based on this same library.  These tests will help eliminate a programming issue or a  device-specific bluetooth stack problem.

closed 2017-03-08 by davidgyoung

damienld22, 2017-03-09:
@davidgyoung Thank you for your fast reply and your time !
The off-the-self scan app run correctly on my phone but even with a reboot, my application doesn't run.
I have the same problem when I test the application on an other device.

I will create a question on StackOverflow.com.