AltBeacon/android-beacon-library #480: Multiple PDU beacon advertisements

safakadir, 2017-03-16:
Hello,

I'm not sure what I want is a new feature or can be consider as a bug. Here is the thing:

### Background Info and Current Behavior
In BeaconParser class, bytes of advertisement data is splited into PDUs. Then in a loop, the first encountered pdu with the type GATT_SERVICE_UUID_PDU_TYPE (0x16) or MANUFACTURER_DATA_PDU_TYPE (0xff) is selected as pduToParse if its, loop breaks. Beacon payloads are present in the 0x16 frame for Eddystone, in the 0xff frame for IBeacon and others.

### Motivation
Sometimes, some specific hardware manufacturers may put both 0x16 and 0xff pdu types in the single advertisement of their beacons. I don't know why they choose this way instead of advertising another packet, but I observed this case. There was ibeacon payload in 0xff pdu and some small custom data was present in 0x16 pdu in a single advertisement packet. We guess that the custom data in our case, is battery level and some other manufacturer specific things since we are able to see these information in manufacturer's application. The second pdu is thrown away in such cases in current version of the library.

### Wished Behavior
Instead of throwing away other pdus after discovering one of 0xff or 0x16, the library should process both of them if present. The user may define different parsers for both, so that user can receive 2 beacon objects for different fields in an advertisement packet.

Can it be added to library?
I hope I could express what I want correctly.

davidgyoung, 2017-03-26:
Hi, @safakadir, I think this is a reasonable feature request.  I agree that in theory a beacon may be implemented this way, but the limited size of the advertising payload in Bluetooth 4.1 suggests to me it may not be practical to do this.

Do you know of specific beacon manufacturers that work this way?  If so, it would be helpful to see byte sequence for a real-world advertisement like this.

safakadir, 2017-03-29:
Hi @davidgyoung ,

Here is the byte sequence in hex that make me think of this feature. Manufacturer is Minew Technologies.
`0201061AFF4C000215E2C56DB5DFFB48D2B060D0F5A71096E00000000EC5020A000816F0FF150000000E11094D696E69426561636F6E5F30303031340000`

Let me split into PDUs:
`"020106" "1AFF4C000215E2C56DB5DFFB48D2B060D0F5A71096E000000013C5" "020A00" "0816F0FF3000000013" "11094D696E69426561636F6E5F3030303139" 0000`

`"020106" -> Type 01: Flags`

`"1AFF4C000215E2C56DB5DFFB48D2B060D0F5A71096E000000013C5" -> Type FF: iBeacon. Here minor is 0x0013 or 19 in decimal.`

`"020A00" -> Type 0A: Bluetooth Generic Access Profile says that it's Tx Power but it seems empty`

`"0816F0FF3000000013" -> Type 16: I couldn't figure what data is put by manufacturer here.`

`"11094D696E69426561636F6E5F3030303139" -> Type 09: Name, MiniBeacon_00019 in this example.`

I guess last two 00 bytes are padding.

davidgyoung, 2017-03-29:
Very helpful,  @safakadir.  Based on the above info it should be possible to modify the beacon parser to handle this layout.  The main trick is that a single packet will now potentially yield multiple beacon instances.  I will have to see how  difficult of a change this is.

phuongtran030818, 2019-06-27:
hi @safakadir @davidgyoung , is this problem solved already?  If it is, can you give details solution, please?

Thanks,

davidgyoung, 2019-06-27:
@phuongtran030818, this has not been implemented.  While possible, it is a significant amount of work to implement, perhaps 10-30 hours of coding and testing in my estimation.   Given the other priorities with the library, I would need a volunteer to do the work or a sponsor to move it forward.

phuongtran030818, 2019-06-27:
Hi @davidgyoung , thanks for your response. I think I found the way to fix this, now I can detect already, but will try to test with more devices to make sure it works with all device got from the supplier of my client. Btw still better if have this update in feature (some next version ^^).

Thanks,