AltBeacon/android-beacon-library #481: Background mode received "start scan is blocked"

desmond0321, 2017-03-21:
### Expected behavior
App go to background and still able to scan & detect beacon every 30sec.

### Actual behavior
After app go to background, it scan & detected few times the beacon, after that keep showing: 
    I/BluetoothLeScanner: start scan is blocked 
    D/BluetoothAdapter: stopLeScan()
and never detect beacon anymore.

### Steps to reproduce this behavior
the beaconManager setup as below:
`backgroundPowerSaver = new BackgroundPowerSaver(this.context);
beaconManager = BeaconManager.getInstanceForApplication(this.context);
beaconManager.setBackgroundBetweenScanPeriod(30000l);
beaconManager.setBackgroundBetweenScanPeriod(01);
beaconManager.setForegroundBetweenScanPeriod(1000l);
beaconManager.setAndroidLScanningDisabled(true);`

### Mobile device model and OS version
Testing with Xiao redmi 3s. Android 6.0.1

### Android Beacon Library version
android_beacon_library2.9.2

I try with Samsung Android 5.0, it work find and scan in background normally. I have already grand every permission that needed.

davidgyoung, 2017-03-26:
@desmond0321, I noticed in the steps to reproduce, you mention that you set `beaconManager.setBackgroundBetweenScanPeriod(01);`, which effectively does constant scans in the background, stopping and restarting every 5 minutes.  

Do you see this same behavior if you use the default background scan settings with `beaconManager.setBackgroundBetweenScanPeriod(30000l); `?

hyo1411, 2017-07-07:
The root cause is that, xiaomi device blocks location service for the app in the background, without location access, ble scanning is blocked. To turn off you need to go to Setting->Battery & performace to grant location access for the app in background

desmond0321, 2017-07-07:
@hyo1411 so grant location access must be user do it manually?
@davidgyoung Yes, same behaviour, but I think @hyo1411 is right, I think is Xiao mi security problem.

hyo1411, 2017-07-07:
@desmond0321 as far I know, it must be done manually for granting location service for the app in background for xiaomi devices (or you can turn off this feature for all apps in Battery & Performance)

nofal-barefoot, 2018-01-31:
@davidgyoung Hey Greetings of the day, How can we restart the background scan services, like the wait scan alarm of 5 minutes which restarts the ble scan , how can we do that programmatically, i have already used stopService() & startService()methods, but the service scan cycles are not restarted, like they are when the application is launched first time?

davidgyoung, 2020-02-28:
This is the same core issue as #643.  We need assistance characterizing what appears to be a custom Xiaomi/Redmi behavior blocking some BLE scans in the background.  Please comment on the linked issue.

closed 2020-02-28 by davidgyoung