AltBeacon/android-beacon-library #488: Cannot build model distance calculations

labwebsites, 2017-04-03:
Hello!

This exception (from the screenshot) is happening every time i run my app for the first time after reinstalling. It seems that the file "model-distance-calculations.json" cannot be found on my device. After the exception, the file is downloaded from AltBeacon server and saved to my device.

Is this an issue or am i doing something wrong?

Thank you!

[image]

davidgyoung, 2017-04-03:
This error message suggest that the library's built-in model-distance-calculations.json file, which is included in the AAR file, somehow is not on the classpath of your app.   That's not normal, and it's not clear why this is happening.  Perhaps it has something to do with how you are packaging your AAR file.  Are you using Android Studio and a release version of this library?

You might try seeing if you can find that file in the APK that your app generates.  If I do this for the android-beacon-library-reference app, I can unzip its APK file and see what is inside.  I get:

```
$ cd   android-beacon-library-reference/app/build/outputs/apk
$ unzip app-debug.apk 
Archive:  app-debug.apk
  inflating: AndroidManifest.xml     
  inflating: META-INF/CERT.RSA       
  inflating: META-INF/CERT.SF        
  inflating: META-INF/MANIFEST.MF    
  inflating: classes.dex             
  inflating: model-distance-calculations.json  
 extracting: res/drawable-xhdpi-v4/ic_launcher.png  
  inflating: res/layout/activity_monitoring.xml  
  inflating: res/layout/activity_ranging.xml  
 extracting: resources.arsc          
```

Do you see this model-distance-calculations.json file in your APK?

wingjjzhang, 2018-09-27:
I have same exception...
I use the newest aar file "2.15.1".

davidgyoung, 2018-09-27:
@wingjjzhang, can you please try the same procedure I suggested above and tell me if you see the model-distance-calculations.json file in your APK?

wingjjzhang, 2018-09-28:
Yes, there are json files in your apk, but not in my own apk. I put aar files directly in my project, and my project didn't put json files in it. 
Theoretically, the json file should be in the aar file, right? 
I tried to decompress it and it turned out to be there. 
But if I put it in my project, then after packaging, there will be no json file in it.

wingjjzhang, 2018-09-28:
I'll look into it further and let you know when I get the results.thx