AltBeacon/android-beacon-library #495: Different RSSI values on different devices

junaidyasin, 2017-04-10:
@davidgyoung I am currently working on a project with iBeacons using this amazing library, in the testing I have discovered that different phones produce different RSSI values at the same distance and scenario. I am using the beacons for payment and the user has to tap his phone on the beacon to trigger the event. There is an RSSI threshold I have set to achieve that (-63). I came up with this value after a lot of testing on 2 devices. But recently I have found out that some devices trigger the event when they are really far away from the beacon, and some do not trigger the event at all because they never reach that threshold value. I have confirmed it in the log and analyzing the live RSSI values.
So my question is does different devices has different RSSI values at 1 meter lets say? If so, then how does one achieve a solution which is viable for at least 70-80% of the devices. 

Note: The testing was done using only one beacon with the same configurations (advertising interval, TxPower)

MadisonBlake, 2017-04-10:
It really comes down to hardware. It is my understanding that different devices have different antennas and will receive the same signal differently.

My general assumption is that raw RSSI observations should not be used to do any threshold-based logic.

davidgyoung, 2017-04-10:
@junaidyasin, this has been one of the most vexing issues I have dealt with since starting work on this library 3 1/2 years ago.

The core issue is that different Android devices have different bluetooth chipsets and different antennas.  If you put two of these device models exactly one meter away from the same beacon, one device may read an average RSSI level of -50 and another one might read an average of -60.  

iOS devices also have this variation -- you can see it between different iPhone models and iPads.  But Android is so much more fragmented, the variations are much greater in magnitude between models, and there are also many, many more models.

The strategy to cope with this has been to create a device-specific database that compensates for this difference as much as possible.   Apps can then use beacon.getDistance() instead of beacon.getRssi() to get a number compensated for device model.   This is not a perfect strategy, but it can at least adjust for coarse variations.   Read here for more info:  http://altbeacon.github.io/android-beacon-library/distance-calculations.html

The main problem with building this database is taking the measurements for specific devices is time consuming and tedious, and as a result, we have not had as many contributions as I would have hoped.  

So I agree with @MadisonBlake -- you really cannot compare RSSI measurements between two different devices.  You have to build something that compensates.  In your case, you may be able to build an automated database generator depending on your app use case, perhaps by having the user hold the phone a known distance from the beacon to determine a  a model-specific correction factor.  Even if you do this, keep in mind the correction factor will only be good as the conditions used to test.  Orientation, phone cases, reflections and obstructions will all affect the results.

tikorth, 2017-04-28:
Here you can find more informations...

Look at "Device Variations"

http://developer.radiusnetworks.com/2014/12/04/fundamentals-of-beacon-ranging.html