AltBeacon/android-beacon-library #496: Beacons not detected after long time scanning

davidgyoung, 2017-04-14:
### Expected behavior
Beacons continue to be detected for long periods of time

### Actual behavior
As reported by @lendylongli, "When I turned off & on bluetooth,it work again."

### Steps to reproduce this behavior
As reported by @lendylongli:
This is easy to reproduce in my device.I modified some of the code, it will be easy to reproduce. BeaconService.java
-- private ExecutorService mExecutor;//it will be problematic, when i start & stop BeaconService many times, the thread pool will create many times, this can cause memory leaks
++ private static final ExecutorService mExecutor = Executors.newSingleThreadExecutor();

start scan with this params
ScanPeriod 200ms
BetweenScanPeriod 0ms

### Mobile device model and OS version
huawei mate 8, Android 6.0

### Android Beacon Library version
Unspecified

This new issue was created off of comments on #122 so as not to hijack that thread

davidgyoung, 2017-04-14:
A few thoughts:

1. The change to `mExecutor` may be causing this problem.  If only a single thread is provided to process beacon detections and it somehow gets blocked, obviously no other detections will be possible.   If you are concerned about the `BeaconService` leaking the thread pool, it might be better to simply add `mExecutor.shutdownNow();` in the `BeaconService#onDestory` method.

2. Since you say you are starting and stopping the service a large number of times, this may be contributing to the issue.  Some logs with `beaconManager.setDebug(true);` of the most recent startup of when there are no detections may be helpful.

3. The extremely short scan periods may be causing the bluetooth chip or stack to lock up.  It would be useful to know if this problem goes away if the interval is set longer to the default of 1100ms.

4. The extremely short scan periods may make it difficult or impossible to detect beacons in a 200ms window of scanning (actually less because it takes some time for scanning to start up.)   Some beacons often transmit only once every second or less. This may just be a timing issue.

mExecutor.shutdownNow();

lendylongli, 2017-04-14:
my code will stop scan when switch background and start scan in foreground. and then i found a lot of thread pool not release with Android Device Monitor.

davidgyoung, 2017-04-14:
Depending on what you see in Android Device Monitor, it may be that the `ExecutorService` objects may not have been garbage collected yet, but are eligible for garbage collection.  I'd be more concerned if threads are continuing to operate.  If there is any evidence that is happening, then the call to `mExecutor.shutdownNow();` might solve that problem.

lendylongli, 2017-04-14:
There is another problem, if use the thread pool will lead to the beacon callback sequence disorders. so i use singleThreadExecutor.

davidgyoung, 2017-04-18:
Yes, it is true that since this library uses multiple threads to process beacon detections, there is no guarantee that all beacons detected in a single scan cycle will be delivered in a ranging or monitoring callback that comes at the end of the cycle.   The library by design is not intended to give such fine grained access to the timing of BLE advertisements, and instead aggregates them to blocks of time configured by these scan cycles.  Whether or not an advertisement received at the very end of one scan cycle ends up being applied to the next scan cycle probably makes no difference to most users, and I think the edge case described is consistent with the design.

lendylongli, 2017-04-19:
@davidgyoung Thank you for your answer. The threadpool problem I have solved
, the same bluetooth address with the same thread to parse beacon.

The problem of scanning does not call back still exists. I guess it is so to reproduce the problem when `bind BeaconService` and `unbind BeaconService` multi-times. And I found the ble scan still working and print  log in `logcat`.

	04-19 10:54:25.001 25847-25889/? I/Bluetooth: BtGatt.GattService:onScanResult() - address=32:01:D5:06:2E:06, rssi=-99
	04-19 10:54:25.002 25847-25889/? I/Bluetooth: BtGatt.GattService:need phone set:requiresLocationEnabled=false, locationEnabled=true
	04-19 10:54:25.002 25847-25889/? I/Bluetooth: BtGatt.GattService:need ble device set:hasPeersMacAddressPermission=false, hasLocationPermission=true, hasPermission:true
	04-19 10:54:25.002 25847-25889/? I/Bluetooth: BtGatt.GattService:matchesFilters  return true
	04-19 10:54:25.002 25847-25889/? I/Bluetooth: BtGatt.GattService:onScanResult() client is not server, CALLBACK_TYPE_ALL_MATCHES
	04-19 10:54:25.003 25847-25900/? I/bt_btm: btm_identity_addr_to_random_pseudo
	04-19 10:54:25.003 25847-25900/? I/bt_btm: BTM_CheckAdvData type=0x01
	04-19 10:54:25.003 25847-25900/? I/bt_btm: BTM_CheckAdvData type=0x19
	04-19 10:54:25.003 25847-25900/? I/bt_btm: BTM_CheckAdvData type=0x03
	04-19 10:54:25.004 25847-25900/? I/bt_btm: btm_identity_addr_to_random_pseudo
	04-19 10:54:25.004 25847-25900/? I/bt_btm: BTM_CheckAdvData type=0x01
	04-19 10:54:25.004 25847-25900/? I/bt_btm: BTM_CheckAdvData type=0x19
	04-19 10:54:25.004 25847-25900/? I/bt_btm: BTM_CheckAdvData type=0x03
	04-19 10:54:25.004 25847-25900/? I/bt_btm: BTM_CheckAdvData type=0x01
	04-19 10:54:25.004 25847-25900/? I/bt_btm: BTM_InqDbRead: bd addr [240ac40633f6]
	04-19 10:54:25.004 25847-25900/? I/bt_btm: BTM_CheckEirData type=0x09
	04-19 10:54:25.004 25847-25889/? I/bt_btm: BTM_CheckEirData type=0x09
	04-19 10:54:25.004 25847-25889/? I/bt_btm: BTM_CheckEirData type=0x08
	04-19 10:54:25.004 25847-25889/? W/bt_btif: HAL bt_gatt_callbacks->client->scan_result_cb

davidgyoung, 2017-04-19:
There were some similar reports about issues with binding/unbinding in #145 and #161, but I have not been able to set up a reproducible test case.   @lendylongli, do you think you could take a crack at making such a test case by modifying the reference app here?  https://github.com/altbeacon/android-beacon-library-reference

fangchenxuan, 2017-07-20:
@davidgyoung When I test on the google pixel (Android N) phone, there will be opportunities to scan the situation, so that onScanresult callback method sometimes can not be called, how do you deal with this problem？

davidgyoung, 2017-07-20:
@fangchenxuan, can you create a test case with the reference app that will reproduce?

fangchenxuan, 2017-07-21:
@davidgyoung I use the android official Bluetooth api, I need to scan the Bluetooth per second, but in the android N system, frequent scanning will be forced to modify the system to opportunity scan, onScanresult method can not be callback, how do you handle this problem?

davidgyoung, 2017-07-21:
@fangchenxuan, it sounds like you are working directly with Android Bluetooth APIs.   I am happy to assist with general  Android bluetooth programming questions on StackOverflow.com, but understand that this forum is reserved for bug reports with this library.  Feel free to ask a question on StackOverflow and I will do my best to answer there.

aanalmehta, 2017-09-20:
Hello,
I am using Huawei P8 lite device. I heard that there some bluetooth issues with Huawei devices.
The main problem is I am not able to connect with my PulseOximeter device. While debugging it with my application, I realized that onConnectionChanged method was not called successfully.
I saw following logs in LogCat. 

Can anyone please help me with this?

09-20 16:10:10.755 1777-1813/? I/Bluetooth: BtGatt.GattService:onScanResult() - address=E1:03:53:D3:53:5F, rssi=-87
09-20 16:10:10.756 1777-1813/? I/Bluetooth: BtGatt.GattService:need phone set:requiresLocationEnabled=false, locationEnabled=true
09-20 16:10:10.756 1777-1813/? I/Bluetooth: BtGatt.GattService:need ble device set:hasPeersMacAddressPermission=false, hasLocationPermission=true, hasPermission:true
09-20 16:10:10.756 1777-1813/? I/Bluetooth: BtGatt.GattService:matchesFilters  return true
09-20 16:10:10.756 1777-1813/? I/Bluetooth: BtGatt.GattService:onScanResult() client is not server, CALLBACK_TYPE_ALL_MATCHES