AltBeacon/android-beacon-library #499: Create bugfix patch release 2.9.3?

cupakromer, 2017-04-24:
Given that there have been several recent bug fixes would it be possible to have patch 2.9.3 released?

The release would need to be attached to commit ef90455e819b7cf82cfd37aadd97e515d9775ac5 which is the last bug fix prior to the new feature introduced in #491.

davidgyoung, 2017-04-24:
Any reason not to include #491?

cupakromer, 2017-04-24:
It was listed as an enhancement in the changelog. And it was a relatively large change compared to the other bug fixes. Was trying to keep semver-ish compliant and stick to bug fixes for a patch level release. But I'm flexible on that.

I didn't dive too deep into the final version just yet but I guess my only concern at the moment is around performance. Since it seems to be based on the device hardware, how likely is that to change between scan cycles? For example, once it triggers "true", will that ability likely change in any reasonable amount of time without say a device restart?

davidgyoung, 2017-04-24:
My inclination would be to issue a release off of master that includes this feature and call it 2.10.

I have done enough testing of #491 to be quite confident that it will perform reliably on newer devices that can detect multiple identical packets per scan cycle.  I can't imagine any reason to think why such a device would stop being able to detect multiple packets per scan, but if it did, then it would not detect the same beacon again until the app is restarted.  If there is any risk to #491 it is that older devices like the Nexus 4 or Moto G1 would use a bit more CPU and drain their battery a little more when doing constant scans with the library.  But since this is generally done in the foreground, with the screen on, I think it is unlikely the change would be noticeable compared with the battery drain of illuminating the screen.

davidgyoung, 2017-04-25:
I rolled a 2.10 release this morning and made it available on Github releases, JCenter and the library download page.

https://bintray.com/altbeacon/android/android-beacon-library
https://github.com/AltBeacon/android-beacon-library/releases/tag/2.10
http://altbeacon.github.io/android-beacon-library/download.html

@cupakromer if you find you need a release without #491 for some reason, please shout.

closed 2017-04-25 by davidgyoung

cupakromer, 2017-04-25:
Sure that works. I just wasn't sure what other things you had wanted to get into a new minor release.