AltBeacon/android-beacon-library #500: Eddystone UID + TLM how get temperature

tikorth, 2017-04-26:
At link above tell about temperature in topic "Eddystone-TLM"

https://altbeacon.github.io/android-beacon-library/eddystone-support.html

But look a sample at topic "Getting Ranging Updates from Nearby Beacons With Telemetry" there is not a sample how get the temperature.

sampe
https://altbeacon.github.io/android-beacon-library/eddystone-how-to.html

well with AltBecon Library is possible get the temperature using beacon Eddystone UID + TLM

davidgyoung, 2017-04-26:
Reading the temp requires a bit of code to convert a long value for the temperature data field and convert it to an signed double value that represents the temperature in Celsius.  Here's example code that does the conversion:

    long unsignedTemp = (beacon.getExtraDataFields().get(2) >> 8);
    double temperature = unsignedTemp > 128 ? unsignedTemp - 256 : unsignedTemp + (beacon.getExtraDataFields().get(2) & 0xff) / 256.0;

tikorth, 2017-04-26:
Very thanks, works fine 🥇

closed 2017-04-26 by tikorth