AltBeacon/android-beacon-library #502: Kalman Filter Feature Request

davidgyoung, 2017-04-29:
The idea of adding a Kalman Filter for RSSI smoothing was mooted on issue #498 by
@ghassanghassan, @lakshmanpasala and @rrjanbiah.

The library is designed to accept custom RSSI smoothing algorithms by implementing the `RSSIFilter` interface.  Simply make a new `KalmanRssiFilter` class that implements `RSSIFilter` interface, then you can put it in action by calling  `BeaconManager.setRssiFilterImplClass(KalmanRssiFilter.class)`

You can see an example of this in the implementation of the [ArmaRssiFilter](https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/ArmaRssiFilter.java)  class, which is an alternative to the default [RunningAverageRssiFilter](https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/RunningAverageRssiFilter.java).

If you can get this going, please submit it as a pull request so we can add it to the library.

imsrgadich, 2017-09-28:
@lakshmanpasala so what are your states in the KF?

marct83, 2020-02-12:
Any Updates on the Kalman Filter request?

davidgyoung, 2020-02-12:
@marct83, no work to date.  We are looking for a volunteer to take it on.  Interested?

Simonida, 2020-05-19:
@davidgyoung When setting RSSI filter class by calling static function, as you mentioned
BeaconManager.setRssiFilterImplClass(RunningAverageRssiFilter.class)
Is RssiFilter instantiated per specific beacons or it is just one instance used for the app?

For example the RunningAverageRssiFilter keeps the reference of the RSSI measurements and it would be problematic if runningAverage RSSI value is calculated from the RSSI values of different beacons.

davidgyoung, 2020-05-19:
@Simonida, yes, of course the tracked rssi readings are kept distinct per tracked beacon.

Simonida, 2020-05-19:
@davidgyoung Thank you. Sorry, I've accidentally deleted the comment, because I confirmed that behaviour by debugging, before I saw your answer.

I have just one question, can we somehow control the interval when addMeasurement function, in RssiFilter class,  would be called? Setting beacon scan period to BeaconManager seems to doesn't have any impact.

davidgyoung, 2020-05-19:
As designed, any filter implementation gets one RSSI sample each time a matching packet is detected by the bluetooth chip and the operating system.   Changing the scan period will not affect this as multiple packets may be detected in a single scan period and all are passed to the filter by design.   If you don't want that all samples for some reason, you can certainly build a custom filter that discards some based on some time interval.