AltBeacon/android-beacon-library #503: Background service cannot find beacon after stopping from task switcher (in 2.7 works, in 2.10 does not)

smedic, 2017-05-04:
### Expected behavior
Beacon should be detected
### Actual behavior
Beacon is not detected
### Steps to reproduce this behavior
Start app (beacon detected in background service)
Stop app from task switcher
Start app again (beacon not detected)
(beacon cannot be found until app is killed from app settings, phone restart, or app reinstall)
### Mobile device model and OS version
Samsung S7 , Nougat 7.0
### Android Beacon Library version
2.10 (on 2.7 works)

What is changed in 2.10? Do you have any idea?

I read a lot of posts and your description how to restart, but did not find any clue.

Thanks

davidgyoung, 2017-05-04:
To clarify, you are saying that starting with a switch from 2.7 to 2.10, you see that if you ever use the task switcher to kill the app, it breaks further detections?  And the only way to get out of this is to reboot the phone, reinstall the app, or kill the app from settings?

smedic, 2017-05-04:
Yes. All regions are correctly set, didDetermineStateForRegion is called for every region (I have 4) and I call startRangingBeaconsInRegion for all of them. 
I have found some previous issues where you said that we should use stopRangingBeaconsInRegion because data is persisted in a file, but this does not help.

In 2.7 everything is okay.

smedic, 2017-05-04:
When killed from application settings, it's okay.

davidgyoung, 2017-05-04:
This may be caused by region persistence, which was added after 2.7.  Three things would help track down what is causing this for you.

1. Please clarify whether you are not getting `didEnterRegion` callbacks or whether you are not getting `didRangeBeaconsInRegion` callbacks.
2. Please add beaconManager.setDebug(true); and capture a log of 30 seconds+ in the foreground that covers the period where a beacon is around where you are not getting the detections you expect.
3. Please try adding `beaconManager.setRegionStatePersistenceEnabled(false);` and see if this allows you to detect.

smedic, 2017-05-05:
1. In a successful try, I get didEnterRegion once after fresh install. And that works. 
After stopping from task switcher, I don't get didEnterRegion.
didRangeBeaconsInRegion is called 4 times every 9 seconds (for every region), but beacons collection size is 0. In successful try, collection size is 1, which is normal behavior.
2. I will add log later, and post here.
3. setRegionStatePersistenceEnabled is not helpful for me..

smedic, 2017-05-05:
This is the log. In first try you will see my log (beside yours) "beacon in range! Distance: xxx".
Somewhere around the middle of the file, there are a few empty lines, and from that place, app is started again and no new beacons are found.
[beacon_lib_log.txt](https://github.com/AltBeacon/android-beacon-library/files/979505/beacon_lib_log.txt)

davidgyoung, 2017-05-05:
@smedic, good news: I see the problem.

Somehow your BeaconParser is not getting set before the scanning service starts up in the case where the app is not detecting.  As a result, only beacons of type AltBracon may be detected after the app restarts.

I believe there is a subtle timing issue at play, and perhaps it just *barely* worked with 2.7, and the timing is a bit different with 2.10, and even then only after some apps restarts.

Whatever triggers  it, the fix should be simple.  Just move the place where you set the BeaconParser as early as possible in code, before you bind or construct a RegionBootstrap.

smedic, 2017-05-05:
You are right. I moved beacon manager creating in Application class and it worked!
When it is located in onStartCommand in service, it does not work. 

Sorry for bothering you with debugging! 
Library is great and support is event better.

Best regards

closed 2017-05-05 by smedic