AltBeacon/android-beacon-library #504: Android O Preview Background Limitation.

widevisioner, 2017-05-05:
### Expected behavior
The Monitoring of the Beacon should happen. didEnterRegion method should be called.
### Actual behavior
The Monitoring of the Beacon should not happening . didEnterRegion method is not called.
### Steps to reproduce this behavior
Try Monitoring any beacons in an app target Android O Preview OS.
### Mobile device model and OS version
 Google Pixel and Android O Preview
### Android Beacon Library version
compile 'org.altbeacon:android-beacon-library:2+'
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-05-05:
Android O introduces significant changes that block long-running background services.  For this library, it means its BeaconService will be killed by the operating system, blocking beacon detections.

The planned solution is to switch to scheduled jobs to do scanning when the library is installed on Android O or later.  

A very rough proof of concept is available to try in #484.  Please feel free to test this and send feedback, but be advised it is should not be put to production use.

Svoka, 2017-06-30:
Maybe we can scan in Foreground service? I don't see why it might be a bad decision.

davidgyoung, 2017-06-30:
Yes, a foreground service may be a solution for apps like dedicated beacon scanner tools, where a user wants a visual reminder that it is happening.  But this is less good for apps that use beacons to re-launch themselves or send notifications on detection for purposes that users might not be aware of.  In these cases, always showing an icon to indicate the app is running might put off some users.

Fortunately, my testing on Android O shows that the JobScheduer strategy combined with background scanning that delivers detection results between scans works quite well.