AltBeacon/android-beacon-library #505: Measurements added are inversely proportional to the distance

rrjanbiah, 2017-05-05:
### Background

* Was using https://github.com/Bridouille/android-beacon-scanner for scanning real beacons
* Found some inconsistencies in readings. For example, even without moving beacons, the distance reading jumped randomly.
* Thought that Kalman will help to reduce the noise. And so, landed here #502
* Based on suggestions at #502 by @davidgyoung, tried to understand `RunningAverageRssiFilter` and `ArmaRssiFilter`... 
* After enabling logging, noted that "measurements" adding https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/RunningAverageRssiFilter.java#L24 reduced to single call when moved to some distance; and it is higher (say around 16) when scanner is close to beacon. 
* And so, noted `sampleExpirationMilliseconds` logic used to expire beacons (IOW, without this handling, it will be always averaging a single beacon when beacon is kept at some distance)
* Tried to change so many values so that this "measurement samples count" will be higher--when beacons are kept at distance. But, that didn't work.
* (If this is not a bug and intended behavior, please share which values to be adjusted)

### Mobile device model and OS version

Nexus - 5  & Android Version - 6.0.1

### Android Beacon Library version

master

davidgyoung, 2017-05-05:
It is possible and expected that if you increase the distance between the beacon and the Android device that the number of packets it will receive will decrease, as the signal to noise ratio decreases and more packets are lost.  However, I would expect the number of samples received to go to zero at some point and not stay at one.

How far away is "some distance"?  

Are you sure the Android device is still able to detect the beacon?

rrjanbiah, 2017-05-09:
@danialgoodwin Many thanks for your quick reply. 

Taken some samples today. But, unfortunately, didn't get 16 `measurements` now and not sure why. Anyhow added all data here so that if you have any suggestions, please do share:

```
Nearer

2 measurements: [-3.0, -35.0]
3 measurements: [-31.0, -31.0, -30.0]
4 measurements: [-31.0, -84.0, -30.0, -31.0]
1 measurements: [-29.0]
4 measurements: [-32.0, -29.0, -28.0, -29.0]
2 measurements: [-32.0, -32.0]
4 measurements: [-29.0, -1.0, -32.0, -1.0]
3 measurements: [-32.0, -1.0, -31.0]
3 measurements: [-31.0, -2.0, -31.0]

60 cm apart

4 measurements: [-66.0, -65.0, -63.0, -63.0]
3 measurements: [-61.0, -63.0, -64.0]
4 measurements: [-63.0, -63.0, -60.0, -44.0]
4 measurements: [-60.0, -60.0, -60.0, -60.0]
5 measurements: [-60.0, -60.0, -60.0, -60.0, -60.0]
4 measurements: [-61.0, -61.0, -44.0, -60.0]
4 measurements: [-60.0, -60.0, -60.0, -60.0]
5 measurements: [-61.0, -44.0, -61.0, -44.0, -58.0]
3 measurements: [-61.0, -61.0, -62.0]
4 measurements: [-61.0, -61.0, -62.0, -62.0]
4 measurements: [-62.0, -62.0, -61.0, -61.0]
3 measurements: [-61.0, -58.0, -62.0]

150 cm apart

3 measurements: [-66.0, -66.0, -79.0]
3 measurements: [-63.0, -84.0, -71.0]
2 measurements: [-95.0, -85.0]
2 measurements: [-75.0, -70.0]
1 measurements: [-75.0]
3 measurements: [-85.0, -60.0, -80.0]
4 measurements: [-62.0, -79.0, -83.0, -61.0]
4 measurements: [-80.0, -61.0, -79.0, -64.0]
2 measurements: [-79.0, -79.0]
3 measurements: [-63.0, -70.0, -83.0]
3 measurements: [-66.0, -81.0, -70.0]
3 measurements: [-68.0, -81.0, -81.0]
4 measurements: [-83.0, -65.0, -82.0, -64.0]
2 measurements: [-76.0, -77.0]

300 cm apart

2 measurements: [-87.0, -83.0]
1 measurements: [-75.0]
1 measurements: [-74.0]
1 measurements: [-70.0]
3 measurements: [-92.0, -77.0, -75.0]
1 measurements: [-76.0]
2 measurements: [-75.0, -77.0]
2 measurements: [-84.0, -84.0]
1 measurements: [-68.0]
2 measurements: [-87.0, -68.0]

5.33 m apart

2 measurements: [-73.0, -95.0]
2 measurements: [-71.0, -94.0]
2 measurements: [-79.0, -74.0]
1 measurements: [-72.0]
2 measurements: [-75.0, -75.0]
5 measurements: [-88.0, -74.0, -89.0, -71.0, -70.0]
1 measurements: [-71.0]
4 measurements: [-68.0, -69.0, -88.0, -88.0]
1 measurements: [-87.0]
1 measurements: [-90.0]

6.3 m apart

2 measurements: [-90.0, -73.0]
1 measurements: [-76.0]
2 measurements: [-91.0]
1 measurements: [-73.0]
1 measurements: [-100.0]
1 measurements: [-94.0]
1 measurements: [-78.0]
1 measurements: [-80.0]
1 measurements: [-77.0]
1 measurements: [-76.0]
1 measurements: [-75.0]
```

Over the top of my head, I think, if we can filter out `rssi` between `0` and `-5` or so, we may get better averages. Similarly, if we can adjust some parameters to increase the `measurements` count, it will help. I'm kinda stuck in identifying the later parameter--as nothing seems to increase the `measurements` count.

TIA

davidgyoung, 2017-05-09:
I am surprised by how few packets are detected at 6.3 meters.

Before going further, is it possible to do similar tests with a reference transmitter to eliminate the transmitter as possible cause of the issue?

I might suggest using an Android or iOS device at the transmitter.  I have a Nexus 5 and when I get RSSI measurements from beacon packets from an iOS or Android transmitter advertising at 10Hz, I typically see 8-9 packets detected at -60 dBm at 100 cm and 6-6 detected at 10m at around -90 dBm.  If you can do these tests with one of these transmitters, I can repeat the same test with my Nexus 5 and determine if we are seeing the same thing.

rrjanbiah, 2017-05-10:
@davidgyoung Many thanks for your great support.

Above reading was using Beacon simulator app https://play.google.com/store/apps/details?id=net.alea.beaconsimulator at 3Hz frequency. And, below reading are using 10Hz frequency. Still `measurements` count are sometimes low.

```
Nearer

9 measurements: [-22.0, -22.0, -22.0, -22.0, 0.0, -22.0, -24.0, -25.0, -22.0]
9 measurements: [-22.0, -22.0, -22.0, 0.0, -22.0, -22.0, -21.0, -22.0, -24.0]
10 measurements: [-24.0, -24.0, -20.0, -24.0, -24.0, -19.0, -23.0, -24.0, -24.0, -23.0]
8 measurements: [-24.0, -23.0, -24.0, -24.0, -23.0, -24.0, -23.0, -19.0]
7 measurements: [-23.0, -23.0, -19.0, -23.0, -19.0, -23.0, -23.0]
8 measurements: [-19.0, -19.0, -19.0, -19.0, -19.0, -24.0, -23.0, -19.0]
10 measurements: [-23.0, -24.0, -23.0, -23.0, -23.0, -19.0, -19.0, -21.0, -24.0, -21.0]
10 measurements: [-24.0, 0.0, 0.0, -19.0, -24.0, -19.0, -23.0, -77.0, -78.0, 0.0]
9 measurements: [0.0, 0.0, -24.0, -21.0, -23.0, -23.0, -24.0, -23.0, 0.0]
10 measurements: [-23.0, 0.0, -24.0, 0.0, 0.0, 0.0, -24.0, 0.0, -24.0, -23.0]
8 measurements: [-24.0, -24.0, -24.0, -21.0, -24.0, -24.0, -24.0, -22.0]
10 measurements: [-26.0, -22.0, -22.0, -22.0, -22.0, -22.0, -22.0, 0.0, -22.0, 0.0]
10 measurements: [-22.0, -22.0, -25.0, -21.0, -24.0, -25.0, -22.0, -22.0, -25.0, -22.0]

at 1m

9 measurements: [-74.0, -74.0, -55.0, -76.0, -78.0, -75.0, -74.0, -72.0, -72.0]
10 measurements: [-71.0, -71.0, -70.0, -69.0, -52.0, -51.0, -52.0, -70.0, -72.0, -68.0]
9 measurements: [-71.0, -71.0, -72.0, -74.0, -73.0, -55.0, -57.0, -73.0, -74.0]
9 measurements: [-74.0, -74.0, -76.0, -76.0, -75.0, -75.0, -58.0, -75.0, -75.0]
8 measurements: [-75.0, -57.0, -55.0, -74.0, -74.0, -70.0, -70.0, -72.0]
10 measurements: [-71.0, -72.0, -71.0, -68.0, -68.0, -71.0, -51.0, -70.0, -69.0, -69.0]
10 measurements: [-51.0, -51.0, -51.0, -51.0, -69.0, -68.0, -70.0, -69.0, -69.0, -70.0]
9 measurements: [-51.0, -70.0, -51.0, -70.0, -69.0, -70.0, -69.0, -69.0, -70.0]
10 measurements: [-69.0, -51.0, -68.0, -68.0, -69.0, -70.0, -70.0, -68.0, -69.0, -69.0]
3 measurements: [-69.0, -70.0, -86.0]
1 measurements: [-71.0]
5 measurements: [-80.0, -82.0, -79.0, -78.0, -77.0]
10 measurements: [-58.0, -76.0, -74.0, -76.0, -75.0, -75.0, -76.0, -75.0, -75.0, -59.0]

at 2m

9 measurements: [-74.0, -77.0, -76.0, -72.0, -59.0, -55.0, -74.0, -57.0, -57.0]
8 measurements: [-55.0, -71.0, -70.0, -54.0, -61.0, -55.0, -55.0, -56.0]
10 measurements: [-57.0, -59.0, -79.0, -62.0, -59.0, -57.0, -56.0, -55.0, -55.0, -54.0]
8 measurements: [-57.0, -56.0, -58.0, -77.0, -75.0, -55.0, -54.0, -54.0]
8 measurements: [-55.0, -71.0, -71.0, -71.0, -71.0, -71.0, -72.0, -76.0]
10 measurements: [-58.0, -57.0, -57.0, -75.0, -75.0, -56.0, -56.0, -74.0, -56.0, -56.0]
10 measurements: [-57.0, -56.0, -55.0, -54.0, -55.0, -55.0, -56.0, -55.0, -55.0, -54.0]
8 measurements: [-55.0, -55.0, -55.0, -74.0, -54.0, -56.0, -55.0, -57.0]
11 measurements: [-72.0, -54.0, -54.0, -53.0, -53.0, -53.0, -53.0, -54.0, -54.0, -54.0, -53.0]
9 measurements: [-54.0, -55.0, -74.0, -58.0, -58.0, -59.0, -57.0, -59.0, -58.0]
10 measurements: [-59.0, -58.0, -75.0, -76.0, -76.0, -78.0, -77.0, -77.0, -76.0, -77.0]
8 measurements: [-75.0, -60.0, -59.0, -59.0, -59.0, -60.0, -59.0, -60.0]
10 measurements: [-75.0, -59.0, -59.0, -59.0, -59.0, -77.0, -75.0, -76.0, -76.0, -77.0]
9 measurements: [-59.0, -58.0, -60.0, -60.0, -59.0, -59.0, -59.0, -72.0, -56.0]
9 measurements: [-57.0, -56.0, -55.0, -72.0, -55.0, -73.0, -55.0, -72.0, -74.0]
8 measurements: [-72.0, -55.0, -56.0, -73.0, -73.0, -55.0, -55.0, -55.0]
10 measurements: [-73.0, -55.0, -73.0, -56.0, -73.0, -55.0, -54.0, -55.0, -55.0, -55.0]
9 measurements: [-55.0, -73.0, -55.0, -72.0, -55.0, -55.0, -55.0, -55.0, -54.0]
10 measurements: [-55.0, -55.0, -70.0, -53.0, -53.0, -53.0, -53.0, -54.0, -53.0, -71.0]

at 3m

2 measurements: [-68.0, -85.0]
3 measurements: [-83.0, -81.0, -84.0]
1 measurements: [-81.0]
3 measurements: [-84.0, -81.0, -81.0]
5 measurements: [-81.0, -63.0, -63.0, -63.0, -63.0]
9 measurements: [-82.0, -80.0, -82.0, -80.0, -82.0, -63.0, -63.0, -63.0, -62.0]
8 measurements: [-80.0, -80.0, -80.0, -62.0, -63.0, -63.0, -63.0, -62.0]
8 measurements: [-62.0, -63.0, -63.0, -81.0, -81.0, -81.0, -84.0, -82.0]
10 measurements: [-82.0, -63.0, -63.0, -80.0, -82.0, -82.0, -80.0, -81.0, -81.0, -81.0]
[... 28 lines cut ...]
9 measurements: [-63.0, -77.0, -78.0, -80.0, -80.0, -61.0, -79.0, -79.0, -78.0]
10 measurements: [-61.0, -78.0, -79.0, -78.0, -79.0, -79.0, -76.0, -82.0, -78.0, -81.0]
8 measurements: [-61.0, -79.0, -79.0, -79.0, -81.0, -79.0, -79.0, -80.0]
10 measurements: [-80.0, -61.0, -80.0, -78.0, -78.0, -60.0, -79.0, -77.0, -81.0, -61.0]
7 measurements: [-78.0, -81.0, -78.0, -60.0, -78.0, -60.0, -86.0]
6 measurements: [-84.0, -83.0, -87.0, -84.0, -68.0, -83.0]

at 7m

2 measurements: [-87.0, -92.0]
3 measurements: [-73.0, -93.0, -94.0]
1 measurements: [-92.0]
2 measurements: [-87.0, -91.0]
9 measurements: [-82.0, -81.0, -82.0, -83.0, -79.0, -81.0, -80.0, -63.0, -80.0]
9 measurements: [-79.0, -81.0, -82.0, -81.0, -81.0, -82.0, -81.0, -81.0, -79.0]
10 measurements: [-80.0, -83.0, -81.0, -81.0, -82.0, -82.0, -81.0, -80.0, -79.0, -80.0]
8 measurements: [-82.0, -81.0, -80.0, -80.0, -82.0, -82.0, -81.0, -83.0]
7 measurements: [-81.0, -81.0, -81.0, -82.0, -84.0, -83.0, -87.0]
6 measurements: [-84.0, -88.0, -86.0, -81.0, -88.0, -87.0]
4 measurements: [-87.0, -89.0, -89.0, -84.0]
6 measurements: [-89.0, -88.0, -87.0, -90.0, -88.0, -90.0]
6 measurements: [-87.0, -89.0, -89.0, -90.0, -71.0, -74.0]
```

Briefly played with the configurations in https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/BeaconManager.java#L157 , but that don't seem to help in increasing the count. Do you have any suggestions? TIA

davidgyoung, 2017-05-10:
Well this looks better.  You are at least getting 10 measurements sometimes.  I am not sure why it drops down at 4m, but maybe there is some signal interference or multipath at that location?

One other think you might try is using the 2.10 version of the library if you are not already.   That version added an optimization that does not stop scanning between cycles on devices (like the Nexus 5) that support receiving multiple identical packets per scan.

rrjanbiah, 2017-05-18:
@davidgyoung Again many thanks for your wonderful support. When we switched to 2.10, measurements count again went down to 1 for many distances. And, as mentioned earlier, we couldn't able to increase the measurements count by adjusting other constants. This made the distance reading to fluctuate considerably. And due to time constraint, we worked at distance reading level (instead of rssi level) of coding and added couple of flags to streamline it to detect only the `enter` and `exit`.. as it solved our current requirement.

davidgyoung, 2017-05-20:
Glad you were able to solve your requirement after all this trouble!

closed 2017-05-20 by davidgyoung