AltBeacon/android-beacon-library #506: didRangeBeaconsInRegion stops Firing after 30 minutes

fjserrano, 2017-05-07:
### Expected behavior

I run app. (The library is in service that control other things). Lock the screen and not touching phone anymore during 2 hours. The didRangeBeaconsInRegion event must firing every 2 second all the time.

### Actual behavior

When passed 30 minutes from start and I lock screen the didRangeBeaconsInRegion event not firing anymore. If touch the power button and iluminates the lock screen didRangeBeaconsInRegion firing again. If lock the screen again didRangeBeaconsInRegion stops firing again.

### Steps to reproduce this behavior

- Create APP or service with implements BeaconConsumer
- Running APP or service and lock screen.
- Not touch anymore the phone.
- After 30 minutes from lock screen the didRangeBeaconsInRegion not firing anymore.

### Mobile device model and OS version

LG Nexus 5 - Android 7.1.1

### Android Beacon Library version

compile 'org.altbeacon:android-beacon-library:2+'

I test from different parameters and always obtain the same result.

I test with:
beaconManager.setBackgroundScanPeriod(2000);
beaconManager.setBackgroundBetweenScanPeriod(2);

With:
beaconManager.setBackgroundBetweenScanPeriod(30000l);

With:
beaconManager.setAndroidLScanningDisabled(true);

In my APP I need to read beacons every 2 second or maximum 4 second. If I lost a beacon I need to notify the beacon lost.

I reproduce the issue modifying the altbeacon app reference to test. I attach my log and the log of beacon reference app.

In my app the 30 minutes passed in the time "05-06 21:04:36.320"
In beacon reference app the 30 minutes passed in the time "05-07 12:41:24.872"

I think that didRangeBeaconsInRegion not firing because the scanner is not running in background anymore, but I don't know why.

I have the apps in the no optimized battery for doze mode. I tested forcing doze mode manually and always pass after 30 minutes from lock the screen, forcing or not forcing doze mode.

[myapp30minutes1.txt](https://github.com/AltBeacon/android-beacon-library/files/981514/myapp30minutes1.txt)
[appreference2.txt](https://github.com/AltBeacon/android-beacon-library/files/981515/appreference2.txt)

Thanks in advance for your support.

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-05-07:
@fjserrano, if the phone is motionless for long enough, Android 6.0+ will put the device into Doze mode which will stop the AlarmManager from firing to keep the scanning service alive.  I suspect that if you move your phone after it gets into this state, then beacon scanning will resume in the next five minutes.

I am not sure I understand what you say about Doze mode here: "I have the apps in the no optimized battery for doze mode. I tested forcing doze mode manually and always pass after 30 minutes from lock the screen, forcing or not forcing doze mode."  Can you please clarify?

fjserrano, 2017-05-08:
Hi again.

Thanks for your quick response @davidgyoung.

Sorry for my bad expression referring to "I have the apps in the no optimized battery for doze mode....". I clarify my expressions.

I put my apps in the Settings -> Battery -> in option of Battery optimization. I add in this menu the apps to "not optimized" battery mode (https://support.google.com/nexus/answer/7015477?hl=en).

When referring to forcing doze mode, I reffer to put the device in this mode (https://developer.android.com/training/monitoring-device-state/doze-standby.html?hl=es#testing_doze). And I said forcing this mode, the scanner stops at 30 minutes after lock screen.

I test other things like put a alarm in phone every 10 minutes. When ring I lock again the phone with other 10 minutes alarm. Doing this scanner BLE never stops. And all runs well, but is not the way to run in a final APP.

If I put a Alarm extern to phone every 10 minutes for me and only move the phone without unlock the screen at 30 minutes after of lock the screen the scanner BLE stops.

I think the issue is relationated with the thing you tell me, but I didn't see moving the phone, the scanner restart 5 minutes after but I will test.

Anyway If this occurs I will not have function didRangeBeaconsInRegion reading beacons always?? If it works like you say only I can monitor beacons every 5 minutes when enter in this mode the phone no??

I will do others test and I say the result of this tests.

Thanks.

Best regards.

fjserrano, 2017-05-13:
Hi @davidgyoung.

  I compile with the option "compile 'org.altbeacon:android-beacon-library:2.10'" and it runs in background after 30 minutes, but I don't understand one thing. I question you if you can help me. With "compile 'org.altbeacon:android-beacon-library:2.+'" the background mode is not working for me.

 This is my code for timmings of Scan:
```

beaconManager.setForegroundScanPeriod(4001L);
beaconManager.setForegroundBetweenScanPeriod(1001L);
beaconManager.setBackgroundScanPeriod(4001L);
beaconManager.setBackgroundBetweenScanPeriod(1001L);
beaconManager.bind(this);
if (beaconManager.isBound(this)) beaconManager.updateScanPeriods();

```
When the service is in foreground, it is always running continuisly without stopping between scan period.

But when beaconManager detect, it is need to run in background mode the timings is not equivalent to my settings. I will to explain me.

With this values I understand when the system go to background the beaconmanager read for 4 seconds the beacons in every cycle of read. But in the logcat I see the CycledLeScannerForLollipop only read beacons and got results only 1 second. After this its waits about 10 seconds for other 1 second of results.

For my APP if I get 10 seconds between read cycles its OK, but I need at least 4 seconds reading results because my beacons have a advertising cycle of 1,2 seconds.

How can I force to got results for 4 seconds in every cycle of reads with the beaconManager??.

Thanks for your help.

Best Regards.

Here PUT a example of logcat:

[myapp30minutes2.txt](https://github.com/AltBeacon/android-beacon-library/files/998510/myapp30minutes2.txt)

davidgyoung, 2017-05-13:
I do see from the log that the library is not detecting BLE packets at all in some scan cycles, and it looks to me like something is blocking scan results for 15 seconds at a time.  See my filtered log excerpt below.

A similar issue was reported in #483, but this appeared to be a Samsung-specific OS modification that would force  a low power scan in the background that would cause dropped detections.  `D BtGatt.GattService: Its third party background application, change scanmode to low power`   Sorry to ask for even more logs, but is is possible for you to get a similar log with system log messages included?  That way we can see if the operating system is blocking scanning for so long in the background.

```
05-13 12:01:32.094 10061-10061/testexample.com.beaconble D/CycledLeScanner: Scan started
05-13 12:01:36.238 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.269 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.286 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.296 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.348 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.408 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.438 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.454 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.553 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.562 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.565 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.572 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.584 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.654 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.680 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.684 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.699 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.733 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.755 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.810 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.817 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.831 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.836 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.859 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.929 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.944 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.953 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.964 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:36.967 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:37.067 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:37.070 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:37.103 10061-10061/testexample.com.beaconble D/CycledLeScanner: Scan started
05-13 12:01:42.099 10061-10061/testexample.com.beaconble D/CycledLeScanner: Scan started
05-13 12:01:47.099 10061-10061/testexample.com.beaconble D/CycledLeScanner: Scan started
05-13 12:01:51.225 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:51.284 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
05-13 12:01:51.290 10061-10061/testexample.com.beaconble D/CycledLeScannerForLollipop: got record
```

fjserrano, 2017-05-13:
Hi @davidgyoung.

When I will go back at home, I put de logcat of android studio without filters and save all the logcat that system gives to me. And don't worry about to tell me send more data to you, if I can help I will be happy.

Best Regards.

fjserrano, 2017-05-13:
Hi @davidgyoung.

Here I put 2 files, one from android studio with no filters in logcat and other from adb logcat. If you need more things or to test other things tell to me and I do it. 

I tested this on my Nexus 5 android version 7.1.1.

I will going to test too in Moto G android version 7.1.1.

Best Regards.

[logcatsystem.txt](https://github.com/AltBeacon/android-beacon-library/files/998721/logcatsystem.txt)
[myapplogcatwithoutfilters.txt](https://github.com/AltBeacon/android-beacon-library/files/998722/myapplogcatwithoutfilters.txt)

davidgyoung, 2017-05-13:
@fjserrano, can you please tell me if this log line is coming from our code?  if so, can you share what it is doing?

`05-13 16:37:39.803 10061-10289/testexample.com.beaconble D/ServiceBLE: Service: restarting scan`

fjserrano, 2017-05-13:
This log is from my Service in a timer. Is From my application "schedule(mReScanTimerTask, 0L, 2000L);"

This timer isn't doing nothing with bluetooth.

fjserrano, 2017-05-13:
With Moto G i am having the same behaviour. I upload to you the logcat files like with the nexus 5.

If you want I can do any software with your requisites to test the library. 

How i mentioned to you, I only need a scan every 10 seconds and the scanner running at least 4 seconds continuously

Best Regards.

[myapplogcatwithoutfiltersmotog.txt](https://github.com/AltBeacon/android-beacon-library/files/998777/myapplogcatwithoutfiltersmotog.txt)
[logcatsystemmotog.txt](https://github.com/AltBeacon/android-beacon-library/files/998778/logcatsystemmotog.txt)

davidgyoung, 2017-05-13:
@fjserrano, I was able to reproduce your problem with 2.10, but not with the latest code in master, as it seem the fix from #507 also addresses the problem you are seeing.

Please try this binary build and see if it solves your problem:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.11-beta1

fjserrano, 2017-05-13:
@davidgyoung i can reproduce this behavior quickly. 

- I turn on the App.
- After this, i turned off Bluetooth adapter and waits until the library tell in logcat that it enters in background mode to search beacons.
- Turn on the bluetooth adapter and this behavior appears.

From this mode I can test more quickly the different options. if i find the mode that I am looking for, I will test the stability of searching beacons for hours.

I am going to test the binary build and put the results.

Best regards.

fjserrano, 2017-05-13:
@davidgyoung with the version 2.11beta1 of library all is running fine. I will test more hours without touching the phone, and tomorrow i will test in a moto g. 

Next week i will test in a samsung prime with Android 5.0.1.

When I have the results of different devices I put the results here.

I think you are doing a great work with this library. If you need some help to test in differente devices or something, tell me and I can do it.

Best Regards.

davidgyoung, 2017-05-20:
@fjserrano, I am closing this issue since I believe I have documented the way to reproduce the problem and shown the fix in #512

closed 2017-05-20 by davidgyoung