AltBeacon/android-beacon-library #509: didExitRegion called when goes to background in Google Pixel

roshanparmar, 2017-05-12:
I am using AltBeacon library to use iBeacon in my Android application but getting an issue, explained below - 
### Expected behavior
Expected behavior should not exit from region means didExitRegion should not call when go to background in Google pixel and Samsung s6 edge with Android O.S. version 7.1.2.
### Actual behavior
But Actual behavior is - when I go to background in Google pixel and Samsung s6 edge  then immediately didExitRegion called and I disconnects from my beacon.
### Steps to reproduce this behavior
1. Come in the range of Beacon in Google Pixel and Samsung s6 edge .
2. press home button of Android device means make an app to background.
3. now open any other application from Google pixel and Samsung s6 edge.
4. You will be disconnected from beacon means didExitRegion called.

### Mobile device model and OS version
1. Google pixel with Android O.S. version 7.1.2
2. Samsung s6 edge with Android version  7.x

### Android Beacon Library version
2.9.2

Here the code for Beacon Manager - 
    beaconManager = org.altbeacon.beacon.BeaconManager.getInstanceForApplication(this);

            beaconManager.getBeaconParsers().clear();
            beaconManager.getBeaconParsers().add(new BeaconParser().
                    setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
            beaconManager.setBackgroundScanPeriod(1000L);
            beaconManager.setBackgroundBetweenScanPeriod(1000L);
            beaconManager.setForegroundScanPeriod(1000L);
            beaconManager.setForegroundBetweenScanPeriod(1000L);

      /*      beaconManager.updateScanPeriods();
            beaconManager.setDebug(true);*/

            Identifier identifier = Identifier.parse("C48C6716-193F-477B-B73A-C550CE582A22");
            // wake up the app when a beacon is seen
            region = new Region("backgroundRegion",
                    identifier, null, null);
            regionBootstrap = new RegionBootstrap(this, region);

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-05-12:
Thanks for the report, @roshanparmar.  Can you please attach a log with  beaconManager.setDebug(true); that shows the exit event and prior 20 seconds?

davidgyoung, 2017-05-12:
By the way there were some changes in 2.10 to fix some edge cases with Android 7.0 blocking too many scans in a short period of time.  The problem you describe might be intermittently triggered with 2.9.2 using the 1000ms on / 1000 ms off scan settings you have in your code.  But I would not expect it to happen constantly.  Regardless, it may be worth verifying it still happens with 2.10.

roshanparmar, 2017-05-15:
Thanks to look my problem.
As you asked about logs so I can not give it to you because my client have both Google pixel and samsung s6 edge devices and he is facing same issue as I mentioned to you.
After resd some articles, it is clear that altbeacon library does not work properly with some of devices.

I am trying 2.10 and looking for proper work.

If anything goes wrong then I will let you know and If you would find anything more then please let me know.

roshanparmar, 2017-05-15:
After updating the library from 2.9.2 to 2.10, I am facing this issue my end too, in my LG Nexus 5x device, first time beacon runs scuccesfully but after first exit of beacon, it does run good means when beacon exit for first time then didEnter method called after a long time just for few seconds and then didExit calls. This behavior comes frequently on my device. Below I am attaching logs as you said above - 

05-15 17:19:25.229 29726-29726/com.app.dealer_app D/CycledLeScanner: Waiting to stop scan cycle for another 981 milliseconds
05-15 17:19:25.235 29726-29726/com.app.dealer_app D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{6158fab: android.os.BinderProxy@2367608}
05-15 17:19:26.218 29726-29726/com.app.dealer_app D/CycledLeScanner: Done with scan cycle
05-15 17:19:26.219 29726-29726/com.app.dealer_app D/BeaconService: Calling ranging callback
05-15 17:19:26.221 29726-29726/com.app.dealer_app D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
05-15 17:19:26.232 29726-29726/com.app.dealer_app D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
05-15 17:19:26.233 29726-29726/com.app.dealer_app D/CycledLeScanner: Normalizing between scan period from 1000 to 964
05-15 17:19:26.233 29726-29726/com.app.dealer_app D/CycledLeScannerForLollipop: This is Android L. Doing a filtered scan for the background.
05-15 17:19:26.236 29726-29726/com.app.dealer_app D/CycledLeScannerForLollipop: starting filtered scan in SCAN_MODE_LOW_POWER
05-15 17:19:26.239 29726-29726/com.app.dealer_app D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=76, mManufacturerData=[2, 21], mManufacturerDataMask=[-1, -1]]
05-15 17:19:26.240 29726-29726/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 964 milliseconds
05-15 17:19:26.241 29726-29790/com.app.dealer_app D/BluetoothAdapter: isLeEnabled(): ON
05-15 17:19:26.244 29726-29726/com.app.dealer_app D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{6158fab: android.os.BinderProxy@2367608}
05-15 17:19:26.247 29726-1370/com.app.dealer_app D/BeaconIntentProcessor: got an intent to process
05-15 17:19:26.249 29726-1370/com.app.dealer_app D/BeaconIntentProcessor: got ranging data
05-15 17:19:26.252 29726-29738/com.app.dealer_app D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
05-15 17:19:27.212 29726-29726/com.app.dealer_app D/CycledLeScanner: starting a new scan cycle
05-15 17:19:27.213 29726-29726/com.app.dealer_app D/CycledLeScanner: We are already scanning
05-15 17:19:27.214 29726-29726/com.app.dealer_app D/CycledLeScanner: Waiting to stop scan cycle for another 3000 milliseconds
05-15 17:19:27.216 29726-29790/com.app.dealer_app D/BluetoothAdapter: isLeEnabled(): ON
05-15 17:19:27.221 29726-29726/com.app.dealer_app D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{6158fab: android.os.BinderProxy@2367608}
05-15 17:19:27.221 29726-29726/com.app.dealer_app D/CycledLeScanner: Scan started
05-15 17:19:27.331 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.331   pjsua_core.c !Shutting down, flags=0...
05-15 17:19:27.332 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.332   pjsua_core.c  PJSUA state changed: RUNNING --> CLOSING
05-15 17:19:27.343 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.343   pjsua_call.c  .Hangup all calls..
05-15 17:19:27.343 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.343  pjsua_media.c  .Call 0: deinitializing media..
05-15 17:19:27.343 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.343  pjsua_media.c  .Call 1: deinitializing media..
05-15 17:19:27.343 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.343  pjsua_media.c  .Call 2: deinitializing media..
05-15 17:19:27.343 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.343  pjsua_media.c  .Call 3: deinitializing media..
05-15 17:19:27.343 29726-29726/com.app.dealer_app I/PJSIP: 17:19:27.343   pjsua_pres.c  .Shutting down presence..
05-15 17:19:28.349 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.348   pjsua_core.c  .Destroying...
05-15 17:19:28.349 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.349  pjsua_media.c  .Shutting down media..
05-15 17:19:28.813 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.813   opensl_dev.c  ..OpenSL sound library shutting down..
05-15 17:19:28.814 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.814 android_jni_de  ..Android JNI sound library shutting down..
05-15 17:19:28.814 29726-29726/com.app.dealer_app D/PJSIP: 17:19:28.814 sip_endpoint.c  .Destroying endpoing instance..
05-15 17:19:28.814 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.814 sip_transactio  .Stopping transaction layer module
05-15 17:19:28.814 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.814 sip_transactio  .Stopped transaction layer module
05-15 17:19:28.814 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.814 sip_endpoint.c  .Module "mod-unsolicited-mwi" unregistered
05-15 17:19:28.814 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.814 sip_endpoint.c  .Module "mod-pjsua-options" unregistered
05-15 17:19:28.814 29726-29726/com.app.dealer_app I/PJSIP: 17:19:28.814 sip_endpoint.c  .Module "mod-pjsua-im" unregistered
[... cut ...]
05-15 17:20:55.211 29726-29726/com.app.dealer_app D/CycledLeScanner: We are already scanning
05-15 17:20:55.211 29726-29726/com.app.dealer_app D/CycledLeScanner: Waiting to stop scan cycle for another 3000 milliseconds
05-15 17:20:55.213 29726-29790/com.app.dealer_app D/BluetoothAdapter: isLeEnabled(): ON
05-15 17:20:55.218 29726-29726/com.app.dealer_app D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{6158fab: android.os.BinderProxy@2367608}
05-15 17:20:55.218 29726-29726/com.app.dealer_app D/CycledLeScanner: Scan started

Please let me know for solution.

roshanparmar, 2017-05-15:
Now I am not able to do anything because of this issue. I don't know what is the problem in this but its continously entering and exiting from beacon in Nexus 5x. It is not staying more than few seconds. Sometimes It does not enter for a long time.

Please provide the feasible solution ASAP.

davidgyoung, 2017-05-15:
Sorry you are having trouble, @roshanparmar, and thank you for attaching the log.  

The log shows no detections in the background at all, but indicates that beacons had been previously detected, perhaps in the foreground.  A. few possible causes:

1. The beacon you are using may not be matching the Bluetooth filter used in the background.  This would happen if it uses a non-addictive manufacturer ID.  If you can capture a short log segment showing a successful detection (perhaps in the foreground) I could confirm and offer a solution.

2.  The custom background scan periods may be interfering with the operating system's standard low power background scanning cycle.  To confirm if this is the problem, you could try setting the background scan periods back to the defaults for testing purposes to see if it then detects:

```
      beaconManager.setBackgroundScanPeriod(10000L);
        beaconManager.setBackgroundBetweenScanPeriod(300000L);
```

It may be that setting a 1000 ms scan cycle may be too short to ever allow low power detections to succeed.

3. There is a related bug discovered over the weekend in 2.10 (see #506) that is now fixed in 2.11-beta1.   Simply using this release may resolved the issue. https://github.com/AltBeacon/android-beacon-library/releases/tag/2.11-beta1

Please try the items above and let me know your findings so I may assist further.

roshanparmar, 2017-05-15:
Really thanks for your response.

I have tried your solutions and giving the results below - 
For 1. 
I am attaching logs below for foreground successful scan - 

05-15 19:17:39.515 14489-14489/? D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
05-15 19:17:39.517 14489-14489/? D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
05-15 19:17:39.521 14489-14489/? D/BeaconManager: This consumer is not bound.  binding: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@b792985
05-15 19:17:39.524 14489-14489/? D/BeaconManager: consumer count is now: 1
05-15 19:17:39.527 14489-14489/? D/AppStarter: Waiting for BeaconService connection
05-15 19:17:39.559 14489-14548/? D/ApplicationLoaders: ignored Vulkan layer search path /data/app/com.google.android.gms-2/lib/arm64:/system/fake-libs64:/data/app/com.google.android.gms-2/base.apk!/lib/arm64-v8a for namespace 0x76648ff0f0
05-15 19:17:40.677 14489-14587/com.app.dealer_app D/NetworkSecurityConfig: No Network Security Config specified, using platform default
05-15 19:17:40.730 14489-14489/com.app.dealer_app I/BeaconService: beaconService version 2.10 is starting up
05-15 19:17:40.731 14489-14489/com.app.dealer_app D/BluetoothCrashResolver: constructed
05-15 19:17:40.731 14489-14489/com.app.dealer_app W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
05-15 19:17:40.732 14489-14489/com.app.dealer_app D/BluetoothCrashResolver: Read 0 Bluetooth addresses
05-15 19:17:40.734 14489-14489/com.app.dealer_app D/BluetoothCrashResolver: started listening for BluetoothAdapter events
05-15 19:17:40.735 14489-14489/com.app.dealer_app I/CycledLeScanner: This Android 5.0.  We are using new scanning APIs
05-15 19:17:40.775 14489-14489/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47F, Nexus 5X, LGE
05-15 19:17:40.775 14489-14489/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.775 14489-14489/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 0 for Motorola;XT1092;MPE24.49-18;6.0 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/ModelSpecificDistanceCalculator: found a match with score 1
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, LPV79, Nexus 5, LGE
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47F, Nexus 5X, LGE
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 0 for Motorola;XT1092;MPE24.49-18;6.0 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/ModelSpecificDistanceCalculator: found a match with score 1
05-15 19:17:40.776 14489-14489/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, LPV79, Nexus 5, LGE
05-15 19:17:40.777 14489-14489/com.app.dealer_app D/BeaconService: No org.altbeacon.beacon.SimulatedScanData class exists.
05-15 19:17:40.778 14489-14489/com.app.dealer_app I/BeaconService: starting with intent Intent { cmp=com.app.dealer_app/org.altbeacon.beacon.service.BeaconService }
05-15 19:17:40.778 14489-14489/com.app.dealer_app I/BeaconService: binding
05-15 19:17:40.801 14489-14600/com.app.dealer_app I/Adreno: QUALCOMM build                   : 6818200, Idb2b4cb785
                                                            Build Date                       : 11/18/16
                                                            OpenGL ES Shader Compiler Version: XE031.09.00.04
                                                            Local Branch                     : N25
                                                            Remote Branch                    : 
                                                            Remote Branch                    : 
                                                            Reconstruct Branch               : 
05-15 19:17:40.807 14489-14600/com.app.dealer_app I/OpenGLRenderer: Initialized EGL, version 1.4
05-15 19:17:40.807 14489-14600/com.app.dealer_app D/OpenGLRenderer: Swap behavior 1
05-15 19:17:40.813 14489-14489/com.app.dealer_app D/BeaconManager: we have a connection to the service now
[... cut ...]
As you mentioned, I have set scanning time as :
 beaconManager.setBackgroundScanPeriod(10000L);
        beaconManager.setBackgroundBetweenScanPeriod(300000L);

Still it have some issue but yes not worst as previous experience but need improvements.

For 3 - 

I have tried to update 2.10 to 2.11-beta1 but its showing me "Failed to resolve" for 2.11-beta1 library.
I am using compile dependancy in Android studio to use altbeacon.

Thanks again for your precious time. I need your help again to resolve this issue.
Please come to me ASAP.

davidgyoung, 2017-05-15:
@roshanparmar, I believe I see the issue from the latest log.  See this:

```
05-15 19:17:52.050 14489-14528/com.app.dealer_app W/Binder: Binder call failed.
java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
at android.os.Parcel.readException(Parcel.java:1684)
at android.os.Parcel.readException(Parcel.java:1637)
at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:678)
at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:367)
at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
at android.os.Binder.execTransact(Binder.java:565)
```

That indicates the app has not gotten location permission needed to detect BLE devices.  You can read more about how to do this here:

http://altbeacon.github.io/android-beacon-library/requesting_permission.html

On Android 6+, apps that target SDK 23+ but fail to get this permission can not scan for BLE devices at all.  Apps that target SDKs less than 23 and fail to get this permission can only scan in the foreground.  I believe this is the problem that is causing your failure to scan in the background.

roshanparmar, 2017-05-16:
My app have already have permission for Location. The log which is showing by you was just before I didn't give the permission but after that i give the permission to app. But unfortunately, the same behavior exists. App does not stay in region more than few seconds and after that continous entering and exiting happens.

Location permission is not an issue, I give it already before start to scan for my work for beacon. Not an improvement again.

Please let me know to resolve this issue.

roshanparmar, 2017-05-16:
I am again attaching logs for issue, app is frequently entering and exiting beacon - 

05-16 09:50:53.993 29417-18280/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:50:53.994 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 95 lines
05-16 09:51:19.429 29417-18286/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:51:19.431 29417-18287/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:51:32.740 29417-18289/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:51:37.431 29417-29942/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-4 expire 14 lines
05-16 09:51:41.285 29417-18290/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:51:52.867 29417-18294/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:51:57.412 29417-30061/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-6 expire 14 lines
05-16 09:51:57.421 29417-18295/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:52:02.356 29417-30073/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-7 expire 12 lines
05-16 09:52:07.279 29417-29920/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-2 expire 14 lines
05-16 09:52:08.289 29417-18300/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:52:08.332 29417-18301/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:52:20.209 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 52 lines
05-16 09:52:33.866 29417-18307/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:52:37.601 29417-29516/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-1 expire 16 lines
05-16 09:52:37.610 29417-18308/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:52:42.564 29417-18310/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:52:42.565 29417-29929/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-3 expire 13 lines
05-16 09:52:47.519 29417-29997/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-5 expire 12 lines
05-16 09:52:48.530 29417-29459/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) CycledLeScanner expire 10 lines
05-16 09:53:28.108 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 47 lines
05-16 09:53:28.121 29417-18314/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:53:29.569 29417-18315/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:53:32.612 29417-18321/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:53:43.329 29417-18324/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:53:52.795 29417-18326/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:53:53.796 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 40 lines
05-16 09:53:57.424 29417-18328/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:54:02.667 29417-18329/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:54:09.409 29417-18331/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:54:39.264 29417-18337/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:54:39.270 29417-9168/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) Binder:29417_4 expire 10 lines
05-16 09:54:40.265 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 12 lines
05-16 09:54:42.433 29417-18338/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:54:49.378 29417-18339/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:54:50.623 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 33 lines
05-16 09:55:17.648 29417-18347/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:55:22.549 29417-18350/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:55:22.552 29417-29929/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-3 expire 17 lines
05-16 09:55:23.020 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 18 lines
05-16 09:55:27.491 29417-18352/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:55:42.588 29417-29417/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 36 lines
05-16 09:55:51.480 29417-18363/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
05-16 09:55:52.629 29417-18368/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:56:07.715 29417-29942/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) pool-3-thread-4 expire 16 lines
05-16 09:56:07.726 29417-18371/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 4 lines
05-16 09:56:08.154 29417-18372/com.app.dealer_app I/chatty: uid=10517(com.app.dealer_app) expire 1 line
[... cut ...]
05-16 10:32:03.845 29417-30061/com.app.dealer_app D/BeaconParser: Processing pdu type FF: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e609eab80509424b4f4e14160a1805be0b709ae993d6f6ee04000003e609ea0000000000 with startIndex: 5, endIndex: 29
05-16 10:32:03.846 29417-30061/com.app.dealer_app D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
05-16 10:32:03.846 29417-30061/com.app.dealer_app D/BeaconParser: Bytes are: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e609e

roshanparmar, 2017-05-16:
Is it compalsury to bind beacon service as beaconmanager.bind(this); and implement onBeaconServiecConnected() method ?

roshanparmar, 2017-05-16:
Why 2.11-beta1 version showing me "failed to resolve" in Andoid studio dependancy ?

davidgyoung, 2017-05-16:
2.11-beta1 is not on JCenter.  You must download the binary manuallly, but it in a libs folder in your project and reference it that way.  I will put up instructions in a bit and then review your latest logs.

roshanparmar, 2017-05-16:
yes please..I am waiting for your solutions.

davidgyoung, 2017-05-16:
Beta release updated with configuration instructions:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.11-beta1

roshanparmar, 2017-05-16:
ok. I am applying 2.11-beta1 as per your directions. And please look logs which I have sent you earlier and response your thoughts.

Thank you to helping me out.

davidgyoung, 2017-05-16:
@roshanparmar, a few more thoughts:

* If using `RegionBootstrap` as shown in the code posted in this issue, you should not call `beaconManager.bind(this)`, as it is done automatically inside `RegionBootstrap`.
* Based on reviewing the latest log above, I do believe the 2.11-beta1 release will solve the region exit issue you are seeing.    The log line below indicates that beacons were only detected in the first few seconds of the background scan cycle, which is the symptom fixed in 2.11-beta1.

`05-16 10:31:50.247 29417-29417/com.app.dealer_app D/RegionMonitoringState: We are newly outside the region because the lastSeenTime of 84863358 was 16693 seconds ago, and that is over the expiration duration of 10000`

roshanparmar, 2017-05-16:
I am not able to add 2.11-beta1 as a aar file.
It is giving me error like - 
Error:(63, 0) Supplied String module notation 'android-beacon-library-2.11-beta1' is invalid. Example notations: 'org.gradle:gradle-core:2.2', 'org.mockito:mockito-core:1.9.5:javadoc'.
<a href="openFile:/home/navin/Pictures/Aashus_Data/Projects/Dealer_App/code/dealer_app/app/build.gradle">Open File</a>

I have followed steps as link you provided - 
https://github.com/AltBeacon/android-beacon-library/releases/tag/2.11-beta1

roshanparmar, 2017-05-16:
Still Its continously entering and exiting in Google pixel.

davidgyoung, 2017-05-16:
Sorry, the string in the instructions was incorrect.  Please replace the dash before the version number with a colon like this: 'android-beacon-library:2.11-beta1@aar'

davidgyoung, 2017-05-16:
If you can get this configured, please verify from the logs that you are using this version by looking for this line:

`I/BeaconService: beaconService version 2.11-beta1 is starting up on the main process`

roshanparmar, 2017-05-16:
Thanks for your feedback.
yes,
the line "I/BeaconService: beaconService version 2.11-beta1 is starting up on the main process" is same as you said.

But Google pixel still giving me issue.
Steps to produce - 
1. open our app
2. start scan for beacon
3. when comes in range then put the app in background and open another app
4. then come to our app, after that exit will be called automatically.

above scenario comes to my client continously.

please verify at your end and provide some solutions.

davidgyoung, 2017-05-16:
@roshanparmar, can you please attache a log showing the prior 30 seconds before the unexpected region exit when running 2.11-beta1?

roshanparmar, 2017-05-16:
I am not able to give you logs of Google pixel because My client has that device and he is not near to me.
Can we have another way to check it ?

davidgyoung, 2017-05-16:
Troubleshooting errors on client devices where you don't have physical access is always really hard.  The best thing I can suggest is to ask you client to send you a bug report, but this requires your client to turn on Developer tools on the Android device.

If developer tools are enabled, you can send a bug report by going to Settings -> Developer Options -> Take Bug Report.

After tapping this, there will be a delay of a few seconds to a few minutes (!) until it is ready at which time a notification will be sent.  Tapping that notification will allow you to send the bug report via email.

The bug report includes recent logs as well as other state info about the Android device.

The trickiest part of doing this is getting your client to capture the bug report at the right time.  If this is done too late, the logs won't show the desired information.

roshanparmar, 2017-05-17:
Looking really helpful this. I am telling the client to do so.
When I will get bug repost, I will show to you.

Thanks

roshanparmar, 2017-05-18:
_When I use library 2.10 then entering and exiting frequently occur, attaching logs below -_ 

00:00 D/NetworkSecurityConfig: No Network Security Config specified, using platform default
00:00 D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
00:00 D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
00:00 D/BeaconManager: This consumer is not bound.  binding: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@5b8043d
00:00 D/BeaconManager: consumer count is now: 1
00:00 D/AppStarter: Waiting for BeaconService connection
00:00 D/ApplicationLoaders: ignored Vulkan layer search path /data/app/com.google.android.gms-2/lib/arm64:/system/fake-libs64:/data/app/com.google.android.gms-2/base.apk!/lib/arm64-v8a for namespace 0x76648ff0f0
00:01 I/BeaconService: beaconService version 2.10 is starting up
00:01 D/BluetoothCrashResolver: constructed
00:01 W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
00:01 D/BluetoothCrashResolver: Read 0 Bluetooth addresses
00:01 D/BluetoothCrashResolver: started listening for BluetoothAdapter events
00:01 I/CycledLeScanner: This Android 5.0.  We are using new scanning APIs
00:01 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47F, Nexus 5X, LGE
00:01 D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 0 for Motorola;XT1092;MPE24.49-18;6.0 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/ModelSpecificDistanceCalculator: found a match with score 1
00:01 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, KOT49H, Nexus 4, LGE
00:01 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47F, Nexus 5X, LGE
00:01 D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 0 for Motorola;XT1092;MPE24.49-18;6.0 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G47F;7.1.2
00:01 D/ModelSpecificDistanceCalculator: found a match with score 1
00:01 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, KOT49H, Nexus 4, LGE
00:01 D/BeaconService: No org.altbeacon.beacon.SimulatedScanData class exists.
00:01 I/BeaconService: starting with intent Intent { cmp=com.app.dealer_app/org.altbeacon.beacon.service.BeaconService }
00:01 I/BeaconService: binding
00:01 I/Adreno: QUALCOMM build                   : 6818200, Idb2b4cb785
00:01 I/Adreno: Build Date                       : 11/18/16
00:01 I/Adreno: OpenGL ES Shader Compiler Version: XE031.09.00.04
00:01 I/Adreno: Local Branch                     : N25
00:01 I/Adreno: Remote Branch                    : 
00:01 I/Adreno: Remote Branch                    : 
00:01 I/Adreno: Reconstruct Branch               : 
00:01 I/OpenGLRenderer: Initialized EGL, version 1.4
00:01 D/OpenGLRenderer: Swap behavior 1
00:01 D/BeaconManager: we have a connection to the service now
00:01 D/AppStarter: Activating background region monitoring
00:01 D/AppStarter: Background region monitoring activated for region id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null
00:01 D/BeaconManager: Starting monitoring region id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null with uniqueID: backgroundRegion
00:01 D/BeaconManager: callback packageName: com.app.dealer_app
00:01 D/MonitoringStatus: Not restoring monitoring state because it was recorded too many milliseconds ago: 1495084705500
00:01 D/BeaconManager: updating background flag to true
00:01 D/BeaconManager: updating scan period to 10000, 30000
00:01 I/BeaconService: start monitoring received
00:01 D/FA: Connected to remote service
00:01 D/BeaconService: startMonitoring called
00:01 V/FA: Processing queued up service tasks: 2
00:01 D/MonitoringStatus: saveMonitoringStatusIfOn()
00:01 D/BeaconService: Currently monitoring 1 regions.
00:01 D/CycledLeScanner: start called
00:01 D/CycledLeScanner: starting a new scan cycle
00:01 D/CycledLeScanner: starting a new bluetooth le scan
00:01 D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
00:01 D/CycledLeScanner: Scan started
00:01 D/CycledLeScanner: Set scan periods called with 3000, 1000 Background mode must have changed.
00:01 D/CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
00:01 D/CycledLeScanner: cancel wakeup alarm: null
00:01 D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036596099785 ms: PendingIntent{f0e9175: android.os.BinderProxy@f03c50a}
00:01 I/BeaconService: set scan intervals received
00:01 D/CycledLeScanner: Set scan periods called with 10000, 30000 Background mode must have changed.
00:01 D/CycledLeScanner: We are in the background.  Setting wakeup alarm
[... cut ...]
06:09 D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
06:09 D/CycledLeScanner: Normalizing between scan period from 1000 to 987
06:09 D/CycledLeScannerForLollipop: This is Android L. Doing a filtered scan for the background.
06:09 D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
06:09 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 987 milliseconds
06:09 D/BluetoothAdapter: isLeEnabled(): ON
06:09 D/BeaconIntentProcessor: got an intent to pr

roshanparmar, 2017-05-18:
More remaining logs of above comment - 

08:00 D/CycledLeScanner: Waiting to stop scan cycle for another 998 milliseconds
08:01 D/CycledLeScanner: Done with scan cycle
08:01 D/BeaconService: Calling ranging callback
08:01 D/RunningAverageRssiFilter: Running average mRssi based on 2 measurements: -41.0
08:01 D/RangedBeacon: calculated new runningAverageRssi: -41.0
08:01 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
08:01 D/CurveFittedDistanceCalculator: calculating distance based on mRssi of -41.0 and txPower of -72
08:01 D/CurveFittedDistanceCalculator: avg mRssi: -41.0 distance: 0.00358520177597799
08:01 D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
08:01 D/CycledLeScanner: Normalizing between scan period from 1000 to 977
08:01 D/CycledLeScannerForLollipop: This is Android L, but we last saw a beacon only 3898 ago, so we will not keep scanning in background.
08:01 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 977 milliseconds
08:01 D/BeaconIntentProcessor: got an intent to process
08:01 D/BeaconIntentProcessor: got ranging data
08:02 D/CycledLeScanner: starting a new scan cycle
08:02 D/CycledLeScanner: We are already scanning
08:02 D/CycledLeScanner: Waiting to stop scan cycle for another 3000 milliseconds
08:02 D/CycledLeScanner: Scan started
08:03 D/CycledLeScanner: Waiting to stop scan cycle for another 1999 milliseconds
08:04 D/CycledLeScanner: Waiting to stop scan cycle for another 998 milliseconds
08:05 D/CycledLeScanner: Done with scan cycle
08:05 D/BeaconService: Calling ranging callback
08:05 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
08:05 D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
08:05 D/CycledLeScanner: Normalizing between scan period from 1000 to 978
08:05 D/CycledLeScannerForLollipop: This is Android L, but we last saw a beacon only 7896 ago, so we will not keep scanning in background.
08:05 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 978 milliseconds
08:05 D/BeaconIntentProcessor: got an intent to process
08:05 D/BeaconIntentProcessor: got ranging data
08:06 D/CycledLeScanner: starting a new scan cycle
08:06 D/CycledLeScanner: We are already scanning
08:06 D/CycledLeScanner: Waiting to stop scan cycle for another 3000 milliseconds
08:06 D/CycledLeScanner: Scan started
08:07 D/CycledLeScanner: Waiting to stop scan cycle for another 1999 milliseconds
08:08 D/CycledLeScanner: Waiting to stop scan cycle for another 998 milliseconds
08:09 D/CycledLeScanner: Done with scan cycle
08:09 D/RegionMonitoringState: We are newly outside the region because the lastSeenTime of 259152126 was 11876 seconds ago, and that is over the expiration duration of 10000
08:09 D/MonitoringStatus: found a monitor that expired: id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null
08:09 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
08:09 D/MonitoringStatus: saveMonitoringStatusIfOn()
08:09 D/BeaconService: Calling ranging callback
08:09 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
08:09 D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
08:09 D/CycledLeScanner: Normalizing between scan period from 1000 to 966
08:09 D/CycledLeScannerForLollipop: This is Android L. Doing a filtered scan for the background.
08:09 D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
08:09 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 966 milliseconds
08:09 D/BluetoothAdapter: isLeEnabled(): ON
08:09 D/BeaconIntentProcessor: got an intent to process
08:09 D/BeaconIntentProcessor: got monitoring data
08:09 D/BeaconIntentProcessor: Calling monitoring notifier: com.app.dealer_app.DealerApplication@f407f70
08:09 I/DealerApplication: !!!!SSSSSS did exit region..null
08:09 D/BeaconIntentProcessor: got an intent to process
08:09 D/BeaconIntentProcessor: got ranging data
08:09 D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=6 mClientIf=0
08:09 D/CycledLeScannerForLollipop: got record
08:09 D/CycledLeScannerForLollipop: got a filtered scan result in the background.
08:09 D/BeaconParser: Ignoring pdu type 01
08:09 D/BeaconParser: Processing pdu type FF: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e609eab80509424b4f4e14160a1805be0b709ae993d6f6ee04000003e609ea0000000000 with startIndex: 5, endIndex: 29
08:09 D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
08:09 D/BeaconParser: Bytes are: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e609eab80509424b4f4e14160a1805be0b709ae993d6f6ee04000003e609ea0000000000
08:09 D/BeaconService: beacon detected : id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: 998 id3: 2538
08:09 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
08:09 D/MonitoringStatus: saveMonitoringStatusIfOn()
08:09 D/BeaconService: looking for ranging region matches for this beacon
08:09 D/BeaconService: matches ranging region: id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null
[... cut ...]
14:41 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
14:41 D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
14:41 D/CycledLeScanner: Normalizing between scan period from 1000 to 990
14:41 D/CycledLeScannerForLollipop: This is Android L, but we last saw a beacon only 7721 ago, so we will not keep scanning in background.
14:41 D/CycledLeScannerForLollipop: Waiting to start f

roshanparmar, 2017-05-18:
**I will provide you logs for Google pixel as well ASAP but before that please look into Google pixel that is any with that ?

Do you think that this issue is only device specific or Android version specific or any other thing that we didn't seen till now ?**

roshanparmar, 2017-05-18:
When I have tested it on Samsung s6 edge with 7.0 O.S. then its again giving me same behavior means when app goes to background and again comes to background then didExitBeacon called.

Below I am attaching logs of Smasun s6 edge - 
00:00 D/SensorManager: registerListener :: 0, MPU6500 Acceleration Sensor, 200000, 0,  
00:00 D/NetworkSecurityConfig: No Network Security Config specified, using platform default
00:00 I/System.out: Thread-324(ApacheHTTPLog):Reading from variable values from setDefaultValuesToVariables
00:00 I/System.out: Thread-324(ApacheHTTPLog):isSBSettingEnabled false
00:00 I/System.out: Thread-324(ApacheHTTPLog):isShipBuild true
00:00 I/System.out: Thread-324(ApacheHTTPLog):getDebugLevel 0x4f4c
00:00 I/System.out: Thread-324(ApacheHTTPLog):Smart Bonding Setting is false
00:00 I/System.out: Thread-324(ApacheHTTPLog):SmartBonding Setting is false, SHIP_BUILD is true, log to file is false, DBG is false, DEBUG_LEVEL (1-LOW, 2-MID, 3-HIGH) is 1
00:00 I/CrashlyticsCore: Initializing Crashlytics 2.3.16.dev
00:00 I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
00:00 I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
00:00 I/BeaconManager: BeaconManager started up on pid 15378 named 'com.app.dealer_app' for application package 'com.app.dealer_app'.  isMainProcess=true
00:00 D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
00:00 D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
00:00 D/BeaconManager: This consumer is not bound.  binding: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@f1bd8aa
00:00 D/BeaconManager: consumer count is now: 1
00:00 D/AppStarter: Waiting for BeaconService connection
00:00 D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
00:00 D/BluetoothCrashResolver: constructed
00:00 W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
00:00 D/BluetoothCrashResolver: Read 0 Bluetooth addresses
00:00 D/BluetoothCrashResolver: started listening for BluetoothAdapter events
00:00 I/CycledLeScanner: This Android 5.0.  We are using new scanning APIs
00:00 I/BeaconService: beaconService version 2.11-beta1 is starting up on the main process
00:00 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.0, NRD90M, SM-G925I, samsung
00:00 D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to samsung;SM-G925I;NRD90M;7.0
00:00 D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to samsung;SM-G925I;NRD90M;7.0
00:00 D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to samsung;SM-G925I;NRD90M;7.0
00:00 W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
00:00 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.0, NRD90M, SM-G925I, samsung
00:00 D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to samsung;SM-G925I;NRD90M;7.0
00:00 D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to samsung;SM-G925I;NRD90M;7.0
00:00 D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to samsung;SM-G925I;NRD90M;7.0
00:00 W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
00:00 D/BeaconService: No org.altbeacon.beacon.SimulatedScanData class exists.
00:00 I/BeaconService: starting with intent Intent { cmp=com.app.dealer_app/org.altbeacon.beacon.service.BeaconService launchParam=MultiScreenLaunchParams { mDisplayId=0 mFlags=0 } }
00:00 I/BeaconService: binding
00:00 D/BeaconManager: we have a connection to the service now
00:00 D/BeaconManager: Not synchronizing settings to service, as it is in the same process
00:00 D/AppStarter: Activating background region monitoring
00:00 D/AppStarter: Background region monitoring activated for region id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null
00:00 D/BeaconManager: Starting monitoring region id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null with uniqueID: backgroundRegion
00:00 D/BeaconManager: callback packageName: com.app.dealer_app
00:00 D/MonitoringStatus: Not restoring monitoring state because it was recorded too many milliseconds ago: 951916
00:00 D/BeaconManager: updating background flag to true
00:00 D/BeaconManager: updating scan period to 10000, 300000
00:00 I/BeaconService: start monitoring received
00:00 D/BeaconService: startMonitoring called
00:00 D/MonitoringStatus: saveMonitoringStatusIfOn()
00:00 D/BeaconService: Currently monitoring 1 regions.
00:00 D/CycledLeScanner: start called
00:00 D/CycledLeScanner: starting a new scan cycle
00:00 D/CycledLeScanner: starting a new bluetooth le scan
00:00 D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
00:00 D/CycledLeScannerForLollipop: Making new Android L scanner
00:00 D/BluetoothAdapter: STATE_ON
00:00 D/BluetoothAdapter: STATE_ON
00:00 D/BluetoothLeScanner: Start Scan
00:00 D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
00:00 D/CycledLeScanner: Scan started
00:00 D/CycledLeScanner: Set scan periods called with 1100, 0 Background mode must have changed.
00:00 D/CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
00:00 D/CycledLeScanner: cancel wakeup alarm: null
00:00 D/BluetoothAdapter: STATE_ON
00:00 D/BluetoothAdapter: STATE_ON
00:00 D/BluetoothAdapter: STATE_ON
00:00 D/BluetoothAdapter: STATE_ON
[... 1081 lines cut ...]
[... cut ...]
01:02 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
01:02 D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
01:02 D/CycledLeScanner: starting a new scan cycle
01:02 D/CycledLeScanner: We are already scanning
01:02 D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
01:02 D/CycledLeScanner: Scan started
01:02 D/BeaconIntentProcessor: got an intent to process
01:02 D/BeaconIntentProcessor: got ranging data

davidgyoung, 2017-05-18:
Yes, as discussed above, this is a known issue with 2.10 for an app in the background when beacons are continuously around.  The fix is included in 2.11-beta1.

roshanparmar, 2017-05-18:
Yes thats known but I am using 2.11-beta1 however I am getting this issue.

davidgyoung, 2017-05-18:
@roshanparmar, the logs indicate 2.10 is active:

`05-15 19:17:40.730 14489-14489/com.app.dealer_app I/BeaconService: beaconService version 2.10 is starting up`

roshanparmar, 2017-05-18:
ok.
But I am having a problem that when I called startMonitoring.. method then it tells me to bind service however I am using RegionBootstrap.

and you told me that both are alternatives.

roshanparmar, 2017-05-18:
I am getting below message - 
The BeaconManager is not bound to the service.  Call beaconManager.bind(BeaconConsumer consumer) and wait for a callback to onBeaconServiceConnect()

roshanparmar, 2017-05-18:
I am not able to listen beacon in Samsung s6 edge - same previous code but now not running.
Here are logs - 

05-18 08:18:09.821 16668-16668/? E/Zygote: v2
05-18 08:18:09.821 16668-16668/? I/libpersona: KNOX_SDCARD checking this for 10474
05-18 08:18:09.821 16668-16668/? I/libpersona: KNOX_SDCARD not a persona
05-18 08:18:09.822 16668-16668/? E/Zygote: accessInfo : 0
05-18 08:18:09.822 16668-16668/? W/SELinux: SELinux selinux_android_compute_policy_index : Policy Index[2],  Con:u:r:zygote:s0 RAM:SEPF_SECMOBILE_7.0_0005, [-1 -1 -1 -1 0 1]
05-18 08:18:09.824 16668-16668/? I/SELinux: SELinux: seapp_context_lookup: seinfo=untrusted, level=s0:c512,c768, pkgname=com.app.dealer_app 
05-18 08:18:09.827 16668-16668/? I/art: Late-enabling -Xcheck:jni
05-18 08:18:09.840 16668-16668/? D/TimaKeyStoreProvider: TimaKeyStore is not enabled: cannot add TimaSignature Service and generateKeyPair Service
05-18 08:18:09.843 16668-16675/? E/art: Failed sending reply to debugger: Broken pipe
05-18 08:18:09.843 16668-16675/? I/art: Debugger is no longer active
05-18 08:18:09.843 16668-16675/? I/art: Starting a blocking GC Instrumentation
05-18 08:18:09.887 16668-16668/? D/FirebaseApp: com.google.firebase.auth.FirebaseAuth is not linked. Skipping initialization.
05-18 08:18:09.893 16668-16668/? W/InstanceID/Rpc: Found 10016
05-18 08:18:09.902 16668-16668/? D/FirebaseApp: com.google.firebase.crash.FirebaseCrash is not linked. Skipping initialization.
05-18 08:18:09.908 16668-16668/? I/FA: App measurement is starting up, version: 10298
05-18 08:18:09.908 16668-16668/? I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
05-18 08:18:09.916 16668-16668/? I/FA: To enable faster debug mode event logging run:
                                         adb shell setprop debug.firebase.analytics.app com.app.dealer_app
05-18 08:18:09.916 16668-16668/? D/FA: Debug-level message logging enabled
05-18 08:18:09.924 16668-16668/? I/FirebaseInitProvider: FirebaseApp initialization successful
05-18 08:18:09.983 16668-16668/? D/NetworkSecurityConfig: No Network Security Config specified, using platform default
05-18 08:18:09.990 16668-16721/? I/System.out: Thread-1436(ApacheHTTPLog):Reading from variable values from setDefaultValuesToVariables
05-18 08:18:09.991 16668-16721/? I/System.out: Thread-1436(ApacheHTTPLog):isSBSettingEnabled false
05-18 08:18:09.991 16668-16721/? I/System.out: Thread-1436(ApacheHTTPLog):isShipBuild true
05-18 08:18:09.991 16668-16721/? I/System.out: Thread-1436(ApacheHTTPLog):getDebugLevel 0x4f4c
05-18 08:18:09.991 16668-16721/? I/System.out: Thread-1436(ApacheHTTPLog):Smart Bonding Setting is false
05-18 08:18:09.991 16668-16721/? I/System.out: Thread-1436(ApacheHTTPLog):SmartBonding Setting is false, SHIP_BUILD is true, log to file is false, DBG is false, DEBUG_LEVEL (1-LOW, 2-MID, 3-HIGH) is 1
05-18 08:18:10.010 16668-16668/? I/CrashlyticsCore: Initializing Crashlytics 2.3.16.dev
05-18 08:18:10.026 16668-16668/? I/BeaconManager: BeaconManager started up on pid 16668 named 'com.app.dealer_app' for application package 'com.app.dealer_app'.  isMainProcess=true
05-18 08:18:10.028 16668-16668/? D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
05-18 08:18:10.029 16668-16668/? D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
05-18 08:18:10.033 16668-16668/? D/BeaconManager: This consumer is not bound.  binding: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@ab17ed9
05-18 08:18:10.045 16668-16668/? D/BeaconManager: consumer count is now: 1
05-18 08:18:10.045 16668-16668/? D/AppStarter: Waiting for BeaconService connection
05-18 08:18:11.078 16668-16668/com.app.dealer_app D/TESTFAIRYSDK: onActivityResumed com.app.dealer_app.activity.SplashActivity
05-18 08:18:11.083 16668-16764/com.app.dealer_app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
05-18 08:18:11.083 16668-16764/com.app.dealer_app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
05-18 08:18:11.088 16668-16668/com.app.dealer_app D/ViewRootImpl@1ee3298[SplashActivity]: ThreadedRenderer.create() translucent=false
05-18 08:18:11.090 16668-16668/com.app.dealer_app D/InputTransport: Input channel constructed: fd=68
05-18 08:18:11.090 16668-16668/com.app.dealer_app D/ViewRootImpl@1ee3298[SplashActivity]: setView = DecorView@f660af1[SplashActivity] touchMode=true
05-18 08:18:11.095 16668-16668/com.app.dealer_app D/BluetoothCrashResolver: constructed
05-18 08:18:11.095 16668-16668/com.app.dealer_app W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
05-18 08:18:11.095 16668-16668/com.app.dealer_app D/BluetoothCrashResolver: Read 0 Bluetooth addresses
05-18 08:18:11.096 16668-16668/com.app.dealer_app D/BluetoothCrashResolver: started listening for BluetoothAdapter events
05-18 08:18:11.097 16668-16668/com.app.dealer_app I/CycledLeScanner: This Android 5.0.  We are using new scanning APIs
05-18 08:18:11.097 16668-16668/com.app.dealer_app I/BeaconService: beaconService version 2.11-beta1 is starting up on the main process
05-18 08:18:11.100 16668-16772/com.app.dealer_app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
05-18 08:18:11.100 16668-16772/com.app.dealer_app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
05-18 08:18:11.129 16668-16668/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.0, NRD90M, SM-G925I, samsung
[... cut ...]
05-18 08:19:01.395 16668-16675/com.app.dealer_app W/art: Suspending all threads took: 6.477ms
05-18 08:19:01.936 16668-16668/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 27873 milliseconds
05-18 08:19:02.938 16668-16668/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 26869 milliseconds
05-18 08:19:03.939 16668-16668/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 25868 milliseconds

roshanparmar, 2017-05-18:
Please help me out, not able to understand what is going on.
I am able to find beacon in LG Nexus 5x but not in Samsung s6 edge with 7. O.S.
Here are logs for samsung - 

05-18 08:28:51.916 31931-31931/? E/Zygote: v2
05-18 08:28:51.916 31931-31931/? I/libpersona: KNOX_SDCARD checking this for 10476
05-18 08:28:51.916 31931-31931/? I/libpersona: KNOX_SDCARD not a persona
05-18 08:28:51.917 31931-31931/? E/Zygote: accessInfo : 0
05-18 08:28:51.917 31931-31931/? W/SELinux: SELinux selinux_android_compute_policy_index : Policy Index[2],  Con:u:r:zygote:s0 RAM:SEPF_SECMOBILE_7.0_0005, [-1 -1 -1 -1 0 1]
05-18 08:28:51.918 31931-31931/? I/SELinux: SELinux: seapp_context_lookup: seinfo=untrusted, level=s0:c512,c768, pkgname=com.app.dealer_app 
05-18 08:28:51.920 31931-31931/? I/art: Late-enabling -Xcheck:jni
05-18 08:28:51.936 31931-31931/? D/TimaKeyStoreProvider: TimaKeyStore is not enabled: cannot add TimaSignature Service and generateKeyPair Service
05-18 08:28:51.999 31931-31931/com.app.dealer_app D/FirebaseApp: com.google.firebase.auth.FirebaseAuth is not linked. Skipping initialization.
05-18 08:28:52.002 31931-31931/com.app.dealer_app D/FirebaseApp: com.google.firebase.crash.FirebaseCrash is not linked. Skipping initialization.
05-18 08:28:52.009 31931-31931/com.app.dealer_app I/FA: App measurement is starting up, version: 10298
05-18 08:28:52.009 31931-31931/com.app.dealer_app I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
05-18 08:28:52.022 31931-31931/com.app.dealer_app I/FA: To enable faster debug mode event logging run:
                                                          adb shell setprop debug.firebase.analytics.app com.app.dealer_app
05-18 08:28:52.022 31931-31931/com.app.dealer_app D/FA: Debug-level message logging enabled
05-18 08:28:52.044 31931-31931/com.app.dealer_app I/FirebaseInitProvider: FirebaseApp initialization successful
05-18 08:28:52.090 31931-31931/com.app.dealer_app D/TESTFAIRYSDK: Cached options was found 'video,video-only-wifi,data-only-wifi,video-quality=medium,cpu,memory,logcat,anonymous,shake,anr,user-interactions'
05-18 08:28:52.094 31931-31931/com.app.dealer_app D/SensorManager: registerListener :: 0, MPU6500 Acceleration Sensor, 200000, 0,  
05-18 08:28:52.110 31931-31931/com.app.dealer_app D/NetworkSecurityConfig: No Network Security Config specified, using platform default
05-18 08:28:52.117 31931-31975/com.app.dealer_app I/System.out: Thread-1523(ApacheHTTPLog):Reading from variable values from setDefaultValuesToVariables
05-18 08:28:52.118 31931-31975/com.app.dealer_app I/System.out: Thread-1523(ApacheHTTPLog):isSBSettingEnabled false
05-18 08:28:52.118 31931-31975/com.app.dealer_app I/System.out: Thread-1523(ApacheHTTPLog):isShipBuild true
05-18 08:28:52.118 31931-31975/com.app.dealer_app I/System.out: Thread-1523(ApacheHTTPLog):getDebugLevel 0x4f4c
05-18 08:28:52.118 31931-31975/com.app.dealer_app I/System.out: Thread-1523(ApacheHTTPLog):Smart Bonding Setting is false
05-18 08:28:52.118 31931-31975/com.app.dealer_app I/System.out: Thread-1523(ApacheHTTPLog):SmartBonding Setting is false, SHIP_BUILD is true, log to file is false, DBG is false, DEBUG_LEVEL (1-LOW, 2-MID, 3-HIGH) is 1
05-18 08:28:52.137 31931-31931/com.app.dealer_app I/CrashlyticsCore: Initializing Crashlytics 2.3.16.dev
05-18 08:28:52.153 31931-31977/com.app.dealer_app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
05-18 08:28:52.154 31931-31977/com.app.dealer_app I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
05-18 08:28:52.158 31931-31931/com.app.dealer_app I/BeaconManager: BeaconManager started up on pid 31931 named 'com.app.dealer_app' for application package 'com.app.dealer_app'.  isMainProcess=true
05-18 08:28:52.160 31931-31931/com.app.dealer_app D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
05-18 08:28:52.161 31931-31931/com.app.dealer_app D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
05-18 08:28:52.163 31931-31931/com.app.dealer_app D/BeaconManager: This consumer is not bound.  binding: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@f9f334c
05-18 08:28:52.169 31931-31931/com.app.dealer_app D/BeaconManager: consumer count is now: 1
05-18 08:28:52.170 31931-31931/com.app.dealer_app D/AppStarter: Waiting for BeaconService connection
05-18 08:28:52.171 31931-31931/com.app.dealer_app D/DealerApplication: !!!!SSSSS did enter monitor..
05-18 08:28:52.172 31931-31931/com.app.dealer_app W/System.err: android.os.RemoteException: The BeaconManager is not bound to the service.  Call beaconManager.bind(BeaconConsumer consumer) and wait for a callback to onBeaconServiceConnect()
05-18 08:28:52.172 31931-31931/com.app.dealer_app W/System.err:     at org.altbeacon.beacon.BeaconManager.startMonitoringBeaconsInRegion(BeaconManager.java:825)
05-18 08:28:52.172 31931-31931/com.app.dealer_app W/System.err:     at com.app.dealer_app.DealerApplication.initializeBeaconManager(DealerApplication.java:292)
05-18 08:28:52.172 31931-31931/com.app.dealer_app W/System.err:     at com.app.dealer_app.DealerApplication.onCreate(DealerApplication.java:247)
05-18 08:28:52.172 31931-31931/com.app.dealer_app W/System.err:     at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1032)
05-18 08:28:52.172 31931-31931/com.app.dealer_app W/System.err:     at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5876)
[... cut ...]
05-18 08:30:14.846 31931-31931/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 34961 milliseconds
05-18 08:30:15.850 31931-31931/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 33959 milliseconds
05-18 08:30:16.853 31931-31931/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 32955 milliseconds
05-18 08:30:17.854 31931-31931/com.app.dealer_app D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 31953 milliseconds

roshanparmar, 2017-05-18:
Please response.

roshanparmar, 2017-05-18:
and here logs from Google pixel in which frequently exiting region - 

00:00 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 23259 milliseconds
00:00 E/Volley: [2402] BasicNetwork.performRequest: Request at https://app.callproof.com/hustle/check/token has been redirected to http://app.callproof.com/hustle/check/token/
00:01 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 22258 milliseconds
00:01 E/Volley: [2402] BasicNetwork.performRequest: Request at https://app.callproof.com/hustle/check/token has been redirected to https://app.callproof.com/hustle/check/token/
00:01 I/AppCompatViewInflater: app:theme is now deprecated. Please move to using android:theme instead.
00:02 D/BeaconManager: This consumer is not bound.  binding: com.app.dealer_app.activity.HomeActivity@523d95d
00:02 D/BeaconManager: consumer count is now: 4
00:02 I/ClockInFragment: nnnnnnn clock - in.. 
00:02 D/BeaconManager: updating background flag to false
00:02 D/BeaconManager: updating scan period to 1100, 0
00:02 D/BeaconManager: we have a connection to the service now
00:02 D/BeaconManager: Not synchronizing settings to service, as it is in the same process
00:02 I/DealerApplication: nnnnnn time matrtersFri Jan 02 11:04:42 EST 1970
00:02 I/System.out: true
00:02 D/BeaconManager: callback packageName: com.app.dealer_app
00:02 I/DealerApplication: SSSSSSSS SCHEDULE ALERT
00:02 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 21207 milliseconds
00:02 I/BeaconService: set scan intervals received
00:02 D/CycledLeScanner: Set scan periods called with 1100, 0 Background mode must have changed.
00:02 D/CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
00:02 D/CycledLeScanner: cancel wakeup alarm: PendingIntent{dba5860: android.os.BinderProxy@9851119}
00:02 D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036751886922 ms: PendingIntent{dba5860: android.os.BinderProxy@9851119}
00:02 I/CycledLeScanner: Adjusted nextScanStartTime to be Thu May 18 11:00:04 EDT 2017
00:02 I/BeaconService: start ranging received
00:02 I/BeaconService: Already ranging that region -- will replace existing region.
00:02 D/BeaconService: Currently ranging 1 regions.
00:02 D/CycledLeScanner: start called
00:02 D/CycledLeScanner: scanning already started
00:02 D/CycledLeScanner: Set scan periods called with 1100, 0 Background mode must have changed.
00:02 D/CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
00:02 D/CycledLeScanner: cancel wakeup alarm: PendingIntent{dba5860: android.os.BinderProxy@9851119}
00:02 D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036751886916 ms: PendingIntent{dba5860: android.os.BinderProxy@9851119}
00:02 D/ClockInFragment: !!!!!! User Details found.. {"status_code": 200, "status_message": "user details", "data": {"first_name": "Robert", "last_name": "Hartline", "email": "callproof@goabsolutewireless.com", "is_manager": true, "state_name": ""}}
00:02 I/ClockInFragment: !!!!! status .. {"status_code": 200, "is_call": true}
00:03 D/CycledLeScanner: starting a new scan cycle
00:03 D/CycledLeScanner: restarting a bluetooth le scan
00:03 D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
00:03 D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
00:03 D/CycledLeScanner: Scan started
00:03 D/BluetoothAdapter: isLeEnabled(): ON
00:03 D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5 mClientIf=0
00:03 D/CycledLeScannerForLollipop: got record
00:03 D/BeaconParser: Ignoring pdu type 01
00:03 D/BeaconParser: Processing pdu type FF: 02011a1aff4c0002150629af9103560cc62a93332c786ef0cc00000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
00:03 D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
00:03 D/BeaconParser: Bytes are: 02011a1aff4c0002150629af9103560cc62a93332c786ef0cc00000000000000000000000000000000000000000000000000000000000000000000000000
00:03 D/BeaconService: Beacon packet detected for: id1: 0629af91-0356-0cc6-2a93-332c786ef0cc id2: 0 id3: 0 with rssi -45
00:03 D/BeaconService: beacon detected : id1: 0629af91-0356-0cc6-2a93-332c786ef0cc id2: 0 id3: 0
00:03 D/MonitoringStatus: This region (id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null) does not match beacon: id1: 0629af91-0356-0cc6-2a93-332c786ef0cc id2: 0 id3: 0
00:03 D/BeaconService: looking for ranging region matches for this beacon
00:03 D/BeaconService: This region (id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null) does not match beacon: id1: 0629af91-0356-0cc6-2a93-332c786ef0cc id2: 0 id3: 0
00:03 I/PresenceService: SSSSSSSS SYNC START
00:03 D/CycledLeScannerForLollipop: got record
00:03 D/BeaconParser: Ignoring pdu type 01
00:03 D/BeaconParser: Processing pdu type FF: 02011a1aff4c0002150629af9103560cc62a93332c786ef0cc00000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 29
00:03 D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
00:03 D/BeaconParser: Bytes are: 02011a1aff4c0002150629af9103560cc62a93332c786ef0cc00000000000000000000000000000000000000000000000000000000000000000000000000
00:03 I/DealerApplication: ++++++ params sync {store_id=4807, hm=11:03:42, hm2=11:04:42, type=presence, enddate=2017-05-18, user_id=218, startdate=2017-05-18}
[... cut ...]
00:42 D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
00:42 D/CycledLeScanner: Normalizing between scan period from 300000 to 290551
00:42 D/CycledLeScannerForLollipop: This is Android L, but we last saw a beacon only 963 ago, so we will not keep scanning in background.
00:42 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 290551 milliseconds
00:42 D/BluetoothAdapter: isLeEnabled(): ON
00:42 D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{dba5860: android.os.BinderProxy@9851119}

roshanparmar, 2017-05-18:
More logs from Google 

pixel -

roshanparmar, 2017-05-18:
00:00 D/NetworkSecurityConfig: No Network Security Config specified, using platform default
00:00 I/CrashlyticsCore: Initializing Crashlytics 2.3.16.dev
00:00 I/BeaconManager: BeaconManager started up on pid 9474 named 'com.app.dealer_app' for application package 'com.app.dealer_app'.  isMainProcess=true
00:00 D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
00:00 D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
00:00 D/BeaconManager: This consumer is not bound.  binding: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@e664714
00:00 D/BeaconManager: consumer count is now: 1
00:00 D/AppStarter: Waiting for BeaconService connection
00:00 D/ApplicationLoaders: ignored Vulkan layer search path /data/app/com.google.android.gms-2/lib/arm64:/system/fake-libs64:/data/app/com.google.android.gms-2/base.apk!/lib/arm64-v8a for namespace 0x73e6ef90f0
00:01 D/BluetoothCrashResolver: constructed
00:01 W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
00:01 D/BluetoothCrashResolver: Read 0 Bluetooth addresses
00:01 D/BluetoothCrashResolver: started listening for BluetoothAdapter events
00:01 I/CycledLeScanner: This Android 5.0.  We are using new scanning APIs
00:01 I/BeaconService: beaconService version 2.11-beta1 is starting up on the main process
00:01 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47O, Pixel, Google
00:01 D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Google;Pixel;N2G47O;7.1.2
00:01 D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to Google;Pixel;N2G47O;7.1.2
00:01 D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Google;Pixel;N2G47O;7.1.2
00:01 W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
00:01 D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47O, Pixel, Google
00:01 D/AndroidModel: Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Google;Pixel;N2G47O;7.1.2
00:01 D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to Google;Pixel;N2G47O;7.1.2
00:01 D/AndroidModel: Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Google;Pixel;N2G47O;7.1.2
00:01 W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
00:01 D/BeaconService: No org.altbeacon.beacon.SimulatedScanData class exists.
00:01 I/BeaconService: starting with intent Intent { cmp=com.app.dealer_app/org.altbeacon.beacon.service.BeaconService }
00:01 I/BeaconService: binding
00:01 I/Adreno: QUALCOMM build                   : 6818200, Idb2b4cb785
00:01 I/Adreno: Build Date                       : 11/17/16
00:01 I/Adreno: OpenGL ES Shader Compiler Version: XE031.09.00.04
00:01 I/Adreno: Local Branch                     : N25
00:01 I/Adreno: Remote Branch                    : 
00:01 I/Adreno: Remote Branch                    : 
00:01 I/Adreno: Reconstruct Branch               : 
00:01 D/BeaconManager: we have a connection to the service now
00:01 D/BeaconManager: Not synchronizing settings to service, as it is in the same process
00:01 D/AppStarter: Activating background region monitoring
00:01 D/AppStarter: Background region monitoring activated for region id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null
00:01 D/BeaconManager: Starting monitoring region id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null with uniqueID: backgroundRegion
00:01 D/BeaconManager: callback packageName: com.app.dealer_app
00:01 D/MonitoringStatus: Not restoring monitoring state because it was recorded too many milliseconds ago: 1493625097880
00:01 D/BeaconManager: updating background flag to true
00:01 D/BeaconManager: updating scan period to 10000, 300000
00:01 I/OpenGLRenderer: Initialized EGL, version 1.4
00:01 D/OpenGLRenderer: Swap behavior 1
00:01 I/BeaconService: start monitoring received
00:01 D/BeaconService: startMonitoring called
00:01 D/MonitoringStatus: saveMonitoringStatusIfOn()
00:01 D/BeaconService: Currently monitoring 1 regions.
00:01 D/CycledLeScanner: start called
00:01 D/CycledLeScanner: starting a new scan cycle
00:01 D/CycledLeScanner: starting a new bluetooth le scan
00:01 D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
00:01 D/CycledLeScanner: Scan started
00:01 D/CycledLeScanner: Set scan periods called with 1100, 0 Background mode must have changed.
00:01 D/CycledLeScanner: We are not in the background.  Cancelling wakeup alarm
00:01 D/CycledLeScanner: cancel wakeup alarm: null
00:01 D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036751759419 ms: PendingIntent{2e7b1ea: android.os.BinderProxy@9d83cdb}
00:01 I/BeaconService: set scan intervals received
00:01 D/CycledLeScanner: Set scan periods called with 10000, 300000 Background mode must have changed.
00:01 D/CycledLeScanner: We are in the background.  Setting wakeup alarm
00:01 D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{2e7b1ea: android.os.BinderProxy@9d83cdb}
00:01 E/Volley: [4076] BasicNetwork.performRequest: Request at https://app.callproof.com/hustle/check/token has been redirected to http://app.callproof.com/hustle/check/token/
00:01 E/MyFirebaseInstanceIDService: sendRegistrationToServer: fOSgNJQeue0:APA91bHx35scTseaF-zAwTFCuoHbb8UyHzuWFfes-QUNx345ubhKZRBRg4fGO5WAHmAkmAw9PCEvqJDUpoodjWpimx-t485tybr6uh52dw6r3aVH3D1vmHrVD60Bw-5EqGNrkN85SSMR
[... cut ...]
00:31 D/BeaconParser: Processing pdu type FF: 02011b0bff4c0009060309c0a801f70000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 14
00:31 D/BeaconParser: This is not a matching Beacon advertisement. (Was expecting 02 15.  The bytes I see are: 02011b0bff4c0009060309c0a801f70000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
00:31 D/CycledLeScanner: Done with scan cycle
00:31 D/BeaconService: Calling ranging callback
00:31 D/RunningAverageRssiFilter: Running average mRssi based on 10 measurements: -42.166666666666664

davidgyoung, 2017-05-18:
@roshanparmar, sorry, this is getting confusing to follow.    It's really important to focus on one issue at a time, so this thread doesn't go off track.  Otherwise I cannot help.

This issue was opened about background region exits using:

`regionBootstrap = new RegionBootstrap(this, region);`

And not about using `beaconManger.bind(...)`.  If you need help on the latter means of using the library, please post a new question on StackOverflow.com, and I'll be happy to answer there.

Please only post logs in this thread that use this setup from the start of the question, otherwise I won't be able to understand what is going on:

```
        Identifier identifier = Identifier.parse("C48C6716-193F-477B-B73A-C550CE582A22");
        // wake up the app when a beacon is seen
        region = new Region("backgroundRegion",
                identifier, null, null);
        regionBootstrap = new RegionBootstrap(this, region);
```

For any log you post, I also need to know what library version was running.  I can't tell that from the most recent Pixel log you posted above, as there is no log line that shows the version.  Is one of these logs using *only* the regionBootstrap setup code described above and using version 2.11-beta1 of the library?

roshanparmar, 2017-05-18:
00:31 D/RangedBeacon: calculated new runningAverageRssi: -42.166666666666664
00:31 D/Callback: attempting callback via intent: ComponentInfo{com.app.dealer_app/org.altbeacon.beacon.BeaconIntentProcessor}
00:31 D/CurveFittedDistanceCalculator: calculating distance based on mRssi of -42.166666666666664 and txPower of -72
00:31 D/CurveFittedDistanceCalculator: avg mRssi: -42.166666666666664 distance: 0.0047464372492001024
00:31 D/CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
00:31 D/CycledLeScanner: starting a new scan cycle
00:31 D/CycledLeScanner: We are already scanning
00:31 D/CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
00:31 D/CycledLeScanner: Scan started
00:31 D/BeaconIntentProcessor: got an intent to process
00:31 D/BeaconIntentProcessor: got ranging data
00:31 D/CycledLeScannerForLollipop: got record
00:31 D/BeaconParser: Ignoring pdu type 01
00:31 D/BeaconParser: Processing pdu type FF: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e604ecb80509424b4f4e14160a1805be0b0b3b97461fc55a00000003e604ec0000000000 with startIndex: 5, endIndex: 29
00:31 D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
00:31 D/BeaconParser: Bytes are: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e604ecb80509424b4f4e14160a1805be0b0b3b97461fc55a00000003e604ec0000000000
00:31 D/BeaconService: Beacon packet detected for: id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: 998 id3: 1260 with rssi -32
00:31 D/BeaconService: beacon detected : id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: 998 id3: 1260
00:31 D/BeaconService: looking for ranging region matches for this beacon
00:31 D/BeaconService: matches ranging region: id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null
00:31 D/RangeState: adding id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: 998 id3: 1260 to existing range for: org.altbeacon.beacon.service.RangedBeacon@831d979
00:31 D/CycledLeScannerForLollipop: got record
00:31 D/CycledLeScannerForLollipop: with service uuid: 0629af91-0356-0cc6-2a93-332c786ef0cc
00:31 D/BeaconParser: Ignoring pdu type 07
00:31 D/BeaconParser: Ignoring pdu type 09
00:31 D/BeaconParser: No PDUs to process in this packet.
00:32 D/CycledLeScannerForLollipop: got record
00:32 D/CycledLeScannerForLollipop: with service uuid: 0629af91-0356-0cc6-2a93-332c786ef0cc
00:32 D/BeaconParser: Ignoring pdu type 07
00:32 D/BeaconParser: Ignoring pdu type 09
00:32 D/BeaconParser: No PDUs to process in this packet.
00:32 D/CycledLeScannerForLollipop: got record
00:32 D/CycledLeScannerForLollipop: with service uuid: 0629af91-0356-0cc6-2a93-332c786ef0cc
00:32 D/BeaconParser: Ignoring pdu type 07
00:32 D/BeaconParser: Ignoring pdu type 09
00:32 D/BeaconParser: No PDUs to process in this packet.
00:32 D/CycledLeScannerForLollipop: got record
00:32 D/CycledLeScannerForLollipop: with service uuid: 0629af91-0356-0cc6-2a93-332c786ef0cc
00:32 D/BeaconParser: Ignoring pdu type 07
00:32 D/BeaconParser: Ignoring pdu type 09
00:32 D/BeaconParser: No PDUs to process in this packet.
00:32 D/CycledLeScannerForLollipop: got record
00:32 D/CycledLeScannerForLollipop: with service uuid: 0629af91-0356-0cc6-2a93-332c786ef0cc
00:32 D/BeaconParser: Ignoring pdu type 07
00:32 D/BeaconParser: Ignoring pdu type 09
00:32 D/BeaconParser: No PDUs to process in this packet.
00:32 D/CycledLeScannerForLollipop: got record
00:32 D/CycledLeScannerForLollipop: with service uuid: 0629af91-0356-0cc6-2a93-332c786ef0cc
00:32 D/BeaconParser: Ignoring pdu type 07
00:32 D/BeaconParser: Ignoring pdu type 09
00:32 D/BeaconParser: No PDUs to process in this packet.
00:32 D/CycledLeScannerForLollipop: got record
00:32 D/CycledLeScannerForLollipop: with service uuid: 0629af91-0356-0cc6-2a93-332c786ef0cc
00:32 D/BeaconParser: Ignoring pdu type 07
00:32 D/BeaconParser: Ignoring pdu type 09
00:32 D/BeaconParser: No PDUs to process in this packet.
00:32 D/CycledLeScanner: Waiting to stop scan cycle for another 99 milliseconds
00:32 D/CycledLeScannerForLollipop: got record
00:32 D/BeaconParser: Ignoring pdu type 01
00:32 D/BeaconParser: Processing pdu type FF: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e604ecb80509424b4f4e14160a1805be0b0b3b97461fc55a00000003e604ec0000000000 with startIndex: 5, endIndex: 29
00:32 D/BeaconParser: This is a recognized beacon advertisement -- 02 15 seen
00:32 D/BeaconParser: Bytes are: 0201061aff4c000215c48c6716193f477bb73ac550ce582a2203e604ecb80509424b4f4e14160a1805be0b0b3b97461fc55a00000003e604ec0000000000
00:32 D/BeaconService: Beacon packet detected for: id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: 998 id3: 1260 with rssi -32
00:32 D/BeaconService: beacon detected : id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: 998 id3: 1260
00:32 D/BeaconService: looking for ranging region matches for this beacon
00:32 D/BeaconService: matches ranging region: id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null
00:32 D/RangeState: adding id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: 998 id3: 1260 to existing range for: org.altbeacon.beacon.service.RangedBeacon@831d979
00:32 D/CycledLeScanner: Done with scan cycle
00:32 D/BeaconService: Calling ranging callback
00:32 D/RunningAverageRssiFilter: Running average mRssi based on 11 measurements: -40.714285714285715
[... 1910 lines cut ...]
01:42 D/BeaconParser: Ignoring pdu type 01
[... cut ...]
01:42 D/BeaconService: This region (id1: c48c6716-193f-477b-b73a-c550ce582a22 id2: null id3: null) does not match beacon: id1: 0629af91-0356-0cc6-2a93-332c786ef0cc id2: 0 id3: 0
01:42 D/BeaconManager: updating background flag to true
01:42 D/BeaconManager: updating scan period to 10000, 300000
01:42 I/BeaconService: set scan intervals received
01:42 D/CycledLeScanner: Set scan periods called with 10000, 300000 Background mode must have changed.
01:42 D/CycledLeScanner: We are in the background.  Setting wakeup alarm
01:42 D/CycledLeScanner: Set a wakeup alarm to go off in 300000 ms: PendingIntent{2e7b1ea: android.os.BinderProxy@9d83cdb}

roshanparmar, 2017-05-18:
I am using 2.11-beta1 version in Google pixel.
In this post, only response for Google pixel issue.

For more queries I will post another question.

roshanparmar, 2017-05-18:
Google pixel doing the same thing again and again as I updated the library to 2.11-beta1.

davidgyoung, 2017-05-18:
Looking at the latest log for the pixel that run from
   00:00-00:48
and then from
   01:22-01:42

I don't see any region exits in the log.

davidgyoung, 2017-05-20:
@roshanparmar, I have built a test case with steps to reproduce this problem and prove that it is fixed with 2.11-beta1.  Details are in #512.  Using the reference app to reproduce the problem allows us to have a test case that more than one person can run.  If you think your problem reported here is not fixed as described in #512, please help create a reproducible test case with the reference app.

closed 2017-05-20 by davidgyoung

roshanparmar, 2017-05-22:
[dealerapp-10-104.logcat.txt](https://github.com/AltBeacon/android-beacon-library/files/1017977/dealerapp-10-104.logcat.txt)

I am attaching more logs of Google Pixel for frequently entering and exiting region.
Please response.

roshanparmar, 2017-05-22:
I just want to confirm that is beacon enter/exit in some times automatically means beacon continuous do this ?

roshanparmar, 2017-05-22:
May I test it on Google pixel emulator ?
I am asking this because emulator have limitation for bluetooth.

roshanparmar, 2017-05-22:
Hi,
I faced this issue in LG Nexus 5x too.
Here is logs - 

05-22 16:57:31.117 21701-21701/? I/art: Late-enabling -Xcheck:jni
05-22 16:57:31.183 21701-21701/com.app.dealer_app W/InstanceID/Rpc: Found 10018
05-22 16:57:31.195 21701-21701/com.app.dealer_app I/FA: App measurement is starting up, version: 10298
05-22 16:57:31.195 21701-21701/com.app.dealer_app I/FA: To enable debug logging run: adb shell setprop log.tag.FA VERBOSE
05-22 16:57:31.200 21701-21701/com.app.dealer_app I/FA: To enable faster debug mode event logging run:
                                                          adb shell setprop debug.firebase.analytics.app com.app.dealer_app
05-22 16:57:31.200 21701-21701/com.app.dealer_app D/FA: Debug-level message logging enabled
05-22 16:57:31.204 21701-21701/com.app.dealer_app I/FirebaseInitProvider: FirebaseApp initialization successful
05-22 16:57:31.258 21701-21701/com.app.dealer_app D/NetworkSecurityConfig: No Network Security Config specified, using platform default
05-22 16:57:31.291 21701-21701/com.app.dealer_app I/CrashlyticsCore: Initializing Crashlytics 2.3.16.dev
05-22 16:57:31.312 21701-21701/com.app.dealer_app I/BeaconManager: BeaconManager started up on pid 21701 named 'com.app.dealer_app' for application package 'com.app.dealer_app'.  isMainProcess=true
05-22 16:57:31.313 21701-21701/com.app.dealer_app D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
05-22 16:57:31.314 21701-21701/com.app.dealer_app D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
05-22 16:57:31.315 21701-21701/com.app.dealer_app D/DealerApplication: setting up background monitoring for beacons and power saving
05-22 16:57:31.317 21701-21701/com.app.dealer_app D/BeaconManager: This consumer is not bound.  binding: org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer@71e5a98
05-22 16:57:31.319 21701-21701/com.app.dealer_app D/BeaconManager: consumer count is now: 1
05-22 16:57:31.319 21701-21701/com.app.dealer_app D/AppStarter: Waiting for BeaconService connection
05-22 16:57:31.343 21701-21745/com.app.dealer_app D/ApplicationLoaders: ignored Vulkan layer search path /data/app/com.google.android.gms-2/lib/arm64:/system/fake-libs64:/data/app/com.google.android.gms-2/base.apk!/lib/arm64-v8a for namespace 0x7b16cf40f0
05-22 16:57:32.347 21701-21701/com.app.dealer_app D/TESTFAIRYSDK: onActivityResumed com.app.dealer_app.activity.SplashActivity
05-22 16:57:32.387 21701-21701/com.app.dealer_app D/BluetoothCrashResolver: constructed
05-22 16:57:32.388 21701-21701/com.app.dealer_app W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
05-22 16:57:32.388 21701-21701/com.app.dealer_app D/BluetoothCrashResolver: Read 0 Bluetooth addresses
05-22 16:57:32.390 21701-21701/com.app.dealer_app D/BluetoothCrashResolver: started listening for BluetoothAdapter events
05-22 16:57:32.391 21701-21701/com.app.dealer_app I/CycledLeScanner: This Android 5.0.  We are using new scanning APIs
05-22 16:57:32.393 21701-21701/com.app.dealer_app I/BeaconService: beaconService version 2.11-beta1 is starting up on the main process
05-22 16:57:32.456 21701-21701/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47O, Nexus 5X, LGE
05-22 16:57:32.456 21701-21701/com.app.dealer_app D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.456 21701-21701/com.app.dealer_app D/AndroidModel: Score is 0 for Motorola;XT1092;MPE24.49-18;6.0 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/ModelSpecificDistanceCalculator: found a match with score 1
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, KOT49H, Nexus 4, LGE
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 7.1.2, N2G47O, Nexus 5X, LGE
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 0 for XT1115;Moto X Pro;LXG22.67-7.1;5.0.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 0 for Motorola;XT1092;MPE24.49-18;6.0 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 4;KOT49H;4.4.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.457 21701-21701/com.app.dealer_app D/AndroidModel: Score is 1 for LGE;Nexus 5;LPV79;4.4.2 compared to LGE;Nexus 5X;N2G47O;7.1.2
05-22 16:57:32.458 21701-21701/com.app.dealer_app D/ModelSpecificDistanceCalculator: found a match with score 1
05-22 16:57:32.458 21701-21701/com.app.dealer_app D/ModelSpecificDistanceCalculator: Finding best distance calculator for 4.4.2, KOT49H, Nexus 4, LGE
[... cut ...]
05-22 16:58:26.765 21701-22881/com.app.dealer_app D/BeaconIntentProcessor: got an intent to process
05-22 16:58:26.767 21701-22881/com.app.dealer_app D/BeaconIntentProcessor: got ranging data
05-22 16:58:26.769 21701-22881/com.app.dealer_app D/BeaconIntentProcessor: got an intent to process
05-22 16:58:26.770 21701-22881/com.app.dealer_app D/BeaconIntentProcessor: got ranging data

In above logs, I have entered in region, go to background and again come to foreground and exit from region.
Please review logs and respond.

roshanparmar, 2017-05-22:
[dealerapp_lgnexus5x.logcat.txt](https://github.com/AltBeacon/android-beacon-library/files/1018623/dealerapp_lgnexus5x.logcat.txt)

I have attached more logs in file of LG Nexus 5x with same entering/exiting issue.

davidgyoung, 2017-05-22:
Sorry, @roshanparmar, I do not think I can help debug your app as I do not have the source code and many things can cause the symptoms you describe.

If you suspect the problem is with the library, please try to reproduce the problem with a minimal code example starting with the reference app.  See here for my tests showing I do not get spurious region exits:

https://github.com/AltBeacon/android-beacon-library/issues/512

Can you reproduce your problem with that reference project?

A couple of tips on your above comments:

1. You cannot test in an emulator because you cannot get BLE callbacks.

2. I still see permissions problems in the 5X logs posted earlier:

java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results