AltBeacon/android-beacon-library #511: Empty beacon advertisement with Eddystone-UID and Eddystone-URL

davigonz, 2017-05-18:
I'm trying to get the battery level from a beacon detected in the didRangeBeaconsInRegion method by using RangeNotifier but mDataFields and mExtraDataFields beacon fields are empty.

### Expected behavior

Beacon advertisement (mDataFields or mExtraDataFields) includes data

### Actual behavior

Beacon advertisement  (mDataFields or mExtraDataFields) is empty

### Steps to reproduce this behavior

Use RangeNotifier to detect beacons. Wait for didRangeBeaconsInRegion method to be called and afterwards call the beacon methods getDataFields() or getExtraDataFields()

### Mobile device model and OS version

Samsung Galaxy Note 4, Android 6

### Android Beacon Library version

2.9.2

davidgyoung, 2017-05-18:
What beacon transmitter are you using and what layout (AltBeacon, iBeacon, etc).  Understand the battery level is only present with AltBeacon on some manufacturers like Radius Networks.

davigonz, 2017-05-18:
Hi @davidgyoung , thank you for your answer. I'm not using neither AltBeacon nor iBeacon, but Eddystone.

Battery level is also present in Eddystone TLM by default. I'm using Eddystone-UID and Eddystone-URL instead of TLM but the manufacturer which I bought this beacons from commented me that battery level should appear in beacon advertisement as well.

davidgyoung, 2017-05-18:
@davigonz, you'll have to get more information from the manufacturer, as battery level is often a proprietary manufacturer-specific implementation.  I do not believe it could be present in Eddystone-UID or Eddystone-URL, because there is no standard field to hold this information.

The only two ways I know of to get a battery level from an advertisement are from `beacon.getDatafields(0)` for AltBeacon implementations by Radius Networks and some other manufacturers, and from Eddystone-TLM as documented here:  https://altbeacon.github.io/android-beacon-library/eddystone-how-to.html

If you can get details from the manufacturer about where the battery level should appear in the specific beacon advertisement you are looking for, but still don't see it working with the library, please re-open this issue with that info.

closed 2017-05-18 by davidgyoung