AltBeacon/android-beacon-library #512: Background region monitoring stability with 2.11-beta1

davidgyoung, 2017-05-20:
Library version 2.10 caused background monitoring to cause intermittent exit/entry events even when beacons are always around.

The problem was caused by a bug that unintentionally left low power scanning enabled at all times after a particular sequence.  This has been resolved in 2.11-beta in this change : 14c0508e8863a94e8a86ec4b3da3bb98a2c34de2 .  The problem did not exist in 2.9.x and earlier versions of the library.

Sequence to reproduce the problem:

1. Start the reference app
2. Turn on a beacon transmitting at 10 Hz or more so it is detected multiple times in a scan cycle (causes a region entry)
3. Turn off the beacon and wait for a region exit
4. Put the app to the background
5. Wait for a lower power scan to start in the background
6. Turn on the beacon so it is detected **during the low power scan** (causes a region entry)

After the above a region exit will happen periodically followed by an entry.  This is caused by the fact that OS-level low power scanning has its own duty cycle that will delay results, and because the library does not take this into account, it will cause region exits inappropriately if results don't come quite quickly enough.

You  will know you have entered this state with 2.10 after seeing the debug line "We've been detecting for a bit.  Stopping Android L background scanning".

For this test, the following branch of the reference app (modified to use 2.10) was installed and run on  a  Nexus 9 on Android 7.0 with the app always running in the background.  https://github.com/AltBeacon/android-beacon-library-reference/tree/multi-process-support

```
adb logcat | grep -e "BeaconReferenceApp:" -e "beaconService version" -e "Stopping Android L background scanning"

05-20 19:31:34.676 28386 28386 D BeaconReferenceApp: this is the main process
05-20 19:31:34.676 28386 28386 D BeaconReferenceApp: setting up background monitoring for beacons and power saving
05-20 19:31:34.698 28386 28386 I BeaconService: beaconService version 2.10 is starting up
05-20 19:31:52.575 28386 28717 D BeaconReferenceApp: did enter region.
05-20 19:31:52.575 28386 28717 D BeaconReferenceApp: auto launching MainActivity
05-20 19:32:46.969 28386 29510 D BeaconReferenceApp: did exit region.
05-20 19:33:04.563 28386 29795 D BeaconReferenceApp: did enter region.
05-20 19:33:04.563 28386 29795 D BeaconReferenceApp: Sending notification.
05-20 19:33:15.039 28386 28386 D CycledLeScannerForLollipop: We've been detecting for a bit.  Stopping Android L background scanning
05-20 19:33:26.962 28386 30152 D BeaconReferenceApp: did exit region.
05-20 19:33:27.091 28386 30161 D BeaconReferenceApp: did enter region.
05-20 19:33:27.091 28386 30161 D BeaconReferenceApp: Sending notification.
05-20 19:33:37.976 28386 28386 D CycledLeScannerForLollipop: We've been detecting for a bit.  Stopping Android L background scanning
05-20 19:34:06.954 28386 30682 D BeaconReferenceApp: did exit region.
05-20 19:34:07.033 28386 30683 D BeaconReferenceApp: did enter region.
05-20 19:34:07.033 28386 30683 D BeaconReferenceApp: Sending notification.
05-20 19:34:17.970 28386 28386 D CycledLeScannerForLollipop: We've been detecting for a bit.  Stopping Android L background scanning
05-20 19:34:46.968 28386 31216 D BeaconReferenceApp: did exit region.
05-20 19:34:47.220 28386 31225 D BeaconReferenceApp: did enter region.
05-20 19:34:47.220 28386 31225 D BeaconReferenceApp: Sending notification.
05-20 19:34:57.975 28386 28386 D CycledLeScannerForLollipop: We've been detecting for a bit.  Stopping Android L background scanning
05-20 19:35:26.949 28386 31746 D BeaconReferenceApp: did exit region.
05-20 19:35:27.166 28386 31748 D BeaconReferenceApp: did enter region.
05-20 19:35:27.166 28386 31748 D BeaconReferenceApp: Sending notification.
05-20 19:35:37.970 28386 28386 D CycledLeScannerForLollipop: We've been detecting for a bit.  Stopping Android L background scanning

```

In the above sequence, a beacon was enabled at 19:31:52, then disabled at 19:32:46.  It was then re-enabled at 19:33:27 and left on.  Despite the fact that the beacon was left on, there were several spurious exit events at 19:34:06, 19:34:46, 19:35:26. 

Running the same test with 2.11-beta1:

```
adb logcat | grep -e "BeaconReferenceApp:" -e "beaconService version" -e "Stopping Android L background scanning"

05-20 19:40:58.205  4246  4246 D BeaconReferenceApp: this is the main process
05-20 19:40:58.205  4246  4246 D BeaconReferenceApp: setting up background monitoring for beacons and power saving
05-20 19:40:58.452  4246  4246 I BeaconService: beaconService version 2.11-beta1 is starting up on the main process
05-20 19:42:53.135  4246  6214 D BeaconReferenceApp: did enter region.
05-20 19:42:53.135  4246  6214 D BeaconReferenceApp: auto launching MainActivity
05-20 19:43:27.040  4246  6739 D BeaconReferenceApp: did exit region.
05-20 19:43:32.633  4246  6811 D BeaconReferenceApp: did enter region.
05-20 19:43:32.633  4246  6811 D BeaconReferenceApp: Sending notification.
05-20 19:43:43.101  4246  4246 D CycledLeScannerForLollipop: We've been detecting for a bit.  Stopping Android L background scanning
05-20 19:50:06.977  4246 12252 D BeaconReferenceApp: did exit region
```

In the above sequence, a beacon is turned on at 19:42:53, turned off at 19:43:27 and on again at 19:43:32.  After that the beacon is left on until ~19:49:45, and shortly thereafter the region exits.   Note there are no spurious exits.

The core problem was also reported in #506 and #509

hirenpatelazilen, 2017-05-23:
@davidgyoung, thanks for prompt response. 

I want to use this latest version 2.11 beta 1 library inside my project. Can I have access for this 2.11=beta1 version in order to test frequent entry/exit events for Android L.

Best regards,
Hiren Patel

Kristea, 2017-06-30:
Was having this issue too. Constant entry exits after the initial one.
Has this been fixed in the new 2.11 update?

davidgyoung, 2017-06-30:
Yes, the fix described in this issue is in release 2.11

closed 2017-06-30 by davidgyoung