AltBeacon/android-beacon-library #520: Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions.

sungjin0213, 2017-06-01:
### Expected behavior
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/> will not conflict with <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/> of beacon library.

### Actual behavior
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/> conflicts with <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/> of beacon library preventing the APK being uploaded to play store.

### Steps to reproduce this behavior
1. Have a project with beacon library and targetSdkVersion to 25 and following permission in manifest:
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
2. Try to upload an APK to see if it's saying "Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions."

### Mobile device model and OS version
Android Studio 2.2.3

### Android Beacon Library version
compile 'org.altbeacon:android-beacon-library:2.8.1'

sungjin0213, 2017-06-01:
https://stackoverflow.com/questions/44298951/android-manifest-duplicate-permissions is my exact issue that someone else have

MaksymDovbnia, 2017-06-01:
The same problem when upload app in Google Play

sungjin0213, 2017-06-01:
Awesome! That fixed it!

closed 2017-06-01 by sungjin0213