AltBeacon/android-beacon-library #521: Unable to publish app containing Altbeacon and another library both requiring ACCESS_COARSE_LOCATION

bitbytedog, 2017-06-01:
### Expected behavior

To be able to use the Altbeacon library in an application targeting an API greater than 22, that also uses other third party libraries also targeted at API levels greater than 22 that declare the uses-permission android.permission.ACCESS_COARSE_LOCATION.

### Actual behavior

App links and runs, but cannot be uploaded to Google Play Developer Console for publishing to the Google Play store.

> Upload failed 
> <br>Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions

### Steps to reproduce this behavior

Create an app that uses the altbeacon library and also another library that has the following declared in the AndroidManifest.xml

~~~~
<uses-sdk
       android:minSdkVersion="17"
       android:targetSdkVersion="25" />

<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
~~~~

### Mobile device model and OS version

All apps targeting Android API 23 or greater, that use another library component that requires the ACCESS_COARSE_LOCATION permission.

### Android Beacon Library version

v2.9.2, I also tried v2.10

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

see https://stackoverflow.com/questions/44298951/android-manifest-duplicate-permissions also.

davidgyoung, 2017-06-01:
Please see workaround here: #520

This problem just showed up and appears to be caused by either new behavior in manifest merging, or from new Google Play validation rules.

bitbytedog, 2017-06-01:
This is a slightly different issue. The two instances of ACCESS_COARSE_LOCATION are in AAR library manifests, not in the APP manifest. In general it will not work because the AAR manifest cannot be easily modified.

In our case the conflicting AAR was one of ours, so we tried the specific workaround you mention, it still didn't work and the workaround in #520 results in a merged manifest containing the following lines:

~~~~
<uses-permission
       android:name="android.permission.ACCESS_COARSE_LOCATION"
       android:maxSdkVersion="22" />
<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION" />
~~~~

bitbytedog, 2017-06-01:
My workaround is to place this in the APP AndroidManifest.xml.

~~~~
<uses-permission
       android:name="android.permission.ACCESS_COARSE_LOCATION"
       tools:node="remove"/>
~~~~

This fixes the issue but removes the wrong permission from our point of view (it removes the permission on our library, not the Altbeacon library), but it may be satisfactory for others.

For reference the following workaround should have worked perfectly but failed to remove anything.

~~~~
 <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="remove"
        tools:selector="org.altbeacon.beacon"/>
~~~~

Another bug in the manifest merger I expect.

davidgyoung, 2017-06-02:
Reviewing older projects, it appears that the manifest merger has long added both permissions to the manifest in this case:

```
<uses-permission
       android:name="android.permission.ACCESS_COARSE_LOCATION">
 <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="remove"/>
```

What has changed, apparently in the last few days,  is the validation in the Google Play store.

Unfortunately, I also see that the correct workaround  does not work:

```
 <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="remove"/>
```

I can look at alternatives to this tag for a future release of the library, but the for the exiting one, I don't see a better solution than adding this to the manifest:

```
<uses-permission
       android:name="android.permission.ACCESS_COARSE_LOCATION"
       tools:node="remove"/>
```

The problem is that this will only request that permission for devices with SDK 23 or higher (Android 6+)  If the app is run on devices with Android 5.x or lower, location will not work.   A better alternative my be to do this:

```
<uses-permission
       android:name="android.permission.ACCESS_COARSE_LOCATION"
       tools:node="remove"/>
<uses-permission
       android:name="android.permission.ACCESS_FINE_LOCATION"/>
```

This will allow (fine) location access on all devices and presumably avoid the validation error in the Google Play Store.

Rob-Houweling, 2017-06-06:
> The problem is that this will only request that permission for devices with SDK 23 or higher (Android 6+) If the app is run on devices with Android 5.x or lower, location will not work. A better alternative may be to do this:
`<uses-permission
       android:name="android.permission.ACCESS_COARSE_LOCATION"
       tools:node="remove"/>
<uses-permission
       android:name="android.permission.ACCESS_FINE_LOCATION"/>`
> This will allow (fine) location access on all devices and presumably avoid the validation error in the Google Play Store.

For me, this doesn't work either. I can't get anything to work to be honest. Can't get my app into the store now, this really sucks...

davidgyoung, 2017-06-06:
@Rob-Houweling, any more details on why this doesn't work?  Do you get an error from Google Play or do you get some kind of error during your build?

You can check the generated merged manifest by looking in this folder after a build:

`./app/build/intermediates/manifests/full/release/AndroidManifest.xml`

(In the above path, the app folder is the name of your application module, yours may differ.)

If the suggested workaround works, you should only see one location permission line in that file like this:

`<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />`

What do you see?

matthewmichihara, 2017-06-06:
@davidgyoung I think your workaround gets around the validation issues, but wouldn't the `<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION" />` permission remain? 

Or were you expecting the `<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" tools:node="remove"/>` line to remove the `<uses-permission-sdk-23 />` permission as well?

davidgyoung, 2017-06-06:
My assumption (unverified) is that GooglePlay will not complain about both  of the following being in the manifest:

```
<uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION">
```

Which would leave the app with the needed location permissions in the manifest to access location for all SDK versions.  The presence of uses-permission-sdk-23 would be effectively ignored in this setup.

matthewmichihara, 2017-06-06:
Ah yes, that matches what I was expecting. I just got slightly confused by your earlier comment 

```
"If the suggested workaround works, you should only see one location permission line in that file like this:

<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />"
```

Rob-Houweling, 2017-06-07:
Hi,

Thanks for the response.
This is my generated merged manifest, partial but all of the permission nodes in the generated order:
`<uses-sdk
      minSdkVersion='19'
      targetSdkVersion='25'>
  </uses-sdk>
  <uses-permission
      name='android.permission.ACCESS_COARSE_LOCATION'>
  </uses-permission>
  <uses-permission
      name='android.permission.ACCESS_FINE_LOCATION'>
  </uses-permission>
  <uses-permission
      name='android.permission.BLUETOOTH'>
  </uses-permission>
  <uses-permission
      name='android.permission.BLUETOOTH_ADMIN'>
  </uses-permission>
  <uses-permission
      name='android.permission.INTERNET'>
  </uses-permission>
  <uses-permission
      name='android.permission.RECEIVE_BOOT_COMPLETED'>
  </uses-permission>
  <uses-permission
      name='com.android.alarm.permission.SET_ALARM'>
  </uses-permission>
  <uses-permission
      name='android.permission.WAKE_LOCK'>
  </uses-permission>
  <uses-permission
      name='android.permission.ACCESS_NETWORK_STATE'>
  </uses-permission>
  <uses-permission
      name='android.permission.ACCESS_WIFI_STATE'>
  </uses-permission>
  <permission
      name='nl.spotyou.droid.permission.C2D_MESSAGE'>
  </permission>
  <uses-permission
      name='android.permission.WRITE_EXTERNAL_STORAGE'>
  </uses-permission>
  <uses-permission
      name='nl.spotyou.droid.permission.C2D_MESSAGE'>
  </uses-permission>
  <uses-permission
      name='com.google.android.c2dm.permission.RECEIVE'>
  </uses-permission>
  <uses-permission
      name='android.permission.GET_ACCOUNTS'>
  </uses-permission>
  
  <uses-permission-sdk-23
      name='android.permission.ACCESS_COARSE_LOCATION'>
  </uses-permission-sdk-23>`

davidgyoung, 2017-06-07:
@Rob-Houweling,

Can you show the permission entries from your app's manifest?  Does it include an entry like this?

```
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="remove"/>
```

Rob-Houweling, 2017-06-07:
Yes it does. Here's my app manifest:

~~~ xml
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" xmlns:tools="http://schemas.android.com/tools" android:installLocation="preferExternal" package="nl.spotyou.droid" android:versionName="1.1" android:versionCode="2">
	<uses-sdk android:minSdkVersion="19" android:targetSdkVersion="25" />
  <uses-permission
       android:name="android.permission.ACCESS_COARSE_LOCATION"
       tools:node="remove"/>
  <uses-permission
         android:name="android.permission.ACCESS_FINE_LOCATION"/>
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.INTERNET" />
	<uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" />
	<uses-permission android:name="com.android.alarm.permission.SET_ALARM" />
	<uses-permission android:name="android.permission.WAKE_LOCK" />
	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />
	<application android:label="SpotYou" android:icon="@drawable/icon"></application>
</manifest>
~~~

davidgyoung, 2017-06-07:
Clearly the` tools:node="remove"` is not working for you as it is supposed to, which I cannot explain.  

An alternative solution is to avoid manifest merging from the library entirely using this technique:

http://altbeacon.github.io/android-beacon-library/alternate-configuration.html

This will require you to manually add all of the desired entries to your manifest as described on that page.

Rob-Houweling, 2017-06-08:
Thanks for that.
Check this:
https://issuetracker.google.com/issues/62218594

davidgyoung, 2017-06-30:
This issue has been resolved in 2.11 via #527

closed 2017-06-30 by davidgyoung