AltBeacon/android-beacon-library #522: java.lang.NoClassDefFoundError: org.altbeacon.beacon.logging.Loggers

jmvines, 2017-06-02:
### Expected behavior
not exception

### Actual behavior

 E/AndroidRuntime: FATAL EXCEPTION: main
                                                   java.lang.ExceptionInInitializerError
                                                       at org.altbeacon.beacon.powersave.BackgroundPowerSaver.<init>(BackgroundPowerSaver.java:44)
                                                       at com.naranya.claroapps.ClaroApplication.onCreate(ClaroApplication.java:121)
                                                       at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1017)
                                                       at android.app.ActivityThread.handleBindApplication(ActivityThread.java:4590)
                                                       at android.app.ActivityThread.access$1400(ActivityThread.java:157)
                                                       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349)
                                                       at android.os.Handler.dispatchMessage(Handler.java:99)
                                                       at android.os.Looper.loop(Looper.java:176)
                                                       at android.app.ActivityThread.main(ActivityThread.java:5317)
                                                       at java.lang.reflect.Method.invokeNative(Native Method)
                                                       at java.lang.reflect.Method.invoke(Method.java:511)
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1102)
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:869)
                                                       at dalvik.system.NativeStart.main(Native Method)
                                                    Caused by: java.lang.NoClassDefFoundError: org.altbeacon.beacon.logging.Loggers
                                                       at org.altbeacon.beacon.logging.LogManager.<clinit>(LogManager.java:27)
                                                       at org.altbeacon.beacon.powersave.BackgroundPowerSaver.<init>(BackgroundPowerSaver.java:44) 

### Steps to reproduce this behavior

when call new BackgroundPowerSaver(this);

### Mobile device model and OS version

alps m4 SS4045 4.2

### Android Beacon Library version

 2.10

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-06-02:
I have heard a number of reports of this problem that were caused by apps using libraries that run a service in a separate process like Google Cloud Messaging and Nokia Here Maps.  You may not even be aware that you are using a library that does this.

A fix for this was put in #479 and is available for testing in this binary release:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.11-beta1

@jmvines, please try that release and let me know if it solves your problem.

jmvines, 2017-06-05:
Thanks for your response @davidgyoung .

i will tell you if this problem is solved with this fix.

Regards.

davidgyoung, 2017-06-30:
This fix described has been released in 2.11.  Please reopen if that does not solve the problem.

closed 2017-06-30 by davidgyoung