AltBeacon/android-beacon-library #525: Without Location permission foreground scanning does not work too

ali-bugdayci, 2017-06-05:
### Expected behavior

Expecting to find the devices while the app is in the foreground without ACCESS_COARSE_LOCATION permission
 
### Actual behavior

Not getting the permission ACCESS_COARSE_LOCATION on >= Build.VERSION_CODES.M devices results in not finding the devices in range

### Steps to reproduce this behavior

* Remove the ACCESS_COARSE_LOCATION
* Or not ask for the permission ACCESS_COARSE_LOCATION to the user
* Scan results with no devices

### Mobile device model and OS version

* Samsung SM-N910C
* Samsung SM-G532F

### Android Beacon Library version

Latest:
    compile 'org.altbeacon:android-beacon-library:2+'

android compile version:
    compileSdkVersion 25
    buildToolsVersion "25.0.3"

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-06-05:
This is expected behavior.  Android M introduced restrictions disallowing Bluetooth scans without the location permission.  See here: http://developer.radiusnetworks.com/2015/09/29/is-your-beacon-app-ready-for-android-6.html

Let me know if I misunderstand the issue.

closed 2017-06-05 by davidgyoung

ali-bugdayci, 2017-06-05:
Ok. Thank you for the info