AltBeacon/android-beacon-library #526: Android Things scanning stops after 30 minutes

davidgyoung, 2017-06-11:
### Expected behavior

Scanning continues indefinitely once ranging starts

### Actual behavior

Scanning stops after 30 minutes

### Steps to reproduce this behavior

Create an Android Things app that starts ranging 

### Mobile device model and OS version

Raspberry Pi 3, androidthings_rpi3_devpreview_4

### Android Beacon Library version

2.10

halindrome, 2017-06-14:
I am seeing similar behavior - although it is not always after 30.  With a long running scan for two different UUIDs it will at some point stop seeing any advertisements at all.  I added a watchdog to my code that cycles the bluetooth off and on, which seems to kickstart the thing.  Is there anything I can capture via logcat that might help?

davidgyoung, 2017-06-15:
Upon further research, it seems Android 7.0 introduced a BLE scan timeout, where any scan running for 30 minutes or more is effectively stopped automatically and only resumed "opportunistically" 

You can see the code that does this in the AOSP source here:

https://android.googlesource.com/platform/packages/apps/Bluetooth/+/android-7.0.0_r1/src/com/android/bluetooth/gatt/ScanManager.java#72

A workaround for this is to not keep scans going that long. You can simply stop them and restart them periodically.

fabriziogueli, 2017-06-19:
Hi @davidgyoung , i made some tests with my Samsung Galaxy S7 (Android 7) and with my Huawei Nexus 6P (Android 7.1.2). I started ranging with foregroundScanPeriod set to 1100 ms and the library seems work as expected. The ranging not stop after 30 minutes. I don't see the BtGatt logs that you mentioned. 

Library version: 2.9.2

davidgyoung, 2017-06-19:
@fabriziogueli you are correct that you will not see these symptoms in 2.9.x and earlier versions of the library.  This is because those versions inadvertently circumvented these Android 7 long scan protections by always stopping and restarting scanning  at the end of each `beaconManager.getForegroundScanPeriod()` (1100 ms by default) in order to reset the detected devices cache, which is needed on older Android devices.  Stopping and restarting scanning causes missed detections at the stop/start boundary, so it is not desirable to do unless necessary.  Version 2.10 added code that automatically detects if clearing the detected devices cache is unnecessary, and if that is the case, no longer stops scanning at the end of each cycle if the `beaconManager.getForegroundBetweenScanPeriod()` is zero (which it is by default.)

fabriziogueli, 2017-06-19:
@davidgyoung  Thanks for the explanation

davidgyoung, 2017-06-30:
A workaround has been published in 2.11.

closed 2017-06-30 by davidgyoung

ElieMakhoul, 2017-07-13:
@davidgyoung would really appreciate clarity on behavior of 2.11;
There's no more need to use setForegroundBetweenScanPeriod and setBacgroundBetweenScanPeriod?

davidgyoung, 2017-07-13:
Android 7+ will block scanning that lasts for 30 minutes or more without stopping.  So if you have a betweenScanPeriod of 0, and you scan for 30 minutes, the operating system will cut off your scan on Android 7.

Library versions 2.0-2.9 prevented the above from happening (inadvertently) so long as the scanPeriod was 30 minutes or less.  These library versions always stopped scanning and restarted it at the end of a scan cycle.

Library version 2.10 allowed the OS to block scans as described above, because it no longer stopped scanning between cycles if the betweenScanPeriod is 0.  This library version provided no way of circumventing this 30 minute limit by the operating system.

Library version 2.11+ works like 2.10.  Constant scans of 30+ minutes will be blocked by Andorid 7+. 
 But this library version provides a way to keep constantly scanning for longer than 30 minutes if you really have to.  To do this, you add the following to your AndoridManifest.xml:

```
    <service android:name="org.altbeacon.beacon.service.BeaconService"
             tools:node="replace">
        <meta-data android:name="longScanForcingEnabled" android:value="true"/>
    </service>
```

See here for details:  https://github.com/AltBeacon/android-beacon-library/pull/529

halindrome, 2017-07-13:
I can confirm that this works very well.  I am having some other problems
with scanning, but I suspect they are not related to this issue.

ElieMakhoul, 2017-07-13:
Thanks much @davidgyoung @halindrome.
For clarification, I do need to be always scanning but I run my beacon at slower paste (2.5sec adv. interval).
The implementation is targeted at the enterprise market and the focus is on reliability (no data missed).
I set Foreground/Background scanning period to 6sec in order to make sure the service had enough time to capture two scans.

Do you see any concerns with that flow, or can you recommend other optimal settings to ensure reliability while ensuring some level of battery optimization?

Thanks again

davidgyoung, 2017-07-13:
Two thoughts:

1. If you have a betweenScanPeriod of 0 it really does not matter what you set the scanPeriod to (in terms of battery), because the library will no longer stop scanning in version 2.10+ so long as you have a modern Android unit that can detect multiple BLE adverts from a unique device in the same scan cycle.  The only thing the scanPeriod will really affect in this case, is the frequency of callbacks to `didRangeBecaconsInRegion`.  Depending on how quickly you want these, I would leave it at about the default of 1100ms.  Setting it longer adds lag.  Setting it shorter burns more CPU.

2. Constant scanning certainly will burn a lot of battery -- probably 4x what you would burn from just having the cell standby going.    I understand this is appropriate for some use cases, which is why we added the ability to configure this long scanning for folks who are really sure they want to do it.  But if you really do want "some level of battery optimization" you really shouldn't be scanning constantly for 30 minutes or more.

andersborgabiro, 2017-09-09:
For my understanding:

Do I need to do more (e.g. set foregroundScanPeriod and the "between" setting) if using the longScanForcingEnabled=true approach?

Is it possible to set this value programmatically?

Cheers

davidgyoung, 2017-09-09:
You simply set the entry in the manifest, then you may set as long of a scanPeriod as you wish. The library will automatically stop scanning every 30 minutes if this flag is set in the manifest which will circumvent one specific Android 7 Open Source scanning restriction, preventing scans longer that 30 min.

Note that there is no guarantee that this will let you scan forever on all devices -- they may have other ROM-specific restrictions that block it.  

No, there is no way to set this programmatically.  It is intended to be slightly difficult to set up so as to respect Android restrictions.

andersborgabiro, 2017-09-09:
Thanks for the clarification.

andersborgabiro, 2017-09-12:
Is it correct that the fix is activated by default? That's what Peter Metz indicated when he upgraded the library used in cordova-plugin-ibeacon.

g4gaj, 2017-09-15:
Hi, I am not able to import the library project in android studio, need help, I dont want to use .aar

davidgyoung, 2017-09-15:
@g4gaj, there are instructions here: 
http://altbeacon.github.io/android-beacon-library/alternate-configuration.html

If you need more help, please open a question on StackOverflow.com

g4gaj, 2017-09-19:
Hi, Actually i want to import the source code of library in android studio, but i am unable to import.

davidgyoung, 2017-09-20:
@g4gaj, I created a separate issue here:  https://github.com/AltBeacon/android-beacon-library/issues/596.  Please see there and comment on that thread so as  not to hijack this thread.