AltBeacon/android-beacon-library #528: Ranging stops seemingly randomly

halindrome, 2017-06-15:
(with the understanding that this is NOT in the context of the sample app)

We are using the cordova-plugin-ibeacon module.  This module is built atop THIS library.  As you are aware, on Android 7 there are issues with scanning too frequently.  Thanks for working on ways around that limitation!  We have installed version 2.10 of this library as a jar file into that plugin.  When operating with that configuration, we are seeing some really odd behavior - as documented below.  

NOTE: This is all happening in the foreground.  We are aware there are other things that might happen in the background, but we wanted to reduce the number of variables so that diagnosing this might be easier.

NOTE 2: We think this is related to #526 but wanted to open this thread because 1) it is under the ibeacon cordova plugin and 2) we have detailed logs that demonstrate the behavior.

### Expected behavior

Set up ranging for two different beacons via the ibeacon plugin.  Ranging is continuous.  As beacons come in range, receive call backs from the plugin with their information.  This should continue until we stop ranging, as there is no timeout set.

### Actual behavior

Everything starts as expected.  Then, at some seemingly random point, the AltBeacon library stops relaying beacon information to the ibeacon plugin's callbacks.   It appears "hung".  Initially we assumed it was the BT subsystem locking up, but going into the bluetooth settings panel on the android device shows that BT is still "on" and that it can see BT devices advertising.

Using 'adb logcat' and enabling debugging in the AltBeacon (and ibeacon plugin)  shows something interesting.  

First, we see that the scan is starting:

```
06-15 09:52:37.564  1628  1644 D BtGatt.GattService: start scan without filters
06-15 09:52:37.564  2967  2967 D SystemWebChromeClient: file:///android_asset/www/js/bt.js: Line 322 : Not adding a stop timer - there is no interval set
06-15 09:52:37.565  2967  2967 I chromium: [INFO:CONSOLE(322)] "Not adding a stop timer - there is no interval set", source: file:///android_asset/www/js/bt.js (322)
06-15 09:52:37.566  2967  2967 D SystemWebChromeClient: file:///android_asset/www/js/bt.js: Line 322 : Not adding a stop timer - there is no interval set
06-15 09:52:37.566  2967  2967 I chromium: [INFO:CONSOLE(322)] "Not adding a stop timer - there is no interval set", source: file:///android_asset/www/js/bt.js (322)
06-15 09:52:37.566   942  2912 W PackageManager: checkUidPermission(): android.permission.PEERS_MAC_ADDRESS of 10097 is denied.
06-15 09:52:37.566   287   287 I DisplayDevice: [Built-in Screen (type:0)] fps:17.124743,dur:1401.48,max:900.31,min:14.52
06-15 09:52:37.568  2967  3544 D OpenGLRenderer: ~CanvasContext() 0x7c5c1f8800
06-15 09:52:37.569  2967  3544 D OpenGLRenderer: ~CanvasContext() 0x7c8335c800
06-15 09:52:37.571  2967  3544 D OpenGLRenderer: ~CanvasContext() 0x7c83359000
06-15 09:52:37.571  1628  1719 D BtGatt.ScanManager: handling starting scan
06-15 09:52:37.571  2967  3565 I BluetoothLeScanner: startRegisteration: mLeScanClients={org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$4@7d9ef2c=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@6b26df5}
06-15 09:52:37.572  1628  1719 D BluetoothAdapter: getLeState() returning 12
06-15 09:52:37.573  1628  1719 D BluetoothAdapter: getLeState() returning 12
06-15 09:52:37.573  1628  1719 D BluetoothAdapter: getLeState() returning 12
06-15 09:52:37.574  1628  1719 I BtGatt.ScanManager: configureScanFilters: client=com.android.bluetooth.gatt.ScanClient@24
06-15 09:52:37.574  1628  1719 I BtGatt.ScanManager: getDeliveryMode:  DELIVERY_MODE_IMMEDIATE
06-15 09:52:37.574  1628  1719 I BtGatt.ScanManager: gattClientScanFilterEnableNative(com.android.bluetooth.gatt.ScanClient@24,true);
06-15 09:52:37.578  1628  1701 D BtGatt.GattService: onScanFilterEnableDisabled() - clientIf=5, status=0, action=1
06-15 09:52:37.578  1628  1701 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
06-15 09:52:37.578  1628  1719 I BtGatt.ScanManager: configureScanFilters: shouldUseAllPassFilter
06-15 09:52:37.578  1628  1719 I BtGatt.ScanManager: getDeliveryMode:  DELIVERY_MODE_IMMEDIATE
06-15 09:52:37.578  1628  1719 D BtGatt.ScanManager: configureFilterParamter 500 10000 1 0
06-15 09:52:37.578  1628  1719 I BtGatt.ScanManager: configureFilterParamter: deliveryMode=0 ,rssiThreshold=-128
06-15 09:52:37.578  1628  1719 I BtGatt.ScanManager: gattClientScanFilterParamAddNative
06-15 09:52:37.581  1628  1701 D BtGatt.GattService: onScanFilterParamsConfigured() - clientIf=5, status=0, action=0, availableSpace=49
06-15 09:52:37.581  1628  1701 D BtGatt.ScanManager: callback done for clientIf - 5 status - 0
06-15 09:52:37.581  1628  1719 I BtGatt.ScanManager: gattClientScanNative(true);
06-15 09:52:37.581  1628  1719 D BtGatt.ScanManager: configureRegularScanParams() - queue=1
06-15 09:52:37.581  1628  1719 D BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=2 mLastConfiguredScanSetting=-2147483648
06-15 09:52:37.581  1628  1719 I BtGatt.ScanManager: gattClientScanNative(false);
06-15 09:52:37.581  1628  1719 D BtGatt.ScanManager: configureRegularScanParams - scanInterval = 8000configureRegularScanParams - scanWindow = 8000
06-15 09:52:37.581  1628  1719 I BtGatt.ScanManager: gattClientScanNative(true);
06-15 09:52:37.581  1628  1701 D BtGatt.GattService: onScanParamSetupCompleted : 0
```

AltBeacon correctly determines that the device in question can return multiple detections per scan so it does not cycle the scanning (thus avoiding the Android 7 nonsense):

```
06-15 09:52:38.656  2967  2967 D CycledLeScanner: Not stopping scanning.  Device capable of multiple indistinct detections per scan.
06-15 09:52:38.656  2967  2967 D CycledLeScanner: starting a new scan cycle
06-15 09:52:38.656  2967  2967 D CycledLeScanner: We are already scanning
06-15 09:52:38.657  2967  2967 D CycledLeScanner: Waiting to stop scan cycle for another 1100 milliseconds
06-15 09:52:38.657  2967  2967 D CycledLeScanner: Scan started
```

Scanning proceeds, returning information to the ibeacon plugin's BeaconParser instance.  Everything is working great!

Then, about 30 minutes later (in this example) the beacon scanner for some reason decides to stop scanning:

```
06-15 10:22:38.396  1628  1719 D BtGatt.ScanManager: clientIf set to scan opportunisticly: 5
06-15 10:22:38.396  1628  1719 D BtGatt.ScanManager: configureRegularScanParams() - queue=1
06-15 10:22:38.396  1628  1719 D BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=-1 mLastConfiguredScanSetting=2
06-15 10:22:38.396  1628  1719 D BtGatt.ScanManager: configureRegularScanParams() - queue emtpy, scan stopped
06-15 10:22:38.396  1628  1719 D BtGatt.ScanManager: stop scan
```

### Steps to reproduce this behavior

I have not created a simple test case for this, but it is 100% reproducible in our environment.  I feel like the 30 minutes thing is telling.  It is also interesting that
it says "queue empty".  We have not removed anything from the queue.  

### Mobile device model and OS version

This is a BlackView bv6000 running their lightly customized Android 7.0.  It is basically stock 7.0

### Android Beacon Library version

2.10

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-06-15:
Thanks for this report.   I have confirmed  the same result in #526, so I am going to mark this issue as a duplicate and continue tracking there.  

Android 7+ will stop scanning if it continues without stopping at all for 30 minutes or more. This was added as a feature to prevent battery drain in case developers had inadvertently or abusively left scanning on forever.

You can work around the problem by configuring a small between scan period (just 100ms would be plenty) to force stopping and restarting scanning.   But understand that this is essentially working against what Android 7 is trying to do here.

I'd be curious to hear what you think the library should do in this case.   I don't think it is wise to automatically circumvent this protection in Android 7 unless the user explicitly asks to do so.

closed 2017-06-15 by davidgyoung

halindrome, 2017-06-15:
I understand the work around - we have implemented it already.   

What do I think?  Hmm.  I think the people at Google made a mistake in introducing a breaking change without telling anyone.

As to how it should deal with it... I think if a library user requested continuous scanning, the library should provide that service.  So on 7 (and later?) I think it would be reasonable to trigger an automatic restart if the scan times out.  This would make the behavior backward compatible for people who wanted to scan all the time (like a BT traffic monitor, for example).

If that doesn't feel quite right to you, or if you wanted to be more forward looking, perhaps you could provide another callback hook that a consumer could register for that would get invoked if the OS terminates a scan (for whatever reason).  Then smart consumers could react as they need in a way that is potentially forward compatible.

Anyway, thanks for the hard work on this.  I would be happy to help out on enhancements.  Let me know!

davidgyoung, 2017-06-17:
@halindrome, I have added the ability to trigger an automatic restart in this case in #529 as discussed.  I'd be curious to hear your thoughts on the mechanism I put in for enabling this.

BTW, it is an interesting idea you offered.   about adding a callback when the OS force stops a scan.  However, I don't think that is possible as the only way I know of to detect when it happens is to look at LogCat, which apps are restricted from doing.

halindrome, 2017-06-17:
Good point.   I was hoping there was a way to subscribe to a notification about being relegated to an opportunistic scan.  Guess not :/

SEAbdulbasit, 2019-01-29:
Hello @danialgoodwin 
i am integrating alt beacon library in our android app. it is working fine in foreground. i want my app to be event driven, as i enter in the vicinity of beacon, my app should open. But the app opens after 5 min if i finish the application. i have also changed its scan period and between scan period time but no luck.

davidgyoung, 2019-01-29:
@SEAbdulbasit, the best way to get help with using the library with your project is to post a new question on StackOverflow.com.  I'm glad to answer there.  Please include (1) The library version you are using (2) The Operating System version and phone model you are testing with, (3) Show the code you use to set up beacon monitoring that shows the context of where you put the setup code in your app.

SEAbdulbasit, 2019-01-30:
@davidgyoung  sorry to bother you sir, but i don,t have the required reputation to post question on Stakeoverflow. 
1- libarary version 
    implementation 'org.altbeacon:android-beacon-library:2+'
2- operating system version
   7.0
3- code
I am using the code which is given in the examples of the library. It is detecting perfectly but after 5 min of app closing. I have read that above android 8.0 they are not allowing services to run more than 15 min in background. But i am using lower version so it should not have any issue to search continuously.

package com.example.beacondetector;

import android.app.Application;
import android.app.NotificationManager;
import android.app.PendingIntent;
import android.app.TaskStackBuilder;
import android.content.Context;
import android.content.Intent;
import android.os.RemoteException;
import android.support.v4.app.NotificationCompat;
import android.util.Log;
import android.widget.Toast;

import org.altbeacon.beacon.Beacon;
import org.altbeacon.beacon.BeaconConsumer;
import org.altbeacon.beacon.BeaconManager;
import org.altbeacon.beacon.RangeNotifier;
import org.altbeacon.beacon.Region;
import org.altbeacon.beacon.startup.BootstrapNotifier;
import org.altbeacon.beacon.startup.RegionBootstrap;
import org.altbeacon.beacon.utils.UrlBeaconUrlCompressor;

import java.util.Collection;
import java.util.Collections;

public class MyApplicationName extends Application implements BootstrapNotifier, RangeNotifier, BeaconConsumer {
    private static final String TAG = ".MyApplicationName";
    private RegionBootstrap regionBootstrap;
    BeaconManager beaconManager;

    @Override
    public void onCreate() {
        super.onCreate();
        Log.d(TAG, "App started up");
        beaconManager = BeaconManager.getInstanceForApplication(this);
        beaconManager.setBackgroundScanPeriod(1000l);
        beaconManager.setDebug(true);
        beaconManager.setBackgroundBetweenScanPeriod(3000l);
        try {
            beaconManager.updateScanPeriods();
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        beaconManager.setBackgroundMode(true);
        Region region = new Region("com.example.myapp.boostrapRegion", null, null, null);
        regionBootstrap = new RegionBootstrap(this, region);
    }

    @Override
    public void didDetermineStateForRegion(int arg0, Region arg1) {
        // Don't care
    }

    @Override
    public void didEnterRegion(Region arg0) {
        Log.d(TAG, "Got a didEnterRegion call");
        regionBootstrap.disable();
        Intent intent = new Intent(this, MainActivity.class);
        intent.setFlags(Intent.FLAG_ACTIVITY_NEW_TASK);
        String  uniqueId=arg0.getUniqueId();
        String  bluetoothAddress=arg0.getBluetoothAddress();
        Log.d(TAG, "Unique Identified ID : "+uniqueId);
        intent.putExtra("uniqueIdentifier",uniqueId);
        intent.putExtra("bluetoothAddress",bluetoothAddress);

        NotificationCompat.Builder builder =
                new NotificationCompat.Builder(this)
                        .setContentTitle("Beacon Reference Application")
                        .setContentText("A beacon is nearby.")
                        .setSmallIcon(R.drawable.ic_launcher_background);

        TaskStackBuilder stackBuilder = TaskStackBuilder.create(this);
        stackBuilder.addNextIntent(new Intent(this, MainActivity.class));
        PendingIntent resultPendingIntent =
                stackBuilder.getPendingIntent(
                        0,
                        PendingIntent.FLAG_UPDATE_CURRENT
                );
        builder.setContentIntent(resultPendingIntent);
        NotificationManager notificationManager =
                (NotificationManager) this.getSystemService(Context.NOTIFICATION_SERVICE);
        notificationManager.notify(1, builder.build());

        Toast.makeText(getApplicationContext(),"Hello, i found a beacon",Toast.LENGTH_LONG).show();
        if(arg0.getId2()!=null) {
            Toast.makeText(getApplicationContext(), arg0.getId2().toString(), Toast.LENGTH_LONG).show();
        }
        startActivity(intent);

    }

    @Override
    public void didExitRegion(Region arg0) {
        // Don't care
    }

    @Override
    public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
        for (Beacon beacon: beacons) {
            if (beacon.getServiceUuid() == 0xfeaa && beacon.getBeaconTypeCode() == 0x10) {
                // This is a Eddystone-URL frame
                String url = UrlBeaconUrlCompressor.uncompress(beacon.getId1().toByteArray());
                Log.d(TAG, "I see a beacon transmitting a url: " + url +
                        " approximately " + beacon.getDistance() + " meters away.");
            }
        }
    }

    @Override
    public void onBeaconServiceConnect() {
        beaconManager.removeAllRangeNotifiers();
        beaconManager.addRangeNotifier(new RangeNotifier() {
            @Override
            public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
                if (beacons.size() > 0) {
                    Log.i(TAG, "The first beacon I see is about "+beacons.iterator().next().getDistance()+" meters away.");
                    Toast.makeText(MyApplicationName.this, "The first beacon I see is about "+beacons.iterator().next().getDistance()+" meters away.", Toast.LENGTH_SHORT).show();
                }
            }
        });

        try {
            beaconManager.startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
        } catch (RemoteException e) {    }
    }
}

davidgyoung, 2019-01-30:
@SEAbdulbasitplease do not add new questions as comments to issues (especially closed ones) as this pollutes discussion threads.  Again, I am happy to help on StackOverflow.com.  While you need a certain reputation level to comment on existing questions, new users are allowed to open up new questions with no special repurltation.

mmcor, 2019-06-22:
I love the library and librarian t-shirt great tee 

https://amzn.to/2FrGwjp

https://amzn.to/2X0jYkJ

https://amzn.to/2ZFbpsc