AltBeacon/android-beacon-library #531: Nexus 7 not able to detect beacons while it is detected on other devices

vikas344041, 2017-06-23:
### Expected behavior

### Actual behavior

### Steps to reproduce this behavior

nexus 7 android 6.0.1

2+

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-06-23:
There are two models of the Nexus 7, the original and the 2013 edition.  Only the latter models have BLE and can detect beacons.  Check you device model in settings and see if it is one of the following codes (there may be others) from the original edition:  KOT49H, KTU84L, KTU84Q

davidgyoung, 2017-07-12:
@vikas344041, can you please check if the Nexus 7 is one of the models I listed withoug BLE?