AltBeacon/android-beacon-library #534: the size of collections in didRangeBeaconsInRegion is always 1

wangbincumt, 2017-07-03:
I have 4 beacons which in  corners of the room
I set DEFAULT_BACKGROUND_SCAN_PERIOD=2000L In my code

in didRangeBeaconsInRegion，
 Log.e("beaconservice", collections.size()+"");
i find the size is always equal 1，the beacon info is one of thems；
how to get all beacons info in one times

davidgyoung, 2017-07-03:
Do the beacons have different identifiers or are they all the same?

wangbincumt, 2017-07-03:
the uuid of the beacons is the same

davidgyoung, 2017-07-03:
By default, if the UUID, major and minor are all the same, the library will treat the the four beacons as only one.  If you want it to treat them as 4, you can set:

`Beacon.setHardwareEqualityEnforced(true);`

wangbincumt, 2017-07-03:
thank you very much。it works。
i used a app and find the UUID, major and minor are all the same.
Now，collections.size() is
 random From 0 to 4。
i will config the beacons and try again

closed 2017-07-12 by davidgyoung