AltBeacon/android-beacon-library #535: Public Beacon class method to retrieve the current average RSSI

peterdeka, 2017-07-04:
### Expected behavior
The method should return the current averaged RSSI for the beacon.
### Actual behavior
In the docs there is no method. In the latest release there is a `getRunningAverageRssi` method but the code is not clear: it gets a parameter RSSI?? and returns that same RSSI also assigning it to the class' property (nonsense for me).
### Steps to reproduce this behavior
Feature request
### Mobile device model and OS version
Any
### Android Beacon Library version
2.11
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-07-04:
Yes, that is obviously a mistake.  The definition should simply be:

```
public double getRunningAverageRssi() {
    return mRunningAverageRssi;
}
```

peterdeka, 2017-07-04:
Thanks a lot @davidgyoung as the last release was a few days ago i suppose it will be a long wait?! :D any way i can help to speed things up?

davidgyoung, 2017-07-04:
If you are OK with using a beta release I am planning one for the end of the week.

peterdeka, 2017-07-05:
@davidgyoung if it can help i opened a PR #536

davidgyoung, 2017-08-04:
#536 has been merged.

closed 2017-08-04 by davidgyoung