AltBeacon/android-beacon-library #539: Android O - BeaconReference app has stopped : beaconManager.setEnableScheduledScanJobs(true)

danielBC-KIM, 2017-07-12:
### Expected behavior
Using "android-beacon-library-reference" app on Nexus 5X running with Android 8.0.0
Click "Start Ranging" button -->> display Beacon information near me 
 
### Actual behavior
Click "Start Ranging" button -->> App stopped

### Steps to reproduce this behavior
1. Download and add the the latest library in Android Studio Project

Download "android-beacon-library-reference" app and follow the steps described in the release note(i.e. android-beacon-library-2.12-beta1) , Android O Support. 

2. Code changes and its results
   - Disable scan job
       
        a. Changes :    

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
        beaconManager.setEnableScheduledScanJobs(false);

        b. Results : Success
        When selecting a "Start Ranging" button, UI displays Beacons near me. No issue, success.
        When checking "Running services" in Developer options, BeaconReference app shows one running service, Beacon

   - Enable scan job

        a. Changes :

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));
        beaconManager.setEnableScheduledScanJobs(true);

        b. Results : Failure
        When selecting a "Start Ranging" button, UI displays an exception popup, BeaconReference has stopped 

        c. Logcat :Fatal Exception 
```
07-12 14:03:29.405 24427-24816/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
07-12 14:03:29.417 24427-24427/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
07-12 14:03:29.483 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
07-12 14:03:29.483 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified flex for 1 is 0. Clamped to +5m0s0ms
07-12 14:03:29.484 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
07-12 14:03:29.484 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified flex for 1 is 0. Clamped to +5m0s0ms
07-12 14:03:29.509 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified interval for 1 is +5m0s0ms. Clamped to +15m0s0ms
07-12 14:03:29.510 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified flex for 1 is 0. Clamped to +5m0s0ms
07-12 14:03:29.510 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified interval for 1 is +5m0s0ms. Clamped to +15m0s0ms
07-12 14:03:29.510 24427-24427/org.altbeacon.beaconreference W/JobInfo: Specified flex for 1 is 0. Clamped to +5m0s0ms
07-12 14:03:29.587 24427-24427/org.altbeacon.beaconreference I/AssistStructure: Flattened final assist data: 1396 bytes, containing 1 windows, 4 views
07-12 14:03:29.588 24427-24427/org.altbeacon.beaconreference I/ScanJob: Running immdiate scan job: instance is org.altbeacon.beacon.service.ScanJob@18fdb81
07-12 14:03:29.589 24427-24427/org.altbeacon.beaconreference I/ScanJob: scanJob version 2.12-beta1 is starting up on the main process
07-12 14:03:29.603 24427-24427/org.altbeacon.beaconreference I/CycledLeScanner: Using Android O scanner
07-12 14:03:29.606 24427-24427/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
07-12 14:03:29.616 24427-24427/org.altbeacon.beaconreference I/ScanJob: Scan job running for 300000 millis
07-12 14:03:29.617 24427-24858/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
07-12 14:03:29.619 24427-24440/org.altbeacon.beaconreference D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=11 mScannerId=0
07-12 14:03:30.720 24427-24427/org.altbeacon.beaconreference D/AndroidRuntime: Shutting down VM
07-12 14:03:30.724 24427-24427/org.altbeacon.beaconreference E/AndroidRuntime: FATAL EXCEPTION: main
                                                                               Process: org.altbeacon.beaconreference, PID: 24427
                                                                               java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.altbeacon.beacon.service.Callback.call(android.content.Context, java.lang.String, android.os.Bundle)' on a null object reference
                                                                                   at org.altbeacon.beacon.service.ScanHelper.processRangeData(ScanHelper.java:262)
                                                                                   at org.altbeacon.beacon.service.ScanHelper.access$200(ScanHelper.java:59)
                                                                                   at org.altbeacon.beacon.service.ScanHelper$1.onCycleEnd(ScanHelper.java:213)
                                                                                   at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:373)
                                                                                   at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:363)
                                                                                   at org.altbeacon.beacon.service.scanner.CycledLeScanner$2.run(CycledLeScanner.java:359)
                                                                                   at android.os.Handler.handleCallback(Handler.java:769)
                                                                                   at android.os.Handler.dispatchMessage(Handler.java:98)
[... 53 lines cut ...]
                                                                                Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.altbeacon.beacon.service.Callback.call(android.content.Context, java.lang.String, android.os.Bundle)' on a null object reference
                                                                                   at org.altbeacon.beacon.service.ScanHelper.processRangeData(ScanHelper.java:262)
                                                                                   at org.altbeacon.beacon.service.ScanHelper.access$200(ScanHelper.java:59)
                                                                                   at org.altbeacon.beacon.service.ScanHelper$1.onCycleEnd(ScanHelper.java:213)
                                                                                   at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:373)
                                                                                   at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:337)
                                                                                   at org.altbeacon.beacon.service.scanner.CycledLeScanner.stop(CycledLeScanner.java:226)
                                                                                   at org.altbeacon.beacon.service.ScanJob.stopScanning(ScanJob.java:140)
                                                                                   at org.altbeacon.beacon.service.ScanJob.onStopJob(ScanJob.java:133)
                                                                                   at android.app.job.JobService$1.onStopJob(JobService.java:76)
                                                                                   at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:117)
                                                                                   at android.os.Handler.dispatchMessage(Handler.java:105) 
                                                                                   at android.os.Looper.loop(Looper.java:164) 
                                                                                   at android.app.ActivityThread.main(ActivityThread.java:6541) 
                                                                                   at java.lang.reflect.Method.invoke(Native Method)
```

### Mobile device model and OS version
Mobile device : Nexus5X
OS version : Android 8.0.0
 
### Android Beacon Library version
android-beacon-library-2.12-beta1.aar

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2017-07-12:
Thanks for this report.   This is caused by a bug in deserializing the the Scan settings.  I am investigating.

davidgyoung, 2017-07-12:
Fixed in https://github.com/AltBeacon/android-beacon-library/pull/484/commits/956eff4e623e9ce9caea49cda74c9c49b5982b6c

I have verified this prevents crashes on ranging callbacks, and that ranging results are received correctly.

This has been rolled into a new beta2 binary release.   See latest here:  https://github.com/AltBeacon/android-beacon-library/releases

closed 2017-07-12 by davidgyoung